dusty-nv/jetson-containers #538: What's the plan to support ROS2 Jazzy

tonynajjar, 2024-05-23:
ROS2 Jazzy is fresh out of the oven. @dusty-nv what needs to be done to support it? Maybe I can help

dusty-nv, 2024-05-23:
@tonynajjar haven't looked at this yet and am unsure to the extent that Ubuntu 22.04 is supported, but you could add it here and then it should show up as a container that you could try to build:

https://github.com/dusty-nv/jetson-containers/blob/040a0ac03be8a84d38e55bbb2a954846ac23dccc/packages/ros/config.py#L4

tonynajjar, 2024-05-23:
> @tonynajjar haven't looked at this yet and am unsure to the extent that Ubuntu 22.04 is supported

Ah right, I forgot that we are restricted by the ubuntu 22 because of Jetpack. Judging from history, Jetpack for ubuntu 24 is probably not coming any time soon. ROS2 Jazzy does not support ubuntu 24 (at least not officially) so I'll close this issue.

This recurring timeline mismatch between Jetpack and Ubuntu is quite a pain; maybe there is a good reason I don't know about though.

closed 2024-05-23 by tonynajjar

dusty-nv, 2024-05-23:
I compile ROS from source for that reason, and other ROS distros we are able to compile on different versions of JetPack/Ubuntu - just haven't tried out Jazzy on Ubuntu 22.04 yet.  I guess I was going to let the dust settle from the release first - I saw others on social asking similar questions about Jazzy on OS other than 24.04 (especially since 24.04 itself was just released and most people have not changed yet)