dusty-nv/jetson-containers #540: PyPI index (http://jetson.webredirect.org) is down

ai-and-i, 2024-05-27:
I'm trying to build packages and getting this error:

```
Looking in indexes: http://jetson.webredirect.org/jp6/cu124
Processing ./torchvision-0.18.0a0+6043bc2-cp310-cp310-linux_aarch64.whl
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object at 0xffff9427e6b0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu124/pillow/
```

It looks like the default PyPI server is down.

dusty-nv, 2024-05-28:
Sorry about that @ai-and-i , @DmytroIvakhnenkov , it should be back up now.

ai-and-i, 2024-05-28:
Works now, thanks for the quick fix!

closed 2024-05-28 by ai-and-i