dusty-nv/jetson-containers #542: Problems using XTTS

hslr4, 2024-05-29:
I found some problems when using xtts.

- I had to install `libsndfile1-dev` e.g. by adding `RUN apt-get update && apt-get clean -y && apt-get install -y --no-install-recommends libsndfile1-dev && rm -rf /var/lib/apt/lists/*` to the `Dockerfile`. I think `soundfile` needs it and causes an exception like `OSError: cannot load library 'libsndfile.so': libsndfile.so: cannot open shared object file: No such file or directory` or similar when it is missing.
- When using tensorrt `flag` is not defined in the xtts fork here: https://github.com/dusty-nv/TTS/blob/b452e628316e0fe33b8842e5d6ec1eb01fc46ef3/TTS/tts/models/xtts.py#L804 I changed it to `use_fp16 = (use_tensorrt == 'fp16' if isinstance(use_tensorrt, str) else False)`

I think there is still the streaming test failing when building the image, however I currently don't need this functionality 😅

dusty-nv, 2024-05-30:
Aha okay yes, I had tried getting xtts working again and ran into the libsoundfile thing, but was not able to get the streaming test working again

dusty-nv, 2024-06-19:
XTTS streaming is working again with https://github.com/dusty-nv/jetson-containers/commit/f14a7771f3383620df6e60f125fd35a999fb8953

Build/tested/pushed the container image for JP6 to [dustynv/xtts:r36.3.0](https://hub.docker.com/r/dustynv/xtts/tags)