dusty-nv/jetson-containers #551: jetson-containers build [package] results in error

Stuveman, 2024-06-05:
Hello, I did the whole setup guide with setting up docker, cloning the repo and executing the install script.
(I'm using the Nvidia Jetson AGX Orin)
I want to adjust the dockerfile in jetson-containers/packages/audio/whisper/Dockerfile for installing additional Ubnutu packages. So I wanted to create a new docker image with **jetson-containers build whisper** but I ran into following errors:
```
$ jetson-containers build whisper
Namespace(packages=['whisper'], name='', base='', multiple=False, build_flags='', package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', logs='', verbose=False, no_github_api=False)
-- L4T_VERSION=36.3.0
-- JETPACK_VERSION=6.0
-- CUDA_VERSION=12.5
-- PYTHON_VERSION=3.10
-- LSB_RELEASE=22.04 (jammy)
-- Package ros:noetic-ros-base has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:noetic-ros-core has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:noetic-desktop has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:foxy-ros-base has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:foxy-ros-core has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:foxy-desktop has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:galactic-ros-base has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:galactic-ros-core has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:galactic-desktop has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:humble-ros-base has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:humble-ros-core has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:humble-desktop has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:iron-ros-base has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:iron-ros-core has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ros:iron-desktop has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package tritonserver has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ctranslate2:master has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ctranslate2:master-builder has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ctranslate2:4.2.0 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package ctranslate2:4.2.0-builder has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package cupy has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package pycuda has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package nemo has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package nanosam has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package nanoowl has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package sam has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package efficientvit has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package tam has missing dependencies, disabling...  ("couldn't find package:  sam")
-- Package onnxruntime:1.19 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package onnxruntime:1.19-builder has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package l4t-diffusion has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package l4t-text-generation has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package l4t-tensorflow:tf1 has missing dependencies, disabling...  ("couldn't find package:  tensorflow")
-- Package l4t-tensorflow:tf2 has missing dependencies, disabling...  ("couldn't find package:  pycuda")
-- Package l4t-ml has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package l4t-pytorch has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package openai-triton:2.1.0 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package openai-triton:2.1.0-builder has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package deepstream has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package arrow:14.0.1 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package arrow:12.0.1 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package arrow:5.0.0 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package numba has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package xtts has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package audiocraft has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package voicecraft has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package whisper_trt has missing dependencies, disabling...  ("couldn't find package:  onnxruntime")
-- Package faster-whisper has missing dependencies, disabling...  ("couldn't find package:  ctranslate2")
-- Package whisperx has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package piper-tts has missing dependencies, disabling...  ("couldn't find package:  onnxruntime")
-- Package whisper has missing dependencies, disabling...  ("couldn't find package:  numba")
-- Package jetson-inference has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package torch2trt has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package torch_tensorrt has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package torchvision:0.16.2 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package torchvision:0.16.2-builder has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package torchvision:0.17.2 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package torchvision:0.17.2-builder has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package torchvision:0.18.0 has missing dependencies, disabling...  ("couldn't find package:  cuda")
[... 86 lines cut ...]
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/media/jetson/ExtraStorage/jetson-containers/jetson_containers/build.py", line 103, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/media/jetson/ExtraStorage/jetson-containers/jetson_containers/container.py", line 66, in build_container
    packages = resolve_dependencies(packages)
  File "/media/jetson/ExtraStorage/jetson-containers/jetson_containers/packages.py", line 316, in resolve_dependencies
    packages, changed = add_depends(packages)
  File "/media/jetson/ExtraStorage/jetson-containers/jetson_containers/packages.py", line 288, in add_depends
    for dependency in find_package(package).get('depends', []):
  File "/media/jetson/ExtraStorage/jetson-containers/jetson_containers/packages.py", line 174, in find_package
    raise KeyError(f"couldn't find package:  {package}")
KeyError: "couldn't find package:  whisper"
```
Also tried other packages like pytorch but the result was always the same, that the package couldn't be found.
But running **jetson-containers run $(autotag whisper)** works without any issues:
```
jetson-containers run $(autotag whisper)
Namespace(packages=['whisper'], prefer=['local', 'registry', 'build'], disable=[''], user='dustynv', output='/tmp/autotag', quiet=False, verbose=False)
-- L4T_VERSION=36.3.0  JETPACK_VERSION=6.0  CUDA_VERSION=12.5
-- Finding compatible container image for ['whisper']
...
```
Would appreciate any help.

dusty-nv, 2024-06-05:
Hi @Stuveman , can you try running this like `CUDA_VERSION=12.2 jetson-containers build whisper` ?

Stuveman, 2024-06-05:
Hey @dusty-nv , worked like a charm, thank you so much. Kinda mad at myself at how easy it was to solve.

closed 2024-06-05 by Stuveman

dusty-nv, 2024-06-05:
Oh don't be, sorry i think that is a bug on my end, because I detect what version of CUDA is installed on the host, when really that should not matter at all.  I have a TODO to remove that and just use a lookup table for what the default version of CUDA should be for each jetpack...