dusty-nv/jetson-containers #553: can not pull ollama docker

johnsonwag03, 2024-06-09:
I  tried to run ollama docker on agx orin and run script "./run.sh  --name ollama $(./autotag ollama)"
got error message :docker: error pulling image configuration: download failed after attempts=6: dial tcp [2a03:2880:f12a:83:face:b00c:0:25de]:443: i/o timeout.
so, how to fix this issue.

the whole message are:
Unable to find image 'dustynv/ollama:r35.4.1' locally
r35.4.1: Pulling from dustynv/ollama
2378679266ac: Retrying in 1 second 
fbe64e031ca2: Retrying in 1 second 
bc84636d073b: Retrying in 1 second 
4c8a78053e02: Waiting 
c98a245e07e2: Waiting 
54e58b7fa9fd: Waiting 
bd70e595c704: Waiting 
70a7c3288501: Waiting 
0cdb0dfb035f: Waiting 
a4c3570418d1: Waiting 
019c19b28a48: Waiting 
ea8790418f30: Waiting 
a9dca93572f1: Waiting 
1a4f4ffe916b: Waiting 
b1285e2f3ec3: Waiting 
2bd4b945552e: Waiting 
a5c388f5daf7: Waiting 
d9fa505958b2: Waiting 
5f5b8fb8b6a8: Waiting 
1b798dc9c4ba: Waiting 
b48c074786ef: Waiting 
30eefdd3bb15: Waiting 
4b3a805e5195: Waiting 
033e7e4face2: Waiting 
c51d63f8bec7: Waiting 
04342bdef642: Waiting 
fc868bb9f6ae: Waiting 
docker: error pulling image configuration: download failed after attempts=6: dial tcp [2a03:2880:f12a:83:face:b00c:0:25de]:443: i/o timeout.
See 'docker run --help'.

dusty-nv, 2024-06-09:
@johnsonwag03 can you try again?  I think it is network related

zhangyacnhao, 2024-06-24:
+1

> I tried to run ollama docker on agx orin and run script "./run.sh --name ollama $(./autotag ollama)" got error message :docker: error pulling image configuration: download failed after attempts=6: [...]

+1

closed 2025-02-21 by johnnynunez