dusty-nv/jetson-containers #559: Version tag/value not bing passed into Ollama build

hillct, 2024-06-18:
Upon deploying the jetson_containers Ollama docker image, and visiting http://localhost:11434/api/vrsion the return value is ```{"version": "0.0.0"}``` which reflects the fact that the version is no being injected to the build via the apparently preferred GOFLAGS method shown here https://github.com/ollama/ollama/blob/main/scripts/build_docker.sh#L5-L6

I would think the easiest way to inject it using the Jeson_Containers build framework, is to pass the value in via a docker build arg https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/ollama/Dockerfile
If I have some time this week I'll submit a fix, but if someone bas me to it, so much the better

dusty-nv, 2024-06-25:
Thanks @hillct - fixed this in https://github.com/dusty-nv/jetson-containers/commit/6536834fae6b5a560ca68bb502e82a5204a83c7d and updated the docker images 👍

hillct, 2024-06-26:
Tested on AGX Orin 32GB dev kit - confirmed operating as expected.
Thanks @dusty-nv

closed 2024-06-26 by hillct