dusty-nv/jetson-containers #560: Unable to run Llamaspeak on Jetson Orin AGX 32GB

Tosho-01, 2024-06-19:
Hi, I'm trying to run Llamaspeak following the Instructions on https://www.jetson-ai-lab.com/tutorial_llamaspeak.html

Specs:
Jetson AGX Orin (32GB) Developer Kit
Jetpack 6.0 [L4T 36.3.0]

The RIVA server is up and running the ASR and TTS examples works just fine.

When i run the following code in /path/to/jetson-containers:

```python
jetson-containers run --env HUGGINGFACE_TOKEN=hf_MYTOKEN \
  $(autotag nano_llm) \
  python3 -m nano_llm.agents.web_chat --api=mlc --verbose\
    --model meta-llama/Meta-Llama-3-8B-Instruct \
    --asr=riva --tts=piper
``` 
The Response is:
```python
Namespace(packages=['nano_llm'], prefer=['local', 'registry', 'build'], disable=[''], user='dustynv', output='/tmp/autotag', quiet=False, verbose=False)
-- L4T_VERSION=36.3.0  JETPACK_VERSION=6.0  CUDA_VERSION=12.2
-- Finding compatible container image for ['nano_llm']
dustynv/nano_llm:r36.2.0
[sudo] password for cv: 
localuser:root being added to access control list
+ docker run --runtime nvidia -it --rm --network host --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /var/run/dbus:/var/run/dbus --volume /var/run/avahi-daemon/socket:/var/run/avahi-daemon/socket --volume /var/run/docker.sock:/var/run/docker.sock --volume /mnt/jetson-containers/data:/data --device /dev/snd --device /dev/bus/usb -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix -v /tmp/.docker.xauth:/tmp/.docker.xauth -e XAUTHORITY=/tmp/.docker.xauth --device /dev/i2c-0 --device /dev/i2c-1 --device /dev/i2c-2 --device /dev/i2c-3 --device /dev/i2c-4 --device /dev/i2c-5 --device /dev/i2c-6 --device /dev/i2c-7 --device /dev/i2c-8 --device /dev/i2c-9 -v /run/jtop.sock:/run/jtop.sock --env HUGGINGFACE_TOKEN=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 dustynv/nano_llm:r36.2.0 python3 -m nano_llm.agents.web_chat --api=mlc --verbose --model meta-llama/Meta-Llama-3-8B-Instruct --asr=riva --tts=piper
/usr/local/lib/python3.10/dist-packages/transformers/utils/hub.py:124: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
13:31:48 | DEBUG | Namespace(model='meta-llama/Meta-Llama-3-8B-Instruct', quantization=None, api='mlc', vision_api='auto', vision_model=None, vision_scaling=None, prompt=None, save_mermaid=None, chat_template=None, system_prompt=None, wrap_tokens=512, max_context_len=None, max_new_tokens=128, min_new_tokens=-1, do_sample=False, temperature=0.7, top_p=0.95, repetition_penalty=1.0, audio_input_device=None, audio_input_channels=1, audio_output_device=None, audio_output_file=None, audio_output_channels=1, list_audio_devices=False, sample_rate_hz=48000, riva_server='localhost:50051', language_code='en-US', tts='piper', tts_buffering='punctuation', voice=None, voice_speaker=None, voice_rate=1.0, voice_pitch='default', voice_volume='default', asr='riva', asr_confidence=-2.5, asr_silence=-1.0, asr_chunk=1600, boosted_lm_words=None, boosted_lm_score=4.0, profanity_filter=False, inverse_text_normalization=False, automatic_punctuation=True, web_host='0.0.0.0', web_port=8050, ws_port=49000, ssl_key='/etc/ssl/private/localhost.key.pem', ssl_cert='/etc/ssl/private/localhost.cert.pem', upload_dir='/tmp/uploads', web_trace=False, web_title=None, log_level='debug', debug=True)
The token has not been saved to the git credentials helper. Pass `add_to_git_credential=True` in this function directly or `--add-to-git-credential` if using via `huggingface-cli` if you want to set the git credential as well.
13:31:48 | DEBUG | Starting new HTTPS connection (1): huggingface.co:443
13:31:48 | DEBUG | https://huggingface.co:443 "GET /api/whoami-v2 HTTP/1.1" 200 727
Token is valid (permission: fineGrained).
Your token has been saved to /data/models/huggingface/token
Login successful
13:31:48 | DEBUG | https://huggingface.co:443 "GET /api/models/meta-llama/Meta-Llama-3-8B-Instruct/revision/main HTTP/1.1" 200 19884
/usr/local/lib/python3.10/dist-packages/huggingface_hub/file_download.py:1132: FutureWarning: `resume_download` is deprecated and will be removed in version 1.0.0. Downloads always resume when possible. If you want to force a new download, use `force_download=True`.
  warnings.warn(
Fetching 13 files: 100%|██████████████████████████████████████████████| 13/13 [00:00<00:00, 9988.27it/s]
13:31:48 | DEBUG | https://huggingface.co:443 "GET /api/models/meta-llama/Meta-Llama-3-8B-Instruct/revision/main HTTP/1.1" 200 19884
Fetching 17 files: 100%|██████████████████████████████████████████████| 17/17 [00:00<00:00, 4197.51it/s]
13:31:48 | INFO | loading /data/models/huggingface/models--meta-llama--Meta-Llama-3-8B-Instruct/snapshots/e1945c40cd546c78e41f1151f4db032b271faeaa with MLC
Special tokens have been added in the vocabulary, make sure the associated word embeddings are fine-tuned or trained.
13:31:50 | INFO | device=cuda(0), name=Orin, compute=8.7, max_clocks=1300000, multiprocessors=16, max_thread_dims=[1024, 1024, 64], api_version=12020, driver_version=None
13:31:50 | INFO | loading Meta-Llama-3-8B-Instruct from /data/models/mlc/dist/Meta-Llama-3-8B-Instruct-ctx8192/Meta-Llama-3-8B-Instruct-q4f16_ft/Meta-Llama-3-8B-Instruct-q4f16_ft-cuda.so
13:31:50 | WARNING | model library /data/models/mlc/dist/Meta-Llama-3-8B-Instruct-ctx8192/Meta-Llama-3-8B-Instruct-q4f16_ft/Meta-Llama-3-8B-Instruct-q4f16_ft-cuda.so was missing metadata
13:31:50 | DEBUG | using prefill_with_embed() from /data/models/mlc/dist/Meta-Llama-3-8B-Instruct-ctx8192/Meta-Llama-3-8B-Instruct-q4f16_ft/Meta-Llama-3-8B-Instruct-q4f16_ft-cuda.so
13:31:50 | DEBUG | using create_kv_cache() from /data/models/mlc/dist/Meta-Llama-3-8B-Instruct-ctx8192/Meta-Llama-3-8B-Instruct-q4f16_ft/Meta-Llama-3-8B-Instruct-q4f16_ft-cuda.so
┌─────────────────────────┬─────────────────────────────────────────────────────────────────────────────┐
│ architectures           │ ['LlamaForCausalLM']                                                        │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ attention_bias          │ False                                                                       │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ attention_dropout       │ 0.0                                                                         │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ bos_token_id            │ 128000                                                                      │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ eos_token_id            │ 128009                                                                      │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ hidden_act              │ silu                                                                        │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ hidden_size             │ 4096                                                                        │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ initializer_range       │ 0.02                                                                        │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ intermediate_size       │ 14336                                                                       │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ max_position_embeddings │ 8192                                                                        │
├─────────────────────────┼─────────────────────────────────────────────────────────────────────────────┤
│ model_type              │ llama                                                                       │
[... 124 lines cut ...]
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/opt/NanoLLM/nano_llm/agents/web_chat.py", line 310, in <module>
[... cut ...]
  File "/opt/NanoLLM/nano_llm/plugins/bot_functions/__init__.py", line 167, in <listcomp>
    docs = '\n'.join(['* ' + x.docs for x in cls.functions if x.enabled])
TypeError: can only concatenate str (not "dict") to str
``` 
I tried to access the \_\_init\_\_.py file in /docker/overlay2/CONTAINER_ID/diff/opt/NanoLLM/nano_llm/plugins/bot_functions/\_\_init\_\_.py - but can only find the .pyc file and can't open or edit those.

Kinda stuck at this point.

Any help or recommendations are appreciated.

dusty-nv, 2024-06-21:
Hi @Tosho-01 , can you try running `dustynv/nano_llm:24.6-r36.2.0` container instead of `$(autotag nano_llm)` ?   I have been doing some restructuring on the repo, sorry about that.   That 24.6 release is from ~2 weeks ago which I think should still be fine.  I will test llamaspeak again and fix this though for the next release.

Tosho-01, 2024-06-21:
Hi @dusty-nv, thank you for your response. The older version is working for me. Had some troubles in Firefox and switched to Chromium where everything is running smoothly. Thank you for putting out all those jetson-containers :+1:

closed 2024-06-21 by Tosho-01

lq0104, 2024-07-07:
sorry for another question, what's the different between docker 'dustynv/nano_llm:24.6-r36.2.0' and 'dustynv/nano_llm:r36.2.0' @dusty-nv