dusty-nv/jetson-containers #574: CSI Camera not working, deciding to use USB camera

Syncron19, 2024-07-16:
Hi everyone!

I am currently working with the jetson orin nano and intended on using a csi camera for inferencing and using jetson-containers. However my camera (Raspberry Pi V2-attached) for some reason is not being detected by my orin. I am running with jetpack 6 right now and have heard that camera issues like this occur. I am wondering if I could use a usb camera and should I downgrade to jetpack 5 to get the csi camera working instead? Ideally, I don't want to have to downgrade, but will more than likely have to if the usb camera doesn't work. Thanks everyone!
[image]

dusty-nv, 2024-07-16:
Hi @Syncron19, how are you trying to use your MIPI CSI camera from code?  If you started your container with `jetson-containers run`, it will automatically mount the Argus daemon's socket, so CSI camera can be used inside container through GStreamer (`nvarguscamerasrc`) or libargus:

https://github.com/dusty-nv/jetson-containers/blob/e77e05ebcbbb15e315bbf8b5bcd160946995a12c/run.sh#L76

Can you check that it works outside of container, using `nvgstcapture-1.0` program?

Syncron19, 2024-07-17:
Hey Dusty, so I downgraded down to jetpack 5 today and it works. For some reason, I couldn't get the camera to work at all on the Orin while it was using Jetpack 6. I don't know why that is the case, but for now it will be okay.  To answer your question, nvgstcapture-1.0 did not produce anything despite running "ls /dev/video*" giving me /dev/video0.

 I was using a code that was given to me by a friend who is also using the Jetson Orin 

 docker run -it --runtime=nvidia --device=/dev/video0:/dev/video0 \
  -v /usr/lib/aarch64-linux-gnu/libargus_camera.so:/usr/lib/aarch64-linux-gnu/libargus_camera.so \
  main_dock

closed 2024-07-30 by Syncron19