dusty-nv/jetson-containers #579: Error encountered when compiling cuML on Jestson AGX Orin

WRangers, 2024-08-02:
Hi, there.

Thank you firstly for your `cuML` containers.  However, when I tried to compile a newer version, e.g. `cuML 24.04`, based on the container `nvcr.io/nvidia/l4t-cuda:11.4.19-devel`, I encountered an error when compiling the `raft`, which should not be an error --- a variable was declared but never used. I also tried to modify the `CMakeLists.txt`, but failed.

Later, I gave up compiling from source code, installing `cuML 24.04` via conda according to https://docs.rapids.ai/install via
`conda create -n rapids-24.04 -c rapidsai -c conda-forge -c nvidia  cuml=24.04 python=3.9 'cuda-version==11.4'`. Though the installation was successful, the excution of CPP DBSCAN example failed (verbose output is listed in the end).

My physical device information:

-  CUDA Arch BIN: 8.7
-  L4T 35.4.1
-  Jetpack: 5.1.2
-  Based container: `nvcr.io/nvidia/l4t-cuda:11.4.19-devel` from https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-cuda/tags

I wonder how you built `cuML` successfully before. Could you please give me some advice? Thank you very much!

[error.txt](https://github.com/user-attachments/files/16471928/error.txt)

closed 2025-02-21 by johnnynunez

Promethe-us, 2025-04-24:
Hello, have you solved the problem I also want to install cuml on jetson orin, how do you solve this.

WRangers, 2025-04-24:
I didn‘t solve the issue. So I used the compiled container with an older version of `cuml`.