dusty-nv/jetson-containers #585: Ollama pull error: The model you are attempting to pull requires a newer version of Ollama.Hi

iguelkanat, 2024-08-08:
Hi,

I just installed the containers and pulled the ollama docker.. so far it runs, but it cannot load the new models, as it has the version 0.1.46-0-gbc42e60 but it requires at least the version 0.3x to be able to load the new models.

I get the following error message, when I try to load llama3.1:

root@ubuntu:/# ollama pull llama3.1
pulling manifest 
Error: pull model manifest: 412: 

The model you are attempting to pull requires a newer version of Ollama.

Please download the latest version at:

	https://ollama.com/download

How can I update the ollama within the container to a newer version?

RPHllc, 2024-08-08:
I am using the dustynv/ollama:r36.3.0 and it works for me.
From the command prompt you can "docker images" to see the images you have. Then remove the one you have and use "docker pull dustynv/ollama:rxx.x.x" to get the new image for your system. The options for xx.x.x are 35.4.1, 36.2.0, 36.3.0

iguelkanat, 2024-08-08:
it worked.. thanks

RPHllc, 2024-08-08:
I have seen this error when trying llama3.1 8b instruct. The basic llama3.1 works for me.
This error seems to be a problem with the ollama source, see https://github.com/ollama/ollama/issues/6048, which was recently fixed. I believe we need to wait for dustynv to rebuild the container.

dusty-nv, 2024-08-08:
@RPHllc @iguelkanat just kicked this off, it is building on my Jetson running L4T R36.2, so if all goes well there will be an updated ollama image pushed to `dustynv/ollama:r36.2.0`

> ollama -v gives "ollama version is 0.0.0" for me as well.

This issue was previously addressed by commit https://github.com/dusty-nv/jetson-containers/commit/6536834fae6b5a560ca68bb502e82a5204a83c7d , however perhaps the container image you are running was built prior, and this new one building now will include the versioning patch.

iguelkanat, 2024-08-08:
@dusty-nv : Thanks... it works now and I get "ollama version is 7d1c004-dirty" as version.

dusty-nv, 2024-08-08:
ok yes!, looks like it finished uploading `dustynv/ollama:r36.2.0`  (which should work fine on r36.3)

I'm not sure about it reporting the actual version number, IIRC when building from git it extracts the latest commit SHA and uses that for the "version"

RPHllc, 2024-08-08:
Perfect! Thank you.

RPHllc, 2024-08-08:
We are not there yet. 
The new version of Ollama seems to verify model compatibility against the version. 

When I attempted to pull new images I got:

root@jetson:/# ollama pull llama3.1:8b-instruct-q4_0
pulling manifest
Error: pull model manifest: 412:

The model you are attempting to pull requires a newer version of Ollama.

Please download the latest version at:

	https://ollama.com/download
	
Because I have two Jetsons and I had downloaded this model earlier to one of them, I know the new dustynv/ollama:r36.2.0 fixed the issue and works with the model. The issue may be just around the version number.

RPHllc, 2024-08-08:
Workaround to be able to download new models until this is resolved. 

1) docker pull dustynv/ollama:r36.3.0
2) jetson-containers run -d --name ollama $(autotag ollama:r36.3.0)
3) docker exec -it ollama bash
4) # ollama pull <any desired model>
5) # exit
6) docker stop ollama
7) jetson-containers run -d --name ollama $(autotag ollama)

(This works because we have two images, the older r36.3.0 that is able to download all models but do not run the llama3.1 type models. and the new build r36.2.0 that runs the new models but is not downloading them)

dusty-nv, 2024-08-09:
@iguelkanat oh sure yes, you can rebuild the ollama container with the latest anytime like this:

```bash
jetson-containers build ollama
```

Just follow these steps first to get your docker-root on NVME so that you don't run out of storage:
https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

dusty-nv, 2024-08-09:
> Because I have two Jetsons and I had downloaded this llm model earlier to one of them, I know the new dustynv/ollama:r36.2.0 fixed the issue that prevented the llm model to run. But I am unable to download the same model to my second Jetson.

@RPHllc is this because your other Jetson is still running the previous r36.3 image?   In that case, run it with `dustynv/ollama:r36.2.0` instead of `$(autotag ollama)`  (r36.3 and r36.2 are compatible with each other)

My r36.3 machine is still occupied doing a training run, but after will rebuild it with r36.3 for consistency with the tags.

Or are you referring to a different issue, and the ollama version still isn't being set correctly in the container?

RPHllc, 2024-08-09:
@dusty-nv yes, it is a different issue.

The r36.2 version you previously had and the r36.3 version you currently have are able to load llama3.1, llama3.1:8b-instruct-q4_0, etc. In other words, I was able to execute "ollama pull lama3.1" from inside the container. But those container versions are not able to run llama3.1:8b-instruct-q4_0.

Your new r36.2 version is able to run the previously downloaded llm images. The new version fixed the "not able to run" problem. However, the new version is not able to download llama3.1 and possibly not other images.

wilbert-vb, 2024-08-12:
> @iguelkanat oh sure yes, you can rebuild the ollama container with the latest anytime like this:
> 
> ```shell
> jetson-containers build ollama
> ```
> 
> Just follow these steps first to get your docker-root on NVME so that you don't run out of storage: https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

After the command Jetson-containers build ollama i ended up with the following:

[image]

How do I get a single image with the latest version?

xiaohei2022, 2024-08-16:
> @RPHllc @iguelkanat just kicked this off, it is building on my Jetson running L4T R36.2, so if all goes well there will be an updated ollama image pushed to `dustynv/ollama:r36.2.0`
> 
> > ollama -v gives "ollama version is 0.0.0" for me as well.
> 
> This issue was previously addressed by commit [6536834](https://github.com/dusty-nv/jetson-containers/commit/6536834fae6b5a560ca68bb502e82a5204a83c7d) , however perhaps the container image you are running was built prior, and this new one building now will include the versioning patch.

hello,how to do (ollama -v gives "ollama version is 0.0.0"), thank you for your time

ivo-toby, 2024-08-24:
I'm experiencing same issue. 

```
➜ ollama run llama3.1:latest
pulling manifest
Error: pull model manifest: 412:

The model you are attempting to pull requires a newer version of Ollama.
```

```
➜ ollama -v
ollama version is 0.1.42
Warning: client version is 0.3.6
```

Ollama installed using homebrew.

ballerburg9005, 2024-08-24:
I updated ollama to git version from AUR but still get the error
```
ollama pull  llama3.1:8b-text-fp16 
```

dusty-nv, 2024-08-25:
Is the issue that the builds from source use the git tag as the version like upstream did, whereas in the actual ollama releases they assign a specific version number?

ballerburg9005, 2024-08-25:
I have no idea.

```
ollama version is 0.3.7.g0f92b19b
# ollama pull  llama3.1:8b-text-fp16

pulling manifest 
Error: pull model manifest: 412: 

The model you are attempting to pull requires a newer version of Ollama.

Please download the latest version at:

	https://ollama.com/download
```

ballerburg9005, 2024-08-25:
Ah yes, I changed 0.3.7.g234234234 to 0.3.7 and now it works. Thanks.

iguelkanat, 2024-08-28:
Hi, does anyone has a tutorial on how to build Ollama from source on a Jetson Oring AGX.. I could not find instructions on how to build it.
Thanks

dusty-nv, 2024-08-28:
@iguelkanat `jetson-containers build ollama`

iguelkanat, 2024-08-28:
@dusty-nv  I tried twice and each time it generated successfully but I could not load new models such as llama3.1 with the same error message as I stated at top of this post.

dusty-nv, 2024-08-28:
Hi @iguelkanat , @ballerburg9005 said above they got it to work after changing the version - https://github.com/dusty-nv/jetson-containers/issues/585#issuecomment-2308882121

presumably they truncated the commit tag `g234234234` off the version here in the Dockerfile:  https://github.com/dusty-nv/jetson-containers/blob/0d9e9a4d7f70841e03ed6fa9e71869bd7888b78b/packages/llm/ollama/Dockerfile#L36

I am happy to accept PR's should someone confirm the modifications.  You can also run llama-3.1 through llama.cpp easily.  Happy to host the builds and container images for ollama, but I have maintained since the beginning of it that there's a limit to which I will personally dive into that project being written in Go and essentially just a complicated wrapper around llama.cpp.  Case in point:  llama.cpp can easily run llama-3.1 GGUF since day 1, but you are stuck messing with ollama versions telling you it can't run.

dusty-nv, 2024-09-01:
See here for suggestion from @mtebenev to revert to just using version `0.0.0` to avoid these checks (https://github.com/dusty-nv/jetson-containers/issues/592#issuecomment-2323177164).  It was previously requested to have the correct ollama version reported in the container when `ollama --version` is run, but then this llama 3.1 issue was encountered (and presumably others like it in the future)

For now, I am just rebuilding against ollama 3.9 since that's out now and will push it to `dustynv/ollama:r36.3.0` - that should avoid the llama 3.1 issue

dusty-nv, 2024-09-02:
FYI it ended up being `dustynv/ollama:r36.2.0` that was pushed with ollama 3.9

dusty-nv, 2024-09-12:
@CappyT it is here to change it to 0.0.0 and then rebuild with `jetson-containers build ollama`  https://github.com/dusty-nv/jetson-containers/blob/c51be568562c8f00cbf62091188ce332571413b7/packages/llm/ollama/Dockerfile#L37

rayrrr, 2024-09-25:
Hi all. While I was able to pull the `llama3.1` model using the approached described in this issue, I was still landing on a "core dump" with the same error (the core dump log seems to match this [bug in `llama.cpp`](https://github.com/Mozilla-Ocho/llamafile/issues/516)). That `llama.cpp` issue says it was fixed. So it occurred to me: perhaps what I need to be doing here is not upgrading `ollama` within `jetson-containers`, but upgrading `jetson-containers` itself. I did a `git pull` in the root `jetson-containers` directory, then rebuilt `ollama` with:

```
CUDA_VERSION=12.2 jetson-containers build ollama
```

Go have a glass of water. Then when that's done,

```
CUDA_VERSION=12.2 jetson-containers run $(autotag ollama)
```

..and I am now running `llama3.1` successfully after several nights of wrangling, and can report that it seems to be the quantum leap improvement we've all been expecting.

rzo1, 2024-09-30:
@rayrrr Thanks for the tip. Didn't work for me though, because the re-build of ollama just fails with

```bash
[100%] Linking CXX executable ../bin/ollama_llama_server
/usr/bin/ld: cannot find -lcuda: No such file or directory
collect2: error: ld returned 1 exit status
gmake[3]: *** [ext_server/CMakeFiles/ollama_llama_server.dir/build.make:104: bin/ollama_llama_server] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:3480: ext_server/CMakeFiles/ollama_llama_server.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:3487: ext_server/CMakeFiles/ollama_llama_server.dir/rule] Error 2
gmake: *** [Makefile:1388: ollama_llama_server] Error 2
```

rayrrr, 2024-10-02:
@rzo1 the hash I built from: 94422bc9708784ae42a85af7c2f22cd7ba400900

CappyT, 2024-10-07:
> @rayrrr Thanks for the tip. Didn't work for me though, because the re-build of ollama just fails with
> 
> ```shell
> [100%] Linking CXX executable ../bin/ollama_llama_server
> /usr/bin/ld: cannot find -lcuda: No such file or directory
> collect2: error: ld returned 1 exit status
> gmake[3]: *** [ext_server/CMakeFiles/ollama_llama_server.dir/build.make:104: bin/ollama_llama_server] Error 1
> gmake[2]: *** [CMakeFiles/Makefile2:3480: ext_server/CMakeFiles/ollama_llama_server.dir/all] Error 2
> gmake[1]: *** [CMakeFiles/Makefile2:3487: ext_server/CMakeFiles/ollama_llama_server.dir/rule] Error 2
> gmake: *** [Makefile:1388: ollama_llama_server] Error 2
> ```
> 
> Can you maybe share the git hash you did your build from?

I have the same error, tried also with the hash provided.

closed 2025-02-21 by johnnynunez