dusty-nv/jetson-containers #586: Cannot build stable-diffusion-webui

ToeiRei, 2024-08-11:
I tried to run 'jetson-containers build stable-diffusion-webui' and it errors out stating that it cannot access the GPU.

```
RuntimeError: Torch is not able to use GPU; add --skip-torch-cuda-test to COMMANDLINE_ARGS variable to disable this check

```

any ideas?

```
Software part of jetson-stats 4.2.9 - (c) 2024, Raffaello Bonghi
Model: NVIDIA Jetson Orin NX Engineering Reference Developer Kit - Jetpack 6.0 [L4T 36.3.0]
NV Power Mode[0]: MAXN
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - P-Number: p3767-0000
 - Module: NVIDIA Jetson Orin NX (16GB ram)
Platform:
 - Distribution: Ubuntu 22.04 Jammy Jellyfish
 - Release: 5.15.136-tegra
jtop:
 - Version: 4.2.9
 - Service: Active
Libraries:
 - CUDA: 12.2.140
 - cuDNN: 8.9.4.25
 - TensorRT: 8.6.2.3
 - VPI: 3.1.5
 - Vulkan: 1.3.204
 - OpenCV: 4.9.0 - with CUDA: YES
```

tokk-nv, 2024-08-29:
Can you check the contents of `/etc/docker/daemon.json`?

```
cat /etc/docker/daemon.json
```

ToeiRei, 2024-08-30:
There you go.

```json
{
    "runtimes": {
        "nvidia": {
            "args": [],
            "path": "nvidia-container-runtime"
        }
    },
    "default-runtime": "nvidia"
}
```

In my opinion it may be a docker problem in the end. Gonna investigate and probably just ditch my current jetson install and build my own rootfs

closed 2024-08-30 by ToeiRei