dusty-nv/jetson-containers #591: [Bug] Unable to build opencv container with the default opencv tag `4.8.1`

hrishi121, 2024-08-16:
Hello,

I have been trying to build the `NanoLLM` container for following system config:
```
Jetson Orin Nano
-- L4T_VERSION=36.3.0
-- JETPACK_VERSION=6.0
-- CUDA_VERSION=12.4
-- PYTHON_VERSION=3.10
-- LSB_RELEASE=22.04 (jammy)
```
One of the dependencies for the `NanoLLM` container is the `opencv` container.  When building the `opencv` container, I noticed that `install.sh` would fail and it would proceed to build the `opencv-python` repo using the `build.sh` script. However, compiling `opencv-python` from scratch with the `build.sh` would keep failing as well (Unfortunately, I forgot to capture the build errors)

I commented out the following lines in `build.sh`, and was able to build the opencv container
```
#git checkout --recurse-submodules ${OPENCV_VERSION}
#cat modules/core/include/opencv2/core/version.hpp

#cd ../opencv_contrib
#git checkout --recurse-submodules ${OPENCV_VERSION}

#cd ../opencv_extra
#git checkout --recurse-submodules ${OPENCV_VERSION}
```

Commenting out these lines disables the build from using the `4.8.1` submodules for `opencv, opencv_contrib, opencv_extras` and rather use the new default `4.10.0` submodules in the `4.x` branch

I am not totally sure why the build was failing earlier, but my guesses are:
- Incompatibility of CUDA 12.4 with the 4.8.1 submodules
- [OR] Incompatibility of the latest opencv-python branch 4.x with 4.8.1 submodules

Just wanted to give a heads-up to anyone who is building any container that has the `opencv` as a dependency and using CUDA 12.4

hrishi121, 2024-08-16:
@dusty-nv Is there a specific reason for using `4.8.1` as the default opencv version in `config.py` ? Do you think it would be a good idea to update it to `4.10.0` ?

johnnynunez, 2024-08-16:
> @dusty-nv Is there a specific reason for using `4.8.1` as the default opencv version in `config.py` ? Do you think it would be a good idea to update it to `4.10.0` ?

FYI I'm using opencv 4.10 with cuda 12.6 cudnn last version without problems

closed 2025-02-21 by johnnynunez