dusty-nv/jetson-containers #595: Put Flux inside jetson container

martincerven, 2024-08-20:
Put content from https://www.jetson-ai-lab.com/tutorial_comfyui_flux.html

into jetson containers

johnnynunez, 2024-08-20:
I will work on that

johnnynunez, 2024-08-20:
training: https://github.com/dusty-nv/jetson-containers/pull/594

dusty-nv, 2024-08-20:
Johhny is on it  😀   thanks Johhny!

johnnynunez, 2024-08-21:
> Put content from https://www.jetson-ai-lab.com/tutorial_comfyui_flux.html
> 
> into jetson containers

If you want testing: https://github.com/dusty-nv/jetson-containers/pull/594

johnnynunez, 2024-08-21:
```
CUDA_VERSION=12.2 jetson-containers build comfyui
sudo jetson-containers run -it comfyui:r36.3.0-cu122
```

devrosx, 2024-08-28:
hello great :) its working on jetson xavier AGX ? tried to build manually but i have still this error module ‘torch’ has no attribute ‘float8_e4m3fn’

johnnynunez, 2024-08-28:
> hello great :) its working on jetson xavier AGX ? tried to build manually but i have still this error module ‘torch’ has no attribute ‘float8_e4m3fn’

I don't know. ComfyUI is very recently and works with latest version of pytorch.
But this error is more on the pytorch side:
https://github.com/comfyanonymous/ComfyUI/issues/4297

devrosx, 2024-08-29:
on AGX Xavier is latest CUDA 11.8 and torch-2.1.0a0+41361538.nv23.06-cp38-cp38-linux_aarch64.whl
I would probably need a newer version to get rid of this error.