dusty-nv/jetson-containers #600: Cannot install CUDA 12.1

dilerbatu, 2024-08-23:
Hi All,

I want to install CUDA12.1 with my containers but it only installs cuda 11.4 each time when I write cuda:12.2

jetson-containers build --build-flags=--no-cache --name=cu122 cuda:12.2

johnnynunez, 2024-08-23:
> Hi All,
> 
> I want to install CUDA12.1 with my containers but it only installs cuda 11.4 each time when I write cuda:12.2
> 
> jetson-containers build --build-flags=--no-cache --name=cu122 cuda:12.2
> 
> EDIT: Now I can not install anything, each command gives me:
> 
> ` python3.8 Looking in indexes: http://jetson.webredirect.org/jp5/cu122 ERROR: Could not find a version that satisfies the requirement pip (from versions: none) ERROR: No matching distribution found for pip curl -sS https://bootstrap.pypa.io/pip/3.6/get-pip.py python3.6 /tmp/install_python.sh: line 27: python3.6: command not found curl: (23) Failed writing body (1338 != 1379)
> 
> `

did you try with:
CUDA_VERSION=12.2 jetson-containers build cuda

dilerbatu, 2024-08-23:
python issue just solved but it still installs cuda 11.4. I built the image with the command that u suggest. I run it but when I type nvcc --version, it gives 11.4.

dusty-nv, 2024-08-23:
What JetPack are you on?  CUDA 12.2 is not on JetPack 5

dilerbatu, 2024-08-23:
It explains everything, I guess. I am using Jetpack 5. Do I need to upgrade in order to use CUDA 12 ?

dilerbatu, 2024-08-23:
How will I install ros noetic If I use Jetpack 6 ? I kinda know that jetpack 6 works on ubuntu 22.04

johnnynunez, 2024-09-03:
> It explains everything, I guess. I am using Jetpack 5. Do I need to upgrade in order to use CUDA 12 ? I am using Orin.

Yess, jetpack 6 is necessary

closed 2024-09-04 by dilerbatu