dusty-nv/jetson-containers #601: Not able to run Live LLaVA

Fibo27, 2024-08-26:
@dusty-nv I am trying to follow the tutorial of Live LLaVA on my Jetson-NX platform with Jetpack 6.0. After downloading Jetson container and ensuring that docker-runtime is set to nvidia, I used the following command 

jetson-containers run $(autotag nano_llm) \
  python3 -m nano_llm.agents.video_query --api=mlc \
    --model Efficient-Large-Model/VILA1.5-3b \
    --max-context-len 256 \
    --max-new-tokens 32 \
    --video-input /dev/video0 \
    --video-output webrtc://@:8554/output

however, after pulling the image, i get the following error
"ImportError: libnvdla_compiler.so: cannot open shared object file: No such file or directory" I have looked at possible solutions to overcome this - the most likely being setting the runtime correctly.
Can you please suggest possible options?

tokk-nv, 2024-08-26:
It seems that it requires the NVDLA packages installed on the host side.
Will you try installing the package on the host and try running `nano_llm` container again?

```
sudo apt install nvidia-l4t-dla-compiler
jetson-containers run $(autotag nano_llm) \
  python3 -m nano_llm.agents.video_query --api=mlc \
    --model Efficient-Large-Model/VILA1.5-3b \
    --max-context-len 256 \
    --max-new-tokens 32 \
    --video-input /dev/video0 \
    --video-output webrtc://@:8554/output
```

Fibo27, 2024-08-27:
Hi that solved the issue. Now when I try to run, i am confronted with this issue "gstreamer no element nvvidconv" - this seems to be a well known issue as well - i tried the solution suggested here https://forums.developer.nvidia.com/t/jetson-tx2i-erroneous-pipeline-no-element-nvv4l2h264enc/116957, but that did not work. Any idea?

tokk-nv, 2024-08-27:
That would be installing another package, `nvidia-l4t-gstreamer`.
And you want to connect a USB webcam to Jetson.

Will you try the following?

```
sudo apt install nvidia-l4t-gstreamer
lsusb
ls /dev/video*
jetson-containers run $(autotag nano_llm) \
  python3 -m nano_llm.agents.video_query --api=mlc \
    --model Efficient-Large-Model/VILA1.5-3b \
    --max-context-len 256 \
    --max-new-tokens 32 \
    --video-input /dev/video0 \
    --video-output webrtc://@:8554/output
```

Fibo27, 2024-08-27:
ok cool that works! Thanks

Fyi, before i did `sudo apt install nvidia-l4t-gstreamer` which i presume installs this system wide, I 

1) Ran the jetson-inference container and could run `video-viewer /dev/video0` inside that container - my understanding is that it uses jetson-utils for the video-viewer program which is based on gstreamer - so not sure why this ran but the nano_llm did not?
2) Could see that gstreamer 1.0 and gstreamer1.0 existed in /usr/lib/aarch64-linux-gnu/ - so i had guessed that gstreamer was installed already.

So my follow-up is: isnt nvidia-l4t-gstreamer installed by default in Jetpack 6.0?

Fibo27, 2024-08-27:
I am closing this presuming that this is not the thread for my last query

closed 2024-08-27 by Fibo27

tokk-nv, 2024-08-27:
Hi Fibo27,

For question 1, you may want to pull the latest container image and try again.
```
docker pull dustynv/nano_llm:main-r36.3.0
```

For question 2, `nvidia-l4t-gstreamer` is not part of Jetson Linux (L4T), but is a part of JetPack components (or Jetson SDK components, installed by SDK Manager). 
After setting up your Jetson Orin NX, you can check what is installed with a command like this.

```bash
dpkg-query -Wf '${Installed-Size;8} KiB \t${Package;-30}\t${binary:Summary}\n' | grep nvidia-l4t
```