dusty-nv/jetson-containers #605: Image

nmq45698, 2024-08-30:


nmq45698, 2024-08-30:
HI there!
Currently i'm trying to deploy some small models (say gpt-2) with smoothquant on Jetson-AGX Orin. 
However, there is no results upon searching the image "tensorrt_llm:35.2.1" on dockerhub.
I've noticed the your comments on #564 that the trtllm will be available soon. 
So, are there update for docker image to support the "docker run" about tensorrt_llm on Orin?
thx  :P

dusty-nv, 2024-08-30:
@nmq45698 still coming soon but getting closer!   for now I would just run SmoothQuant inference through PyTorch like their repo shows, or use AWQ TinyChat, or MLC/TVM.

johnnynunez, 2024-09-05:
> @nmq45698 still coming soon but getting closer! for now I would just run SmoothQuant inference through PyTorch like their repo shows, or use AWQ TinyChat, or MLC/TVM.

I've updated to the last version of tensort_llm https://github.com/NVIDIA/TensorRT-LLM/releases/tag/v0.12.0

nmq45698, 2024-09-06:
> > @nmq45698 still coming soon but getting closer! for now I would just run SmoothQuant inference through PyTorch like their repo shows, or use AWQ TinyChat, or MLC/TVM.
> 
> I've updated to the last version of tensort_llm https://github.com/NVIDIA/TensorRT-LLM/releases/tag/v0.12.0

Thanks for your remind! :P
Actually, I've noticed such repo about TensorRT-LLM, with engines successfully built on GPU with amd64. 

While for Jetson AGX Orin, I've tried under some docker images pulled from @dusty-nv dustynv/... r35.x.x, which seems not supportive for TensorRT 10.3 due to the GLIBC under Ubuntu 20.04. The kINT64, kBF16 in TensorRT-LLM could not be compiled.

Then, I tried on docker images pulled from dustynv/... r36.x.x (say nano_llm) based on Ubuntu 22.04, the TensorRT 10.3 was successfully installed.  
However, another error about libnvinfer.so occurred when running /scripts/build_wheels.py, as well as import tensorrt in python!
`(Pdb) import tensorrt
*** ImportError: /lib/aarch64-linux-gnu/libnvinfer.so.10: undefined symbol: _ZN5nvdla8IProfile36setCanGenerateDetailedLayerwiseStatsEb
`

Have you ever encountered that upon building TensorRT-LLM on Orin? ^v^

johnnynunez, 2024-09-06:
> Thanks for your remind! :P Actually, I've noticed such repo about TensorRT-LLM, with engines successfully built on GPU with amd64.
>
> While for Jetson AGX Orin, I've tried under some docker images [...]

Do you use jetpack 5?

Hrishikeshh-nd, 2024-11-14:
> > @nmq45698 still coming soon but getting closer! for now I would just run SmoothQuant inference through PyTorch like their repo shows, or use AWQ TinyChat, or MLC/TVM.
> 
> I've updated to the last version of tensort_llm https://github.com/NVIDIA/TensorRT-LLM/releases/tag/v0.12.0

@johnnynunez Does this release support Jetson Orin or AGX? The official [documentation page](https://nvidia.github.io/TensorRT-LLM/reference/support-matrix.html) still states that there's only support for x86_64 architecture at the moment, and nothing about aarch64.

johnnynunez, 2024-11-14:
> @johnnynunez Does this release support Jetson Orin or AGX? The official [documentation page](https://nvidia.github.io/TensorRT-LLM/reference/support-matrix.html) still states that there's only [...]

https://www.jetson-ai-lab.com/tensorrt_llm.html

dusty-nv, 2024-11-15:
> @dusty-nv in the first example, change the name to  dustynv/tensorrt_llm:0.12-r36.4.0

Ahh thanks @johnnynunez, just fixed that in https://github.com/NVIDIA-AI-IOT/jetson-generative-ai-playground/pull/229

@Hrishikeshh-nd the Orin support is in this TRT-LLM branch:  https://github.com/NVIDIA/TensorRT-LLM/tree/v0.12.0-jetson

shahizat, 2024-11-15:
Hi @dusty-nv, I successfully run the openai_server.py with endpoint exposed, but can not send the request, always shows "`port 8000 after 0 ms: Connection refused`"

Hrishikeshh-nd, 2024-11-15:
> > @dusty-nv in the first example, change the name to  dustynv/tensorrt_llm:0.12-r36.4.0
> 
> Ahh thanks @johnnynunez, just fixed that in [NVIDIA-AI-IOT/jetson-generative-ai-playground#229](https://github.com/NVIDIA-AI-IOT/jetson-generative-ai-playground/pull/229)
> 
> @Hrishikeshh-nd the Orin support is in this TRT-LLM branch: https://github.com/NVIDIA/TensorRT-LLM/tree/v0.12.0-jetson

Thank you @dusty-nv @johnnynunez !

doruksonmez, 2024-12-05:
> hi @dusty-nv, @johnnynunez, If I am not mistaken, we also need to add --tokenizer path to the openai_server.py command. See this link: [NVIDIA/TensorRT-LLM#2357](https://github.com/NVIDIA/TensorRT-LLM/issues/2357)

I'm not sure if this is the correct way to point out the tokenizer but dropping my full command here if someone is looking for an answer:
```
python3 /opt/TensorRT-LLM/examples/apps/openai_server.py /data/models/tensorrt_llm/Llama-2-7b-chat-hf-fp16/ \
--tokenizer /data/models/huggingface/models--meta-llama--Llama-2-7b-chat-hf/snapshots/f5db02db724555f92da89c216ac04704f23d4590/
```

It throws an error similar to `[TRT-LLM] [E] Failed to load tokenizer from /tmp/tmpqxzcy4kyllm-workspace/tmp.engine` if the `--tokenizer` flag is not specified.

xiaoyuzju, 2025-01-21:
Could you please add a container for the latest version of TensorRT-LLM? Thank you.

closed 2025-02-21 by johnnynunez