dusty-nv/jetson-containers #613: Stable Diffusion 404 error on Hugging Face

Yggdrasill501, 2024-09-03:
So when you are trying to install and run the Stable Diffusion v1.5 from https://huggingface.co/runwayml/stable-diffusion-v1-5, based on documentation in https://github.com/dusty-nv/jetson-containers/tree/master/packages/diffusion/stable-diffusion-webui#readme
[image]

Yggdrasill501, 2024-09-03:
I suggest to replacing it with https://huggingface.co/benjamin-paine/stable-diffusion-v1-5

Yggdrasill501, 2024-09-03:
Error message that was related to this :
FileNotFoundError: No checkpoints found. When searching for checkpoints, looked at: - file /opt/stable-diffusion-webui/model.ckpt - directory / data/model/stable diffusion/models/Stable-diffusioncan't run without a checkpoint. Find and place a ckpt or safetensors file into any of those
locations.

johnnynunez, 2024-09-03:
Stable diffusion 1.5 was removed from hugging face. I don't know why, but it disappear. Anyways, Now you can use 2.0 or 3.0

Yggdrasill501, 2024-09-03:
A possible solution is to replace it with this https://huggingface.co/Comfy-Org/stable-diffusion-v1-5-archive which is the archived version be making a pull request later today if needed

Yggdrasill501, 2024-09-03:
Pull request at: https://github.com/dusty-nv/jetson-containers/pull/614

closed 2025-02-21 by johnnynunez

Yggdrasill501, 2025-02-22:
Thank you verry much