dusty-nv/jetson-containers #622: Dockerfile for nanl_llm and Ros Iron

Fibo27, 2024-09-06:
@dusty-nv 
You have a container dustynv/nano_llm:humble-36.3.0 in the hub which has Ros version = Humble baked in. I want to create one with Ros version =Iron. Is there a Dockerfile that i can use? Else, what should be the base container that I should use?

dusty-nv, 2024-09-06:
@Fibo27 try `jetson-containers build nano_llm:iron` - there are combinations defined for the ROS2 distros + nano_llm here:

https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/nano_llm/config.py

Fibo27, 2024-09-06:
Thanks - I did that but i get this error
`File "/home/jetson-containers/jetson_containers/packages.py", line 457, in check_requirements
    if Version(f"{CUDA_VERSION.major}{CUDA_VERSION.minor}") not in SpecifierSet(r.replace('cu','')):
AttributeError: 'str' object has no attribute 'major'`

dusty-nv, 2024-09-07:
Are you able to build other containers from jetson-containers, or is it just that one?  Are you setting CUDA_VERSION in your environment, or not have it installed under `/usr/local` ?   Or if you don't have it installed, then run it like:  `CUDA_VERSION=12.2 jetson-containers build nano_llm:iron`

Fibo27, 2024-09-07:
Thanks that works and I am building the container now.

1) To run the container, do i simply use this command "jetson-containers run" followed by the image name or should I modify this script "https://github.com/dusty-nv/jetson-containers/blob/master/run.sh"? This run script takes care of all the volumes that need to be mounted for the docker run and I presume those will remain the same.
2) To modify the container to add features from ROS, should I follow the script in this Dockerfile https://github.com/dusty-nv/jetbot_ros/blob/d8e5ee1b17f5c66d038017b86f8920a496197ea9/Dockerfile#L72? In the past you have directed me to this one. I would presume that I can use the newly created image as the base and the modify the same for additional ROS features.

dusty-nv, 2024-09-07:
1. Yes, after building you can do `jetson-containers run nano_llm:iron-r36.3.0` , it should automatically source the ROS2 workspaces upon container start-up

2.  Yes, and if there are no additional ROS dependencies, this Dockerfile.ros also does a simpler version that doesn't need rosdep/rosinstall_generator:  https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/nano_llm/Dockerfile.ros

Fibo27, 2024-09-08:
@dusty-nv 
This was an interesting (long) build on a Jetson Orin NX -  there were many intermediate images that got built - eventually the default image name was nano_llm:iron-r36.3.0-cu122 so one has to use `jetson-containers run nano_llm:iron-r36.3.0-cu122 `.

I had to prune the intermediate images as it would take up a lot of space (in the process i had to upgrade to a larger ssd storage device)
Thanks

closed 2024-09-11 by Fibo27