dusty-nv/jetson-containers #629: llama.cpp container error

PredyDaddy, 2024-09-12:
Hello dusty,
   I am using `dustynv/llama_cpp:r35.4.1` images with `JetPack 5.1.2`, then I use follow command to start container
```bash
docker run -it --shm-size=64g --privileged --name llama_cpp --network="host" -v $(pwd):/app dustynv/llama_cpp:r35.4.1
```

then I use `benchmark.py` from this repo, I try to use follow command to infer qwen2
```bash
python3 benchmark.py --model /app/vlm_weights/Qwen2-0___5B-Instruct-GGUF/qwen2-0_5b-instruct-fp16.gguf \            --prompt "Once upon a time," \            --n-predict 128 --ctx-size 192 --batch-size 192 \            --n-gpu-layers 999 --threads $(nproc)
```

then I have follow error
```bash
root@ubuntu:/app/inside_container/jetson-containers/packages/llm/llama_cpp# python3 benchmark.py --model /app/vlm_weights/Qwen2-0___5B-Instruct-GGUF/qwen2-0_5b-instruct-fp16.gguf \            --prompt "Once upon a time," \            --n-predict 128 --ctx-size 192 --batch-size 192 \            --n-gpu-layers 999 --threads $(nproc)
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/llama_cpp/llama_cpp.py", line 75, in _load_shared_library
    return ctypes.CDLL(str(_lib_path), **cdll_args)  # type: ignore
  File "/usr/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1: file too short

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "benchmark.py", line 11, in <module>
    from llama_cpp import Llama
  File "/usr/local/lib/python3.8/dist-packages/llama_cpp/__init__.py", line 1, in <module>
    from .llama_cpp import *
  File "/usr/local/lib/python3.8/dist-packages/llama_cpp/llama_cpp.py", line 88, in <module>
    _lib = _load_shared_library(_lib_base_name)
  File "/usr/local/lib/python3.8/dist-packages/llama_cpp/llama_cpp.py", line 77, in _load_shared_library
    raise RuntimeError(f"Failed to load shared library '{_lib_path}': {e}")
RuntimeError: Failed to load shared library '/usr/local/lib/python3.8/dist-packages/llama_cpp/lib/libllama.so': /usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1: file too short
```
Could you help me look at it?

Thanks!

dusty-nv, 2024-09-13:
Hi @PredyDaddy, this error seems to happen when your nvidia container runtime is not properly mounting the CUDA drivers into docker.  Are you able to run other containers, like pytorch and `import torch` or `import tensorrt` ?  What VLM are you trying btw that is using qwen2-0.5?  Looks cool 👍

PredyDaddy, 2024-09-14:
> Hi @PredyDaddy, this error seems to happen when your nvidia container runtime is not properly mounting the CUDA drivers into docker. Are you able to run other containers, like pytorch and `import torch` or `import tensorrt` ? What VLM are you trying btw that is using qwen2-0.5? Looks cool 👍

Thanks for reply, previously I also have trouble when I using the ollama container `dustynv/ollama:r36.2.0`, I can infer the model, but not use GPU. 

I add `--runtime=nvidia` when I docker run ollama container and llama.cpp container, trouble solved. 

I am not using the VLM version of qwen2, I am using LLM version of qwen2. I really hope I can run qwen2vl in jetson device. Then I check issue of llama.cpp and ollama, many people request feature add qwen2vl, so that I just need to wait this two framework add feature that I can run qwen2vl.

closed 2024-09-14 by PredyDaddy