dusty-nv/jetson-containers #630: Why do jp6 containers include l4t-r36.2.0 as opposed to l4t-r36.3.0?

juliangaal, 2024-09-12:
First, thanks for maintaining this extremely valuable repo!

According to the [release page](https://developer.nvidia.com/embedded/jetpack-sdk-60)

> JetPack 6.0 includes Jetson Linux 36.3

while your ros:humble images for jp6 are only available for `l4t-r36.2.0`. 

I can't seem to find where this specific version of l4t is set in your [gh actions file](https://github.com/dusty-nv/jetson-containers/blob/master/.github/workflows/ros-humble-ros-base_jp60.yml) or your python helper files. There only seem to be `get_jetpack()` functions, but none with `set_jetpack()` functionality.

My issue boils down to two questions:

a) where is the `l4t` version set in your build pipeline?
b) can I help you with regards to extending your repo with `l4t-36.3.0` support?

Cheers

dusty-nv, 2024-09-12:
Hi, r36.2 containers are compatible with r36.3, so if I started building that container during 36.2, I keep it on 36.2 so there aren't extra tags floating around confusing at which is the latest.  New containers I start on 36.3.  It will tag it as what L4T you build it on, your you can override by setting L4T_VERSION environment variable.

juliangaal, 2024-09-12:
ah, perfect, thank you!

closed 2024-09-12 by juliangaal