dusty-nv/jetson-containers #647: ROS2 iron install returned non-zero exit status 1

aapatni, 2024-09-26:
Having some trouble building the ros iron containers (and base/desktop) on my Jetson Orin AGX (36.3). I've tracked it down to a failure right during the vcs import step of `packages/ros/ros2_build.sh`. During the vcs import step I see the error attached below. I attempted to debug the vcs import step by adding the --debug flag as well as the --retry 5 steps, but have not made any progress. 

**Command**: `CUDA_VERSION=12.2 jetson-containers build     --name=jetson-iron:latest     ros:iron-ros-base     cudnn:8.9     opencv:4.8.1     tensorrt:8.6`

```
HEAD is now at 8afbf80 Trimmed the branch to only the vision_opencv sub directory
=== src/yaml_cpp_vendor (git) ===
Cloning into '.'...
Note: switching to 'release/iron/yaml_cpp_vendor/8.1.2-3'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 1299c80 Rebase from 'upstream' (no changes)
The command '/bin/bash -c /tmp/ros2_build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/ssd/adam/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/ssd/adam/jetson-containers/jetson_containers/container.py", line 147, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag jetson-iron:latest-r36.3.0-cu122-ros_iron-ros-base --file /ssd/adam/jetson-containers/packages/ros/Dockerfile.ros2 --build-arg BASE_IMAGE=jetson-iron:latest-r36.3.0-cu122-cmake --build-arg ROS_VERSION="iron" --build-arg ROS_PACKAGE="ros_base" /ssd/adam/jetson-containers/packages/ros 2>&1 | tee /ssd/adam/jetson-containers/logs/20240926_114206/build/jetson-iron_latest-r36.3.0-cu122-ros_iron-ros-base.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

dusty-nv, 2024-09-26:
Hi @aapatni, that's strange I don't really see an error in there it just stops - which normally may occur when out on memory, but you have 32GB and that's not during compilation yet.  I've had sporadic network-related failures while it was cloning ros sources before, but it would go away after retry and just attributed it to the volume of requests happening.  One thing you can do is change this line to still save the container even when the build fails:

https://github.com/dusty-nv/jetson-containers/blob/49995ded5a5578b94addbfd3c0ec0a4f161960e8/packages/ros/Dockerfile.ros2#L24

```
RUN /tmp/ros2_build.sh || echo "BUILD FAILED"
```

Then you can run and inspect the container, and resume those commands from the build script to debug.

aapatni, 2024-10-02:
Hey @dusty-nv I was able to run the container as you proposed and resume the commands manually! Not really able to figure out what step of the pipeline is failing...
```
vcs import src < ros2.${ROS_DISTRO}.${ROS_PACKAGE}.rosinstall
SKIP_KEYS="libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv"
rosdep init
rosdep update
rosdep install -y \
	--ignore-src \
	--from-paths src \
	--rosdistro ${ROS_DISTRO} \
	--skip-keys "$SKIP_KEYS"

colcon build \
	--merge-install \
	--cmake-args -DCMAKE_BUILD_TYPE=Release
```

aapatni, 2024-10-03:
Tracked it down to 
```
=== src/ros2cli/ros2topic (git) ===
Could not clone repository 'https://github.com/ros2-gbp/ros2cli-release.git': Cloning into '.'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 7496 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

=== src/geometry2/tf2_ros (git) ===
Could not clone repository 'https://github.com/ros2-gbp/geometry2-release.git': Cloning into '.'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 7486 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
```

Now trying 
`vcs import src < ros2.${ROS_DISTRO}.${ROS_PACKAGE}.rosinstall --retry 3 --skip-existing -w 1 --repos`

aapatni, 2024-10-16:
I moved back over to my home internet and was able to get this building no problem. Wondering if maybe my office's network is throttling me or github's server's are?