dusty-nv/jetson-containers #654: cannot build containers, python_install.sh fails (Orin AGX 64GB, JetPack 6.1)

benswift, 2024-10-02:
I've got a brand new Orin AGX developer kit (64GB) and used the SDK Manager to download and flash the latest Ubuntu (22.04) and JetPack SDK (6.1).

I'm trying to build a container (I need diffusers and transformers for my application code), however when I run the build command from the README (actually a truncated one, just trying to build pytorch without transformers and ros) it bombs out in the "install python" step:

<details><pre><code>
jane@ubuntu:~/Code/jetson-containers$ jetson-containers build --name=my_container pytorch
Namespace(packages=['pytorch'], name='my_container', base='', multiple=False, build_flags='', build_args='', package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', logs='', verbose=False, no_github_api=False)
-- L4T_VERSION=36.4.0
-- JETPACK_VERSION=5.1
-- CUDA_VERSION=12.6
-- PYTHON_VERSION=3.10
-- LSB_RELEASE=22.04 (jammy)
-- Building containers  ['build-essential', 'pip_cache:cu126', 'cuda:12.6', 'cudnn', 'python', 'numpy', 'cmake', 'onnx', 'pytorch']
-- Building container my_container:r36.4.0-build-essential

DOCKER_BUILDKIT=0 docker build --network=host --tag my_container:r36.4.0-build-essential \
--file /home/jane/Code/jetson-containers/packages/build/build-essential/Dockerfile \
--build-arg BASE_IMAGE=ubuntu:22.04 \
/home/jane/Code/jetson-containers/packages/build/build-essential \
2>&1 | tee /home/jane/Code/jetson-containers/logs/20241002_212941/build/my_container_r36.4.0-build-essential.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  19.97kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> 981912c48e9a
Step 3/5 : ENV DEBIAN_FRONTEND=noninteractive     LANGUAGE=en_US:en     LANG=en_US.UTF-8     LC_ALL=en_US.UTF-8
 ---> Using cache
 ---> bee7dec2ca35
Step 4/5 : RUN set -ex     && apt-get update     && apt-get install -y --no-install-recommends         locales         locales-all         tzdata     && locale-gen en_US $LANG     && update-locale LC_ALL=$LC_ALL LANG=$LANG     && locale         && apt-get install -y --no-install-recommends         build-essential         software-properties-common         apt-transport-https         ca-certificates         lsb-release         pkg-config         gnupg         git         gdb         wget         curl         nano         zip         unzip         time 	   sshpass 	   ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*         && gcc --version     && g++ --version
 ---> Using cache
 ---> 5a61c6947b48
Step 5/5 : COPY tarpack /usr/local/bin/
 ---> Using cache
 ---> 990e4037e0a8
Successfully built 990e4037e0a8
Successfully tagged my_container:r36.4.0-build-essential
-- Building container my_container:r36.4.0-pip_cache_cu126

DOCKER_BUILDKIT=0 docker build --network=host --tag my_container:r36.4.0-pip_cache_cu126 \
--file /home/jane/Code/jetson-containers/packages/cuda/cuda/Dockerfile.pip \
--build-arg BASE_IMAGE=my_container:r36.4.0-build-essential \
--build-arg TAR_INDEX_URL="http://jetson.webredirect.org:8000/jp5/cu126" \
--build-arg PIP_INDEX_REPO="http://jetson.webredirect.org/jp5/cu126" \
--build-arg PIP_TRUSTED_HOSTS="jetson.webredirect.org" \
--build-arg PIP_UPLOAD_REPO="http://localhost/jp5/cu126" \
--build-arg PIP_UPLOAD_USER="jp5" \
--build-arg PIP_UPLOAD_PASS="none" \
--build-arg SCP_UPLOAD_URL="localhost:/dist/jp5/cu126" \
--build-arg SCP_UPLOAD_USER="None" \
--build-arg SCP_UPLOAD_PASS="None" \
/home/jane/Code/jetson-containers/packages/cuda/cuda \
2>&1 | tee /home/jane/Code/jetson-containers/logs/20241002_212941/build/my_container_r36.4.0-pip_cache_cu126.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  48.64kB
Step 1/4 : ARG BASE_IMAGE
Step 2/4 : FROM ${BASE_IMAGE}
 ---> 990e4037e0a8
Step 3/4 : ARG PIP_INDEX_REPO     PIP_UPLOAD_REPO     PIP_UPLOAD_USER     PIP_UPLOAD_PASS     PIP_TRUSTED_HOSTS     TAR_INDEX_URL     SCP_UPLOAD_URL     SCP_UPLOAD_USER     SCP_UPLOAD_PASS
 ---> Using cache
 ---> c2efa4465977
Step 4/4 : ENV TAR_INDEX_URL=${TAR_INDEX_URL}     PIP_INDEX_URL=${PIP_INDEX_REPO}     PIP_TRUSTED_HOST=${PIP_TRUSTED_HOSTS}     TWINE_REPOSITORY_URL=${PIP_UPLOAD_REPO}     TWINE_USERNAME=${PIP_UPLOAD_USER}     TWINE_PASSWORD=${PIP_UPLOAD_PASS}     SCP_UPLOAD_URL=${SCP_UPLOAD_URL}     SCP_UPLOAD_USER=${SCP_UPLOAD_USER}     SCP_UPLOAD_PASS=${SCP_UPLOAD_PASS}
 ---> Using cache
[... 373 lines cut ...]
<details><pre><code>
--------------------- PLATFORM -------------------------
Machine: aarch64
System: Linux
Distribution: Ubuntu 22.04 Jammy Jellyfish
Release: 5.15.148-tegra
Python: 3.10.12
-------------------- JETSON RAW OUTPUT -----------------
------------------
Path: /etc/nv_tegra_release
# R36 (release), REVISION: 4.0, GCID: 37537400, BOARD: generic, EABI: aarch64, DATE: Fri Sep 13 04:36:44 UTC 2024
------------------
Path: /sys/firmware/devicetree/base/model
NVIDIA Jetson AGX Orin Developer Kit
------------------
Path: /proc/device-tree/nvidia,boardids
No such file or directory
------------------
Path: /proc/device-tree/compatible
nvidia,p3737-0000+p3701-0005nvidia,p3701-0005nvidia,tegra234
------------------
Path: /proc/device-tree/nvidia,dtsfilename
No such file or directory
------------------
</code></pre></details>

dusty-nv, 2024-10-02:
Hi @benswift, I had updated jetson-containers for initial JetPack 6.1 support a few days ago - can you try doing a git pull in your jetson-containers repo?   Then when you should start seeing jetson-containers to start reporting your board correctly as JetPack 6.1, and the python build should go through.  It takes a while to rebuild everything here, but PyTorch and Transformers are fine as should be diffusers.

benswift, 2024-10-02:
Thanks for the quick response. I am using the latest (53eb0e69d5a9931a21044a52299609bd16f0f6d5), yeah.

benswift, 2024-10-02:
One more update, just in case others are playing along at home: I had to get `pkginfo` from pypi.org, not the Jetson PA, so you'll need to change line 43 of the python `install.sh` to:

```sh
python3 -m pip install --upgrade pip pkginfo --index-url https://pypi.org/simple
```

(note the extra `pkginfo` in the middle there)

benswift, 2024-10-03:
Ok, by commenting out the `optimum` package I was able to build my `jetson-container` (which includes both diffusers and transformers). I pushed up the few small tweaks required [here](https://github.com/benswift/jetson-containers/commit/b22fd8770471b48e0f5919f7e7fb8ac08df6f43b) - @dusty-nv I'm happy to turn it into a proper PR if you want, just let me know.

dusty-nv, 2024-10-03:
Sorry @benswift, that change to l4t_version.py I thought I had merge into master, but it was still hiding in dev - that has now been merged, and you can see the recognition of JP 6.1 here:   https://github.com/dusty-nv/jetson-containers/blob/d1a10b66220a3322d39772027516e08995a919b1/jetson_containers/l4t_version.py#L87

@tokk-nv had reported a similar thing about `optimum` , although neither myself or others had seen that behavior.  Thanks for reporting it - will remove or fix it 👍

Fibo27, 2024-10-03:
@benswift : thanks for this - i have been struggling with the same since yesterday. I am though running into issues even after making the changes that you have suggested.

@dusty-nv : i had built a container for vlm using ros iron (nano_llm:iron-r36.3.0-cu122) please refer https://github.com/dusty-nv/jetson-containers/issues/622. This has been running without issues on an OrinNX whch is still on jetpack version 6.0. 

I tried to run the same container on an Orin AGX Developer Kit which is running on 36.4.0 and Jetpack 6.1, and I started getting errors such as these
`Fetching 13 files: 100%|█████████████████████| 13/13 [00:00<00:00, 53932.69it/s]
Fetching 17 files: 100%|█████████████████████| 17/17 [00:00<00:00, 44788.42it/s]
['/data/models/mlc/dist/VILA1.5-3b/ctx256/VILA1.5-3b-q4f16_ft/mlc-chat-config.json', '/data/models/mlc/dist/VILA1.5-3b/ctx256/VILA1.5-3b-q4f16_ft/params/mlc-chat-config.json']
Fatal Python error: Segmentation fault

Current thread 0x0000ffff99cc8da0 (most recent call first):
  Garbage-collecting
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/parallel/distributed.py", line 1222 in DistributedDataParallel
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/parallel/distributed.py", line 321 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/parallel/__init__.py",
 `

I then tried to build a new container with cuda 12.6 but the build process stops when building the OpenCV container. I am getting the below errors. 

It appears to be the same python related ones - look forward to your feedback

`traceback (most recent call last):
    File "/tmp/pip-build-env-1ci8_g7q/overlay/local/lib/python3.10/dist-packages/skbuild/setuptools_wrap.py", line 668, in setup
      cmkr.make(make_args, install_target=cmake_install_target, env=env)
    File "/tmp/pip-build-env-1ci8_g7q/overlay/local/lib/python3.10/dist-packages/skbuild/cmaker.py", line 696, in make
      self.make_impl(clargs=clargs, config=config, source_dir=source_dir, install_target=install_target, env=env)
    File "/tmp/pip-build-env-1ci8_g7q/overlay/local/lib/python3.10/dist-packages/skbuild/cmaker.py", line 741, in make_impl
      raise SKBuildError(msg)

  An error occurred while building with CMake.
    Command:
      /tmp/pip-build-env-1ci8_g7q/overlay/local/lib/python3.10/dist-packages/cmake/data/bin/cmake --build . --target install --config RELEASE --
    Install target:
      install
    Source directory:
      /opt/opencv-python
    Working directory:
      /opt/opencv-python/_skbuild/linux-aarch64-3.10/cmake-build
  Please check the install target is valid and see CMake's output for more information.

  error: subprocess-exited-with-error
    
  × Building wheel for opencv-contrib-python (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /usr/bin/python3.10 /usr/local/lib/python3.10/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py build_wheel /tmp/tmpi9uk2_pz
  cwd: /opt/opencv-python
  Building wheel for opencv-contrib-python (pyproject.toml): finished with status 'error'
  ERROR: Failed building wheel for opencv-contrib-python
Failed to build opencv-contrib-python
ERROR: Failed to build one or more wheels
The command '/bin/sh -c cd /tmp/opencv && ./install.sh || ./build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)

`

dusty-nv, 2024-10-03:
@benswift as of yesterday morning opencv was building, and the wheels are up on my pip server, which the containers should install instead of needing to build them (like it appears to have in your case).  I'm not sure what the actual compilation error is from that log, but will try again (I was also trying to build ROS for that)

MLC and NanoLLM I haven't gotten to yet, NanoLLM has a lot of dependencies that am in the process of working through.  Does PyTorch in the r36.3 containers still work for you on r36.4, or is it just MLC?

benswift, 2024-10-03:
@dusty-nv ah, I didn't think to look in the dev branch. Do you have any guidance (or are there any docs somewhere) on which packages it might be ok to pull from "regular" pypi.org and which ones have to come from the jetson.webredirect.org one?

@Fibo27 I actually had the exact same error trying to build the opencv container. I ended up just pip installing `opencv-python-headless` in _my_ container (which uses the `jetson-containers`-built container as a base), since that's all I need.

benswift, 2024-10-03:
@dusty-nv whoops - we were writing comments at the same time.

I'll re-pull the latest and check the opencv thing ASAP - will be able to do that in a couple of hours - and report back.

dusty-nv, 2024-10-03:
@benswift if you browse the previously-built wheels for CUDA 12.2, those are the ones that end up getting specially-built with CUDA support.   That pip server blacklists those packages from pypi, but mirrors others (so for example you can `pip install transformers` from it, and it will pull transformers from PyPi but CUDA-enabled pytorch from mine)

benswift, 2024-10-03:
Ok, I just tried to build opencv from the latest `master` (d1a10b66220a3322d39772027516e08995a919b1) and had the same twine/pkginfo version error I was getting before (and fixed in my branch by [getting pkginfo from the main pypi](https://github.com/benswift/jetson-containers/blob/b22fd8770471b48e0f5919f7e7fb8ac08df6f43b/packages/build/python/install.sh#L43). Log attached.

[twine-error.log](https://github.com/user-attachments/files/17250946/twine-error.log)

johnnynunez, 2024-10-03:
testing:
[image]

try to do first:
```
sudo docker system prune -a
```
and rebuild it

dusty-nv, 2024-10-03:
> `sudo docker system prune -a`

@benswift don't do that unless you actually want to remove all your docker images :rofl:

You can disable the build cache though, I think that's what Johnny means.  It's weird you get that twine error, will cherry-pick your patch 👍

johnnynunez, 2024-10-03:
> > `sudo docker system prune -a`
> 
> @benswift don't do that unless you actually want to remove all your docker images 🤣
> 
> You can disable the build cache though, I think that's what Johnny means. It's weird you get that twine error, will cherry-pick your patch 👍

yeap jajajaja sometimes I have conflicts, so I clean everything and rebuild it

johnnynunez, 2024-10-03:
> > `sudo docker system prune -a`
> 
> @benswift don't do that unless you actually want to remove all your docker images 🤣
> 
> You can disable the build cache though, I think that's what Johnny means. It's weird you get that twine error, will cherry-pick your patch 👍

only the "error" that I obtain is this one:
https://github.com/dusty-nv/jetson-containers/pull/656/files

johnnynunez, 2024-10-04:
[image]

benswift, 2024-10-04:
@dusty-nv one more update in case you're wondering.

I can build the xformers container, but the torch_tensorrt (which I'd _like_ to use, but isn't essential for me right now) one doesn't work. `packages/pytorch/torch_tensorrt/Dockerfile:18` copies in a `WORKSPACE.jp46` file which doesn't feel right given that I'm on jp61. And then it bombs out a bit later trying to stat a file that's not there. The intermediate `tensorrt` container does build, though. Logs below.

```
-- Building container stjet:r36.4.0-tensorrt

DOCKER_BUILDKIT=0 docker build --network=host --tag stjet:r36.4.0-tensorrt \
--file /home/jane/Code/jetson-containers/packages/tensorrt/Dockerfile.tar \
--build-arg BASE_IMAGE=stjet:r36.4.0-xformers \
--build-arg TENSORRT_URL="https://developer.nvidia.com/downloads/compute/machine-learning/tensorrt/10.4.0/tars/TensorRT-10.4.0.26.l4t.aarch64-gnu.cuda-12.6.tar.gz" \
/home/jane/Code/jetson-containers/packages/tensorrt \
2>&1 | tee /home/jane/Code/jetson-containers/logs/20241004_165713/build/stjet_r36.4.0-tensorrt.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  26.11kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> a7717d6dc2f7
Step 3/5 : ARG TENSORRT_URL
 ---> Using cache
 ---> 20fdd3940836
Step 4/5 : RUN set -ex &&     echo "Downloading ${TENSORRT_URL}" &&     mkdir -p /tmp/tensorrt &&     cd /tmp/tensorrt &&     wget --quiet --show-progress --progress=bar:force:noscroll ${TENSORRT_URL} -O TensorRT.tar &&     tar -xvf TensorRT.tar -C /usr/src &&     mv /usr/src/TensorRT-* /usr/src/tensorrt
 ---> Using cache
 ---> ec80a2b129b7
Step 5/5 : RUN cd /tmp/tensorrt &&     cp -r /usr/src/tensorrt/lib/* /usr/lib/$(uname -m)-linux-gnu/ &&     cp -r /usr/src/tensorrt/include/* /usr/include/$(uname -m)-linux-gnu/ &&     PY_VERSION=$(python3 -c 'import sys; print(f"{sys.version_info.major}{sys.version_info.minor}")') &&     pip3 install --verbose --no-cache-dir /usr/src/tensorrt/python/tensorrt-*-cp${PY_VERSION}-*.whl &&     rm -rf /tmp/tensorrt
 ---> Using cache
 ---> 5db9ddd5e7d0
Successfully built 5db9ddd5e7d0
Successfully tagged stjet:r36.4.0-tensorrt
-- Testing container stjet:r36.4.0-tensorrt (tensorrt:10.4/test.sh)

docker run -t --rm --runtime=nvidia --network=host \
--volume /home/jane/Code/jetson-containers/packages/tensorrt:/test \
--volume /home/jane/Code/jetson-containers/data:/data \
--workdir /test \
stjet:r36.4.0-tensorrt \
/bin/bash -c '/bin/bash test.sh' \
2>&1 | tee /home/jane/Code/jetson-containers/logs/20241004_165713/test/stjet_r36.4.0-tensorrt_test.sh.txt; exit ${PIPESTATUS[0]}

&&&& RUNNING TensorRT.trtexec [TensorRT v100400] [b26] # /usr/src/tensorrt/bin/trtexec --help
=== Model Options ===
  --onnx=<file>               ONNX model

=== Build Options ===
  --minShapes=spec                   Build with dynamic shapes using a profile with the min shapes provided
  --optShapes=spec                   Build with dynamic shapes using a profile with the opt shapes provided
  --maxShapes=spec                   Build with dynamic shapes using a profile with the max shapes provided
  --minShapesCalib=spec              Calibrate with dynamic shapes using a profile with the min shapes provided
  --optShapesCalib=spec              Calibrate with dynamic shapes using a profile with the opt shapes provided
  --maxShapesCalib=spec              Calibrate with dynamic shapes using a profile with the max shapes provided
                                     Note: All three of min, opt and max shapes must be supplied.
                                           However, if only opt shapes is supplied then it will be expanded so
                                           that min shapes and max shapes are set to the same values as opt shapes.
                                           Input names can be wrapped with escaped single quotes (ex: 'Input:0').
                                     Example input shapes spec: input0:1x3x256x256,input1:1x3x128x128
                                     For scalars (0-D shapes), use input0:scalar or simply input0: with nothing after the colon.
                                     Each input shape is supplied as a key-value pair where key is the input name and
                                     value is the dimensions (including the batch dimension) to be used for that input.
                                     Each key-value pair has the key and value separated using a colon (:).
                                     Multiple input shapes can be provided via comma-separated key-value pairs, and each input name can
                                     contain at most one wildcard ('*') character.
  --inputIOFormats=spec              Type and format of each of the input tensors (default = all inputs in fp32:chw)
                                     See --outputIOFormats help for the grammar of type and format list.
                                     Note: If this option is specified, please set comma-separated types and formats for all
                                           inputs following the same order as network inputs ID (even if only one input
                                           needs specifying IO format) or set the type and format once for broadcasting.
[... 317 lines cut ...]
[... cut ...]
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag stjet:r36.4.0-torch_tensorrt --file /home/jane/Code/jetson-containers/packages/pytorch/torch_tensorrt/Dockerfile --build-arg BASE_IMAGE=stjet:r36.4.0-bazel --build-arg PYTHON_VERSION="3.10" --build-arg JETPACK_MAJOR="6" --build-arg JETPACK_MINOR="0" --build-arg TORCH_TRT_VERSION="v1.4.0" /home/jane/Code/jetson-containers/packages/pytorch/torch_tensorrt 2>&1 | tee /home/jane/Code/jetson-containers/logs/20241004_165713/build/stjet_r36.4.0-torch_tensorrt.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

dusty-nv, 2024-10-04:
OK, there was an additional fix needed for OpenCV (see https://github.com/dusty-nv/jetson-containers/commit/f1b346b95b5e5c9e696ff60e2ba79644bc1a94a7)

Also checked in the changes to pip pkginfo and optimum.   torch_tensorrt I haven't touched in a while as it was challenging to keep building, instead end up using torch2trt a lot (coincidentally which I believe also needs updated for TensorRT 10, because I also need to finish updating jetson-inference for TRT10)

Sorry, you picked good week to start out haha, a lot of JetPack updates don't need a whole rebuild but this one was a pretty big version bump from CUDA 12.2->12.6, cuDNN 9.3, and TRT 10.  It's a big stack and can take a couple weeks for it all to settle down.

benswift, 2024-10-04:
All good mate, I have built and maintained enough software in my life to know that everything is a fragile house of cards and it’s a miracle anything ever works at all. Thanks for your support, and I will try those things that you suggested.

johnnynunez, 2024-10-04:
try now torch_tensorrt. It was pointing to very old version. Now is compatible with: PyTorch 2.4, CUDA 12.4, TensorRT 10.1, Python 3.12 https://github.com/dusty-nv/jetson-containers/pull/657

also torch2trt is compatible with tensorrt 10 in last commit

johnnynunez, 2024-10-04:
@dusty-nv you can see it: https://github.com/pytorch/TensorRT/compare/main...lluo/jp6.1

dusty-nv, 2024-10-04:
OK thanks yea, it's still not building, will wait until updated and merged for CUDA 12.6.   Indeed torch2trt built and is working though!   `dustynv/torch2trt:r36.4.0`

johnnynunez, 2024-10-04:
> OK thanks yea, it's still not building, will wait until updated and merged for CUDA 12.6. Indeed torch2trt built and is working though! `dustynv/torch2trt:r36.4.0`

super mate!

Fibo27, 2024-10-04:
@dusty-nv : i started building the nano_llm container with ros and i could build the following packages - this of course a big progress since yesterday.

[image]
     

However I get the following error
Installing collected packages: protobuf, onnx
  changing mode of /usr/local/bin/backend-test-tools to 755
  changing mode of /usr/local/bin/check-model to 755
  changing mode of /usr/local/bin/check-node to 755
Successfully installed onnx-1.17.0 protobuf-5.28.2
Name: onnx
Version: 1.17.0
Summary: Open Neural Network Exchange
Home-page: https://onnx.ai/
Author: 
Author-email: ONNX Contributors <onnx-technical-discuss@lists.lfaidata.foundation>
License: Apache License v2.0
Location: /usr/local/lib/python3.10/dist-packages
Requires: numpy, protobuf
Required-by: 
1.17.0
 ---> Removed intermediate container ec62e5a2bc46
 ---> 79b28a3da5ae
Successfully built 79b28a3da5ae
Successfully tagged nano_llm:iron-r36.4.0-cu126-onnx
-- Testing container nano_llm:iron-r36.4.0-cu126-onnx (onnx/test.py)

sudo docker run -t --rm --runtime=nvidia --network=host \
--volume /home/test/jetson-containers/packages/ml/onnx:/test \
--volume /home/test/jetson-containers/data:/data \
--workdir /test \
nano_llm:iron-r36.4.0-cu126-onnx \
/bin/bash -c 'python3 test.py' \
2>&1 | tee /home/test/jetson-containers/logs/20241004_141018/test/nano_llm_iron-r36.4.0-cu126-onnx_test.py.txt; exit ${PIPESTATUS[0]}

exec /ros_entrypoint.sh: exec format error
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/test/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/home/test/jetson-containers/jetson_containers/container.py", line 154, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/test/jetson-containers/jetson_containers/container.py", line 327, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo docker run -t --rm --runtime=nvidia --network=host --volume /home/test/jetson-containers/packages/ml/onnx:/test --volume /home/test/jetson-containers/data:/data --workdir /test nano_llm:iron-r36.4.0-cu126-onnx /bin/bash -c 'python3 test.py' 2>&1 | tee /home/test/jetson-containers/logs/20241004_141018/test/nano_llm_iron-r36.4.0-cu126-onnx_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

benswift, 2024-10-07:
hey @dusty-nv I tried again today with the latest master (1f5dee99fee).

The pkginfo thing has been fixed, so my `diffusers transformers` container now builds cleanly on master (and I don't need my branch anymore).

The optimum one still fails, though:

```
 ---> 92e86450ef25
Step 5/7 : RUN pip3 install --no-cache-dir --verbose accelerate &&     pip3 install --no-cache-dir --verbose sentencepiece &&     pip3 install --no-cache-dir --verbose optimum &&         pip3 uninstall -y transformers &&     pip3 install --no-cache-dir --verbose ${TRANSFORMERS_PACKAGE} &&         PYTHON_ROOT=`pip3 show transformers | grep Location: | cut -d' ' -f2` &&     sed -i 's|torch.distributed.is_initialized|torch.distributed.is_available|g' -i ${PYTHON_ROOT}/transformers/modeling_utils.py
 ---> Running in b1962ec3aaa5
Using pip 24.2 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: http://jetson.webredirect.org/jp6/cu126
Collecting accelerate
  Downloading http://jetson.webredirect.org/root/pypi/%2Bf/d69/159e2c4e4a473/accelerate-0.34.2-py3-none-any.whl (324 kB)
Requirement already satisfied: numpy<3.0.0,>=1.17 in /usr/local/lib/python3.10/dist-packages (from accelerate) (1.26.4)
Requirement already satisfied: packaging>=20.0 in /usr/local/lib/python3.10/dist-packages (from accelerate) (24.1)
Requirement already satisfied: psutil in /usr/local/lib/python3.10/dist-packages (from accelerate) (6.0.0)
Requirement already satisfied: pyyaml in /usr/local/lib/python3.10/dist-packages (from accelerate) (6.0.2)
Requirement already satisfied: torch>=1.10.0 in /usr/local/lib/python3.10/dist-packages (from accelerate) (2.4.0)
Requirement already satisfied: huggingface-hub>=0.21.0 in /usr/local/lib/python3.10/dist-packages (from accelerate) (0.25.1)
Collecting safetensors>=0.4.3 (from accelerate)
  Downloading http://jetson.webredirect.org/root/pypi/%2Bf/688/5016f34bef80e/safetensors-0.4.5-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (441 kB)
Requirement already satisfied: filelock in /usr/local/lib/python3.10/dist-packages (from huggingface-hub>=0.21.0->accelerate) (3.16.1)
Requirement already satisfied: fsspec>=2023.5.0 in /usr/local/lib/python3.10/dist-packages (from huggingface-hub>=0.21.0->accelerate) (2024.9.0)
Requirement already satisfied: requests in /usr/local/lib/python3.10/dist-packages (from huggingface-hub>=0.21.0->accelerate) (2.32.3)
Requirement already satisfied: tqdm>=4.42.1 in /usr/local/lib/python3.10/dist-packages (from huggingface-hub>=0.21.0->accelerate) (4.66.5)
Requirement already satisfied: typing-extensions>=3.7.4.3 in /usr/local/lib/python3.10/dist-packages (from huggingface-hub>=0.21.0->accelerate) (4.12.2)
Requirement already satisfied: sympy in /usr/local/lib/python3.10/dist-packages (from torch>=1.10.0->accelerate) (1.13.3)
Requirement already satisfied: networkx in /usr/local/lib/python3.10/dist-packages (from torch>=1.10.0->accelerate) (3.3)
Requirement already satisfied: jinja2 in /usr/local/lib/python3.10/dist-packages (from torch>=1.10.0->accelerate) (3.1.4)
Requirement already satisfied: MarkupSafe>=2.0 in /usr/local/lib/python3.10/dist-packages (from jinja2->torch>=1.10.0->accelerate) (2.1.5)
Requirement already satisfied: charset-normalizer<4,>=2 in /usr/local/lib/python3.10/dist-packages (from requests->huggingface-hub>=0.21.0->accelerate) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in /usr/local/lib/python3.10/dist-packages (from requests->huggingface-hub>=0.21.0->accelerate) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in /usr/local/lib/python3.10/dist-packages (from requests->huggingface-hub>=0.21.0->accelerate) (2.2.3)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.10/dist-packages (from requests->huggingface-hub>=0.21.0->accelerate) (2024.8.30)
Requirement already satisfied: mpmath<1.4,>=1.1.0 in /usr/local/lib/python3.10/dist-packages (from sympy->torch>=1.10.0->accelerate) (1.3.0)
Installing collected packages: safetensors, accelerate
  changing mode of /usr/local/bin/accelerate to 755
  changing mode of /usr/local/bin/accelerate-config to 755
  changing mode of /usr/local/bin/accelerate-estimate-memory to 755
  changing mode of /usr/local/bin/accelerate-launch to 755
  changing mode of /usr/local/bin/accelerate-merge-weights to 755
Successfully installed accelerate-0.34.2 safetensors-0.4.5
Using pip 24.2 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: http://jetson.webredirect.org/jp6/cu126
Collecting sentencepiece
  Downloading http://jetson.webredirect.org/root/pypi/%2Bf/2fd/e4b08cfe237be/sentencepiece-0.2.0-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (1.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.3/1.3 MB 324.7 kB/s eta 0:00:00
Installing collected packages: sentencepiece
Successfully installed sentencepiece-0.2.0
Using pip 24.2 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: http://jetson.webredirect.org/jp6/cu126
ERROR: Could not find a version that satisfies the requirement optimum (from versions: none)
ERROR: No matching distribution found for optimum
The command '/bin/sh -c pip3 install --no-cache-dir --verbose accelerate &&     pip3 install --no-cache-dir --verbose sentencepiece &&     pip3 install --no-cache-dir --verbose optimum &&         pip3 uninstall -y transformers &&     pip3 install --no-cache-dir --verbose ${TRANSFORMERS_PACKAGE} &&         PYTHON_ROOT=`pip3 show transformer [... cut ...] alized|torch.distributed.is_available|g' -i ${PYTHON_ROOT}/transformers/modeling_utils.py' returned a non-zero code: 1
```

Anyway, I can make do with what I've got for now. Re: optimum you mentioned above you checked in some changes - are they in master or only on dev? 

Do you think that tensorrt (via optimum) would give me some easy speedups for a Stable Diffusion pipeline on the AGX Orin? This is all for an interactive art installation where higher throughput for the pipeline would be handy.

benswift, 2024-10-09:
Hey @dusty-nv I actually spoke to soon above - I was still using my local change where I had commented out the optimus dep.

Again, my goal is to build a container with diffusers, transformers, and torch2trt (or torch_tensorrt, or some other way of using TensorRT-accelerated pytorch).

As a first step, I ensure I'm on a clean `master`:

```
jane@ubuntu:~/Code/jetson-containers$ git show HEAD
commit 0189a3455531341c79b5877f58e3b30e92fc10b2 (HEAD -> master, origin/master, origin/HEAD)
Merge: 1f5dee99 e11c43ca
Author: Dustin Franklin <dustinf@nvidia.com>
Date:   Mon Oct 7 03:33:25 2024 -0400

    Merge remote-tracking branch 'origin/dev'

jane@ubuntu:~/Code/jetson-containers$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```
and I try and build the transformers container:

<details><pre><code>
jane@ubuntu:~/Code/jetson-containers$ jetson-containers build --name=my_container transformers
Namespace(packages=['transformers'], name='my_container', base='', multiple=False, build_flags='', build_args='', package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', logs='', verbose=False, no_github_api=False)
-- L4T_VERSION=36.4.0
-- JETPACK_VERSION=6.1
-- CUDA_VERSION=12.6
-- PYTHON_VERSION=3.10
-- LSB_RELEASE=22.04 (jammy)
-- Building containers  ['build-essential', 'pip_cache:cu126', 'cuda:12.6', 'cudnn', 'python', 'numpy', 'cmake', 'onnx', 'pytorch:2.4', 'torchvision', 'huggingface_hub', 'rust', 'transformers']
-- Building container my_container:r36.4.0-build-essential

DOCKER_BUILDKIT=0 docker build --network=host --tag my_container:r36.4.0-build-essential \
--file /home/jane/Code/jetson-containers/packages/build/build-essential/Dockerfile \
--build-arg BASE_IMAGE=ubuntu:22.04 \
/home/jane/Code/jetson-containers/packages/build/build-essential \
2>&1 | tee /home/jane/Code/jetson-containers/logs/20241009_134747/build/my_container_r36.4.0-build-essential.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  19.97kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> 981912c48e9a
Step 3/5 : ENV DEBIAN_FRONTEND=noninteractive     LANGUAGE=en_US:en     LANG=en_US.UTF-8     LC_ALL=en_US.UTF-8
 ---> Using cache
 ---> bee7dec2ca35
Step 4/5 : RUN set -ex     && apt-get update     && apt-get install -y --no-install-recommends         locales         locales-all         tzdata     && locale-gen en_US $LANG     && update-locale LC_ALL=$LC_ALL LANG=$LANG     && locale         && apt-get install -y --no-install-recommends         build-essential         software-properties-common         apt-transport-https         ca-certificates         lsb-release         pkg-config         gnupg         git         gdb         wget         curl         nano         zip         unzip         time 	   sshpass 	   ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*         && gcc --version     && g++ --version
 ---> Using cache
 ---> 5a61c6947b48
Step 5/5 : COPY tarpack /usr/local/bin/
 ---> Using cache
 ---> 990e4037e0a8
Successfully built 990e4037e0a8
Successfully tagged my_container:r36.4.0-build-essential
-- Building container my_container:r36.4.0-pip_cache_cu126

DOCKER_BUILDKIT=0 docker build --network=host --tag my_container:r36.4.0-pip_cache_cu126 \
--file /home/jane/Code/jetson-containers/packages/cuda/cuda/Dockerfile.pip \
--build-arg BASE_IMAGE=my_container:r36.4.0-build-essential \
--build-arg TAR_INDEX_URL="http://jetson.webredirect.org:8000/jp6/cu126" \
--build-arg PIP_INDEX_REPO="http://jetson.webredirect.org/jp6/cu126" \
--build-arg PIP_TRUSTED_HOSTS="jetson.webredirect.org" \
--build-arg PIP_UPLOAD_REPO="http://localhost/jp6/cu126" \
--build-arg PIP_UPLOAD_USER="jp6" \
--build-arg PIP_UPLOAD_PASS="none" \
[... 804 lines cut ...]
Collecting sentencepiece
  Downloading http://jetson.webredirect.org/root/pypi/%2Bf/2fd/e4b08cfe237be/sentencepiece-0.2.0-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (1.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.3/1.3 MB 1.2 MB/s eta 0:00:00
Installing collected packages: sentencepiece
Successfully installed sentencepiece-0.2.0
Using pip 24.2 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
Looking in indexes: http://jetson.webredirect.org/jp6/cu126
ERROR: Could not find a version that satisfies the requirement optimum (from versions: none)
ERROR: No matching distribution found for optimum
The command '/bin/sh -c pip3 install --no-cache-dir --verbose accelerate &&     pip3 install --no-cache-dir --verbose sentencepiece &&     pip3 install --no-cache-dir --verbose optimum &&         pip3 uninstall -y transformers &&     pip3 install --no-cache-dir --verbose ${TRANSFORMERS_PACKAGE} &&         PYTHON_ROOT=`pip3 show transformers | grep Location: | cut -d' ' -f2` &&     sed -i 's|torch.distributed.is_initialized|torch.distributed.is_available|g' -i ${PYTHON_ROOT}/transformers/modeling_utils.py' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
[... cut ...] er:r36.4.0-rust /home/jane/Code/jetson-containers/packages/llm/transformers 2>&1 | tee /home/jane/Code/jetson-containers/logs/20241009_134747/build/my_container_r36.4.0-transformers.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
</code></pre></details>

So `optimus` is still the issue. Can you confirm that this should build (latest JetPack 6.1, Orin AGX 64GB, everything else up to date)? Because if I can't build the transformers container then I can't even get to the tensorrt part.

johnnynunez, 2024-10-09:
> Hey @dusty-nv I actually spoke to soon above - I was still using my local change where I had commented out the optimus dep.
> 
> Again, my goal is to build a container with diffusers, transformers, and torch2trt (or torch_tensorrt, or some other way of using TensorRT-accelerated pytorch).
> 
> As a first step, I ensure I'm on a clean `master`:
> 
> ```
> jane@ubuntu:~/Code/jetson-containers$ git show HEAD
> commit 0189a3455531341c79b5877f58e3b30e92fc10b2 (HEAD -> master, origin/master, origin/HEAD)
> Merge: 1f5dee99 e11c43ca
> Author: Dustin Franklin <dustinf@nvidia.com>
> Date:   Mon Oct 7 03:33:25 2024 -0400
> 
>     Merge remote-tracking branch 'origin/dev'
> 
> jane@ubuntu:~/Code/jetson-containers$ git status
> On branch master
> Your branch is up to date with 'origin/master'.
> 
> nothing to commit, working tree clean
> ```
> 
> and I try and build the transformers container:
> 
> So `optimus` is still the issue. Can you confirm that this should build (latest JetPack 6.1, Orin AGX 64GB, everything else up to date)? Because if I can't build the transformers container then I can't even get to the tensorrt part.

What is yout command/prompt?

benswift, 2024-10-09:
Do you mean how am I invoking the `jetson-containers` tool to build the container? It's at the top of that log in my previous message (collapsed inside a `<details>` element.

```
jane@ubuntu:~/Code/jetson-containers$ jetson-containers build --name=my_container transformers
```

Or do you mean what's my shell environment or something? Here's the output of `env`:

```
jane@ubuntu:~/Code/jetson-containers$ env
SHELL=/bin/bash
SESSION_MANAGER=local/ubuntu:@/tmp/.ICE-unix/2743,unix/ubuntu:/tmp/.ICE-unix/2743
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
SSH_AGENT_LAUNCHER=gnome-keyring
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
GTK_IM_MODULE=ibus
JETSON_CUDA_ARCH_BIN=8.7
GNOME_SHELL_SESSION_MODE=ubuntu
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
GTK_MODULES=gail:atk-bridge
PWD=/home/jane/Code/jetson-containers
LOGNAME=jane
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=x11
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
SYSTEMD_EXEC_PID=2929
JETSON_MODULE=NVIDIA Jetson AGX Orin (64GB ram)
XAUTHORITY=/run/user/1000/gdm/Xauthority
WINDOWPATH=2
HOME=/home/jane
USERNAME=jane
IM_CONFIG_PHASE=1
LANG=en_AU.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=ubuntu:GNOME
VTE_VERSION=6800
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/e7d5b141_7b4e_4cf1_be03_098c8ea617a8
JETSON_L4T=36.4.0
JETSON_MODEL=NVIDIA Jetson AGX Orin Developer Kit
JETSON_SERIAL_NUMBER=1421524282093
HF_HOME=/ssd/huggingface
LESSCLOSE=/usr/bin/lesspipe %s %s
JETSON_SOC=tegra234
XDG_SESSION_CLASS=user
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=jane
GNOME_TERMINAL_SERVICE=:1.284
DISPLAY=:1
SHLVL=1
QT_IM_MODULE=ibus
LD_LIBRARY_PATH=/usr/local/cuda-12.6/lib64:
XDG_RUNTIME_DIR=/run/user/1000
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
PATH=/home/jane/.rye/shims:/home/jane/.local/bin:/usr/local/cuda-12.6/bin:/home/jane/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
GDMSESSION=ubuntu
JETSON_P_NUMBER=p3701-0005
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
JETSON_JETPACK=6.1
_=/usr/bin/env
OLDPWD=/home/jane/Code/storytellers
```

johnnynunez, 2024-10-09:
> Do you mean how am I invoking the `jetson-containers` tool to build the container? It's at the top of that log in my previous message (collapsed inside a `<details>` element.
>
> ``` [...]

in which branch are you?
CUDA_VERSION=12.6 PYTHON_VERSION=3.10 jetson-containers build transformers
[image]

benswift, 2024-10-09:
I'm on `master`.

```
jane@ubuntu:~/Code/jetson-containers$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
jane@ubuntu:~/Code/jetson-containers$ git remote show origin
* remote origin
  Fetch URL: https://github.com/dusty-nv/jetson-containers
  Push  URL: https://github.com/dusty-nv/jetson-containers
  HEAD branch: master
  Remote branches:
    L4T-R32.4.2 tracked
    L4T-R32.4.3 tracked
    L4T-R32.4.4 tracked
    L4T-R32.5.0 tracked
    L4T-R32.6.1 tracked
    L4T-R32.7.1 tracked
    L4T-R34.1.0 tracked
    L4T-R34.1.1 tracked
    L4T-R35.1.0 tracked
    L4T-R35.2.1 tracked
    L4T-R35.3.1 tracked
    L4T-R35.4.1 tracked
    L4T-R36.2.0 tracked
    L4T-R36.3.0 tracked
    crossformer tracked
    dev         tracked
    docs        tracked
    jp6         tracked
    legacy      tracked
    master      tracked
    trt-llm     tracked
  Local branch configured for 'git pull':
    master merges with remote master
  Local ref configured for 'git push':
    master pushes to master (up to date)
```

benswift, 2024-10-09:
Thanks for your help, btw. I can confirm that I just tried literally the same command:

    CUDA_VERSION=12.6 PYTHON_VERSION=3.10 jetson-containers build transformers

and I still get the error about optimum

    ERROR: Could not find a version that satisfies the requirement optimum (from versions: none)
    ERROR: No matching distribution found for optimum

johnnynunez, 2024-10-09:
> Thanks for your help, btw. I can confirm that I just tried literally the same command:
>
> ```
> CUDA_VERSION=12.6 PYTHON_VERSION=3.10 jetson-containers build transformers
> ```
>
> and I still get the error [...]

i'm on master too, wtf jajaj it is strange, optimum has arm wheels

dusty-nv, 2024-10-09:
Yes it is strange, I have still never gotten this error, and had changed how it was installed.  @benswift just comment it out in the transformers dockerfile

johnnynunez, 2024-10-09:
> Yes it is strange, I have still never gotten this error, and had changed how it was installed. @benswift just comment it out in the transformers dockerfile

wow, you are here, hero

benswift, 2024-10-09:
Thanks all. I'm open to building stuff manually if there's any doco (doesn't have to be super polished, just enough for me to fill-in-the-gaps). Or @dusty-nv is there anything else you think I should try (dev branch, run further diagnostics, etc)?

johnnynunez, 2024-10-09:
> Thanks all. I'm open to building stuff manually if there's any doco (doesn't have to be super polished, just enough for me to fill-in-the-gaps). Or @dusty-nv is there anything else you think I should try (dev branch, run further diagnostics, etc)?

If you have nothing changed,
Could you start from inititial steps?
1. Clone the repo again
2. clean all your images : sudo system prune -a
3. CUDA_VERSION=12.6 PYTHON_VERSION=3.10 jetson-containers build transformers

benswift, 2024-10-09:
yep, will do

johnnynunez, 2024-10-09:
@dusty-nv Now I understand the error, we use a newer version of transformer, 4.45 (https://github.com/huggingface/optimum/commit/049b00f61c9bb17bd2b20a3b77d04cc4c0f20d86) which optimum has not yet released to be compatible (tags/release), therefore, it does not install. Maybe, here we need your opinion whether to keep optimum until now and will update it, or create build to make transformer:4.44 & optimum:4.44 type. or another solution

benswift, 2024-10-11:
Oh, that's good news. Do I have to wait for the jetson.webredirect.org pip cache to be updated? I'm not sure if that's a nightly cronjob or has to be triggered manually by an admin.

johnnynunez, 2024-10-11:
> Oh, that's good news. Do I have to wait for the jetson.webredirect.org pip cache to be updated? I'm not sure if that's a nightly cronjob or has to be triggered manually by an admin.

it is updated manually if the index is cuda. For optimum is on root http://jetson.webredirect.org/root/pypi/optimum/stable

dusty-nv, 2024-10-11:
Yea, optimum and transformers don't need CUDA/NVCC compilation directly (all the CUDA stuff they use is in dependencies), so I don't need to have wheels for them.  The 'root index' is just a mirror of pypi.   That's interesting that even though we don't pin transformers version in the dockerfile, it was still failing to resolve optimum...

huuck, 2024-12-28:
Holy molly, 3 months later and it's still happening.
[image]

This is straight from your tutorial pages for text-generation-webui (https://www.jetson-ai-lab.com/tutorial_text-generation.html). At least have the decency to remove that content if it's powered by the same corporate overlords.

closed 2025-02-21 by johnnynunez