dusty-nv/jetson-containers #661: Request adaptation of the Llama-factory container to support VLLM inference

JIANGTUNAN, 2024-10-05:
start command:
```shell
jetson-containers run dustynv/llama-factory:r36.4.0
``` 

log:
```shell
To create a public link, set `share=True` in `launch()`.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/transformers/utils/versions.py", line 102, in require_version
    got_ver = importlib.metadata.version(pkg)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 996, in version
    return distribution(distribution_name).version
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 969, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 548, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: No package metadata was found for vllm

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/gradio/queueing.py", line 575, in process_events
    response = await route_utils.call_process_api(
  File "/usr/local/lib/python3.10/dist-packages/gradio/route_utils.py", line 322, in call_process_api
    output = await app.get_blocks().process_api(
  File "/usr/local/lib/python3.10/dist-packages/gradio/blocks.py", line 1935, in process_api
    result = await self.call_function(
  File "/usr/local/lib/python3.10/dist-packages/gradio/blocks.py", line 1532, in call_function
    prediction = await utils.async_iteration(iterator)
  File "/usr/local/lib/python3.10/dist-packages/gradio/utils.py", line 671, in async_iteration
    return await iterator.__anext__()
  File "/usr/local/lib/python3.10/dist-packages/gradio/utils.py", line 664, in __anext__
    return await anyio.to_thread.run_sync(
  File "/usr/local/lib/python3.10/dist-packages/anyio/to_thread.py", line 56, in run_sync
    return await get_async_backend().run_sync_in_worker_thread(
  File "/usr/local/lib/python3.10/dist-packages/anyio/_backends/_asyncio.py", line 2405, in run_sync_in_worker_thread
    return await future
  File "/usr/local/lib/python3.10/dist-packages/anyio/_backends/_asyncio.py", line 914, in run
    result = context.run(func, *args)
  File "/usr/local/lib/python3.10/dist-packages/gradio/utils.py", line 647, in run_sync_iterator_async
    return next(iterator)
  File "/usr/local/lib/python3.10/dist-packages/gradio/utils.py", line 809, in gen_wrapper
    response = next(iterator)
  File "/opt/LLaMA-Factory/src/llamafactory/webui/chatter.py", line 104, in load_model
    super().__init__(args)
  File "/opt/LLaMA-Factory/src/llamafactory/chat/chat_model.py", line 49, in __init__
    model_args, data_args, finetuning_args, generating_args = get_infer_args(args)
  File "/opt/LLaMA-Factory/src/llamafactory/hparams/parser.py", line 389, in get_infer_args
    _check_extra_dependencies(model_args, finetuning_args)
  File "/opt/LLaMA-Factory/src/llamafactory/hparams/parser.py", line 126, in _check_extra_dependencies
    require_version("vllm>=0.4.3,<=0.6.0", "To fix: pip install vllm>=0.4.3,<=0.6.0")
  File "/usr/local/lib/python3.10/dist-packages/transformers/utils/versions.py", line 104, in require_version
    raise importlib.metadata.PackageNotFoundError(
importlib.metadata.PackageNotFoundError: No package metadata was found for The 'vllm>=0.4.3,<=0.6.0' distribution was not found and is required by this application. 
To fix: pip install vllm>=0.4.3,<=0.6.0
``` 

[image]

I tried entering the container to manually install the dependencies, but it reported an error：
```shell
Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ?─> [13 lines of output]
      Looking in indexes: http://jetson.webredirect.org/jp6/cu126
      Collecting cmake>=3.21
        Downloading http://jetson.webredirect.org/root/pypi/%2Bf/0a5/929e21af39a3a/cmake-3.30.4-py3-none-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (26.3 MB)
           ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 26.3/26.3 MB 5.6 MB/s eta 0:00:00
      Collecting ninja
        Downloading http://jetson.webredirect.org/root/pypi/%2Bf/73b/93c14046447c7/ninja-1.11.1.1-py2.py3-none-manylinux2014_aarch64.manylinux_2_17_aarch64.whl (155 kB)
      Collecting packaging
        Downloading http://jetson.webredirect.org/root/pypi/%2Bf/5b8/f2217dbdbd2f7/packaging-24.1-py3-none-any.whl (53 kB)
      Collecting setuptools>=49.4.0
        Downloading http://jetson.webredirect.org/root/pypi/%2Bf/35a/b7fd3bcd95e6b/setuptools-75.1.0-py3-none-any.whl (1.2 MB)
           ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 5.2 MB/s eta 0:00:00
      ERROR: Could not find a version that satisfies the requirement torch==2.3.1 (from versions: 2.4.0)
      ERROR: No matching distribution found for torch==2.3.1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
?─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

``` 
[image]

johnnynunez, 2024-10-05:
I'm checking

dusty-nv, 2024-10-05:
@johnnynunez @JIANGTUNAN we should just patch llama-factory to remove dependency on vLLM, because vLLM would not build on Jetson previously, and it is too complex for us to support alongside MLC and TRT-LLM (which are both higher performance than vLLM)

closed 2024-10-06 by JIANGTUNAN

johnnynunez, 2024-10-06:
I've created the jetson container to support vllm. Now vllm supports a lot of things, and the new model, is easy to build and support current pytorch installation with the actual cuda. 
https://github.com/dusty-nv/jetson-containers/pull/664

johnnynunez, 2024-10-09:
hello @JIANGTUNAN 
I've created the docker and now llama-factory supports vllm:
https://github.com/dusty-nv/jetson-containers/pull/670

JIANGTUNAN, 2024-10-11:
@johnnynunez Hello! I'm grateful to have a member like you in the community. You've really been a great help. Thank you for your contributions![image]

JIANGTUNAN, 2024-10-11:
Commands:
```shell
jetson-containers update
jetson-containers build llama-factory
```

Description:
After pulling the new build script, I tried to build the new llama-factory. However, the build failed, and from the logs, it seems the memory was insufficient and caused the compilation to be forcibly terminated. I tried stopping other services to allocate more resources for the compilation, but it still failed.

Device:
Jetson Orin NX 16G

Logs:
```shell
[34/104] Building CUDA object CMakeFiles/_C.dir/csrc/quantization/gptq_marlin/gptq_marlin_repack.cu.o
[35/104] Building CUDA object CMakeFiles/_C.dir/csrc/quantization/gptq_marlin/awq_marlin_repack.cu.o
[36/104] Building CUDA object CMakeFiles/_C.dir/csrc/quantization/gptq_marlin/gptq_marlin.cu.o
  FAILED: CMakeFiles/_C.dir/csrc/quantization/gptq_marlin/gptq_marlin.cu.o
  /usr/local/cuda/bin/nvcc -forward-unknown-to-host-compiler -DCUTLASS_ENABLE_DIRECT_CUDA_DRIVER_CALL=1 -DPy_LIMITED_API=3 -DTORCH_EXTENSION_NAME=_C -DUSE_C10D_GLOO -DUSE_C10D_MPI -DUSE_C10D_NCCL -DUSE_DISTRIBUTED -DUSE_RPC -DUSE_TENSORPIPE -D_C_EXPORTS -I/opt/vllm/csrc -I/opt/vllm/.deps/cutlass-src/include -isystem /usr/include/python3.10 -isystem /usr/local/lib/python3.10/dist-packages/torch/include -isystem /usr/local/lib/python3.10/dist-packages/torch/include/torch/csrc/api/include -isystem /usr/local/cuda/include -DONNX_NAMESPACE=onnx_c2 -Xcudafe --diag_suppress=cc_clobber_ignored,--diag_suppress=field_without_dll_interface,--diag_suppress=base_class_has_different_dll_interface,--diag_suppress=dll_interface_conflict_none_assumed,--diag_suppress=dll_interface_conflict_dllexport_assumed,--diag_suppress=bad_friend_decl --expt-relaxed-constexpr --expt-extended-lambda -O2 -g -DNDEBUG -std=c++17 -Xcompiler=-fPIC --expt-relaxed-constexpr -DENABLE_FP8 --threads=1 -DENABLE_SCALED_MM_C2X=1 -D_GLIBCXX_USE_CXX11_ABI=1 -gencode arch=compute_86,code=sm_86 -MD -MT CMakeFiles/_C.dir/csrc/quantization/gptq_marlin/gptq_marlin.cu.o -MF CMakeFiles/_C.dir/csrc/quantization/gptq_marlin/gptq_marlin.cu.o.d -x cu -c /opt/vllm/csrc/quantization/gptq_marlin/gptq_marlin.cu -o CMakeFiles/_C.dir/csrc/quantization/gptq_marlin/gptq_marlin.cu.o
  Killed
  [37/104] Building CUDA object CMakeFiles/_C.dir/csrc/attention/attention_kernels.cu.o
  FAILED: CMakeFiles/_C.dir/csrc/attention/attention_kernels.cu.o
  /usr/local/cuda/bin/nvcc -forward-unknown-to-host-compiler -DCUTLASS_ENABLE_DIRECT_CUDA_DRIVER_CALL=1 -DPy_LIMITED_API=3 -DTORCH_EXTENSION_NAME=_C -DUSE_C10D_GLOO -DUSE_C10D_MPI -DUSE_C10D_NCCL -DUSE_DISTRIBUTED -DUSE_RPC -DUSE_TENSORPIPE -D_C_EXPORTS -I/opt/vllm/csrc -I/opt/vllm/.deps/cutlass-src/include -isystem /usr/include/python3.10 -isystem /usr/local/lib/python3.10/dist-packages/torch/include -isystem /usr/local/lib/python3.10/dist-packages/torch/include/torch/csrc/api/include -isystem /usr/local/cuda/include -DONNX_NAMESPACE=onnx_c2 -Xcudafe --diag_suppress=cc_clobber_ignored,--diag_suppress=field_without_dll_interface,--diag_suppress=base_class_has_different_dll_interface,--diag_suppress=dll_interface_conflict_none_assumed,--diag_suppress=dll_interface_conflict_dllexport_assumed,--diag_suppress=bad_friend_decl --expt-relaxed-constexpr --expt-extended-lambda -O2 -g -DNDEBUG -std=c++17 -Xcompiler=-fPIC --expt-relaxed-constexpr -DENABLE_FP8 --threads=1 -DENABLE_SCALED_MM_C2X=1 -D_GLIBCXX_USE_CXX11_ABI=1 -gencode arch=compute_87,code=sm_87 -MD -MT CMakeFiles/_C.dir/csrc/attention/attention_kernels.cu.o -MF CMakeFiles/_C.dir/csrc/attention/attention_kernels.cu.o.d -x cu -c /opt/vllm/csrc/attention/attention_kernels.cu -o CMakeFiles/_C.dir/csrc/attention/attention_kernels.cu.o
  Killed
  [38/104] Building CUDA object CMakeFiles/_C.dir/csrc/quantization/cutlass_w8a8/scaled_mm_c2x.cu.o
  FAILED: CMakeFiles/_C.dir/csrc/quantization/cutlass_w8a8/scaled_mm_c2x.cu.o
  /usr/local/cuda/bin/nvcc -forward-unknown-to-host-compiler -DCUTLASS_ENABLE_DIRECT_CUDA_DRIVER_CALL=1 -DPy_LIMITED_API=3 -DTORCH_EXTENSION_NAME=_C -DUSE_C10D_GLOO -DUSE_C10D_MPI -DUSE_C10D_NCCL -DUSE_DISTRIBUTED -DUSE_RPC -DUSE_TENSORPIPE -D_C_EXPORTS -I/opt/vllm/csrc -I/opt/vllm/.deps/cutlass-src/include -isystem /usr/include/python3.10 -isystem /usr/local/lib/python3.10/dist-packages/torch/include -isystem /usr/local/lib/python3.10/dist-packages/torch/include/torch/csrc/api/include -isystem /usr/local/cuda/include -DONNX_NAMESPACE=onnx_c2 -Xcudafe --diag_suppress=cc_clobber_ignored,--diag_suppress=field_without_dll_interface,--diag_suppress=base_class_has_different_dll_interface,--diag_suppress=dll_interface_conflict_none_assumed,--diag_suppress=dll_interface_conflict_dllexport_assumed,--diag_suppress=bad_friend_decl --expt-relaxed-constexpr --expt-extended-lambda -O2 -g -DNDEBUG -std=c++17 -Xcompiler=-fPIC --expt-relaxed-constexpr -DENABLE_FP8 --threads=1 -DENABLE_SCALED_MM_C2X=1 -D_GLIBCXX_USE_CXX11_ABI=1 -gencode arch=compute_86,code=sm_86 -MD -MT CMakeFiles/_C.dir/csrc/quantization/cutlass_w8a8/scaled_mm_c2x.cu.o -MF CMakeFiles/_C.dir/csrc/quantization/cutlass_w8a8/scaled_mm_c2x.cu.o.d -x cu -c /opt/vllm/csrc/quantization/cutlass_w8a8/scaled_mm_c2x.cu -o CMakeFiles/_C.dir/csrc/quantization/cutlass_w8a8/scaled_mm_c2x.cu.o
  Killed
  [39/104] Building CUDA object /opt/vllm/.deps/vllm-flash-attn-build/CMakeFiles/vllm_flash_attn_c.dir/csrc/flash_attn/src/flash_fwd_hdim160_bf16_causal_sm80.cu.o
[... cut ...]
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag llama-factory:r36.3.0-vllm --file /opt/jetson-containers/packages/llm/vllm/Dockerfile --build-arg BASE_IMAGE=llama-factory:r36.3.0-torchaudio --build-arg CUDAARCHS="87" --build-arg VLLM_VERSION="0.6.3" /opt/jetson-containers/packages/llm/vllm 2>&1 | tee /opt/jetson-containers/logs/20241011_132809/build/llama-factory_r36.3.0-vllm.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

```

johnnynunez, 2024-10-11:
> I solved this issue by increasing the virtual memory: ![image](https://private-user-images.githubusercontent.com/44857857/375673046-0013b0d5-ce7d-472a-a3d4-6e905522f569.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg2MzY1NTQsIm5iZiI6MTcyODYzNjI1NCwicGF0aCI6Ii80NDg1Nzg1Ny8zNzU2NzMwNDYtMDAxM2IwZDUtY2U3ZC00NzJhLWEzZDQtNmU5MDU1MjJmNTY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDExVDA4NDQxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViMTAyYTkyMTE3MWRhODYzNTEwZmExMjU1NDRjMTUxYWIzYjViMGI5ZGQ2ZDMwZGZiOWMwZmM5MWRlNWNjY2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yffHTq5wjyuNLjlqMsByg6knNQC7OW0DQ0iyKnXk3Ns)

yes vllm, take a lot of memory. They are optimizing it

JIANGTUNAN, 2024-10-11:
I seem to have encountered a version compatibility issue with PyTorch. The log indicates:  
`AttributeError: module 'torch.library' has no attribute 'custom_op'`  

Looking through the previous logs:  
```shell
testing PyTorch...  
PyTorch version: 2.2.0  
CUDA available: True  
cuDNN version: 8904  
PyTorch built with:  
  - GCC 11.4  
  - C++ Version: 201703  
  - OpenMP 201511 (a.k.a. OpenMP 4.5)  
  - LAPACK is enabled (usually provided by MKL)  
  - NNPACK is enabled  
  - CPU capability usage: NO AVX  
  - CUDA Runtime 12.2  
  - NVCC architecture flags: -gencode;arch=compute_87,code=sm_87  
  - CuDNN 8.9.4  
  - Build settings: BLAS_INFO=open, BUILD_TYPE=Release, CUDA_VERSION=12.2, CUDNN_VERSION=8.9.4, CXX_COMPILER=/usr/bin/c++, CXX_FLAGS= -D_GLIBCXX_USE_CXX11_ABI=1 -fvisibility-inlines-hidden -DUSE_PTHREADPOOL -DN  
``` 

Complete log:
```shell
testing vLLM...
/usr/local/lib/python3.10/dist-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 10-11 08:59:41 importing.py:10] Triton not installed; certain GPU-related functions will not be available.
INFO 10-11 08:59:56 config.py:1654] Downcasting torch.float32 to torch.float16.
WARNING 10-11 09:00:08 config.py:295] gguf quantization is not fully optimized yet. The speed can be slower than non-quantized models.
INFO 10-11 09:00:08 llm_engine.py:237] Initializing an LLM engine (v0.1.dev1+gcbc2ef5.d20241011) with config: model='/data/models/huggingface/models--TheBloke--TinyLlama-1.1B-Chat-v1.0-GGUF/snapshots/52e7645ba7c309695bec7ac98f4f005b139cf465/tinyllama-1.1b-chat-v1.0.Q4_0.gguf', speculative_config=None, tokenizer='TinyLlama/TinyLlama-1.1B-Chat-v1.0', skip_tokenizer_init=False, tokenizer_mode=auto, revision=None, override_neuron_config=None, rope_scaling=None, rope_theta=None, tokenizer_revision=None, trust_remote_code=False, dtype=torch.float16, max_seq_len=2048, download_dir=None, load_format=LoadFormat.GGUF, tensor_parallel_size=1, pipeline_parallel_size=1, disable_custom_all_reduce=False, quantization=gguf, enforce_eager=False, kv_cache_dtype=auto, quantization_param_path=None, device_config=cuda, decoding_config=DecodingConfig(guided_decoding_backend='outlines'), observability_config=ObservabilityConfig(otlp_traces_endpoint=None, collect_model_forward_time=False, collect_model_execute_time=False), seed=0, served_model_name=/data/models/huggingface/models--TheBloke--TinyLlama-1.1B-Chat-v1.0-GGUF/snapshots/52e7645ba7c309695bec7ac98f4f005b139cf465/tinyllama-1.1b-chat-v1.0.Q4_0.gguf, use_v2_block_manager=True, num_scheduler_steps=1, chunked_prefill_enabled=False multi_step_stream_outputs=True, enable_prefix_caching=False, use_async_output_proc=True, use_cached_outputs=False, mm_processor_kwargs=None)
Traceback (most recent call last):
  File "/test/test.py", line 40, in <module>
    run_gguf_inference(model)
  File "/test/test.py", line 24, in run_gguf_inference
    llm = LLM(model=model_path,
  File "/usr/local/lib/python3.10/dist-packages/vllm/entrypoints/llm.py", line 177, in __init__
    self.llm_engine = LLMEngine.from_engine_args(
  File "/usr/local/lib/python3.10/dist-packages/vllm/engine/llm_engine.py", line 574, in from_engine_args
    engine = cls(
  File "/usr/local/lib/python3.10/dist-packages/vllm/engine/llm_engine.py", line 335, in __init__
    self.model_executor = executor_class(
  File "/usr/local/lib/python3.10/dist-packages/vllm/executor/executor_base.py", line 47, in __init__
    self._init_executor()
  File "/usr/local/lib/python3.10/dist-packages/vllm/executor/gpu_executor.py", line 38, in _init_executor
    self.driver_worker = self._create_worker()
  File "/usr/local/lib/python3.10/dist-packages/vllm/executor/gpu_executor.py", line 105, in _create_worker
    return create_worker(**self._get_create_worker_kwargs(
  File "/usr/local/lib/python3.10/dist-packages/vllm/executor/gpu_executor.py", line 24, in create_worker
    wrapper.init_worker(**kwargs)
  File "/usr/local/lib/python3.10/dist-packages/vllm/worker/worker_base.py", line 449, in init_worker
    self.worker = worker_class(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/vllm/worker/worker.py", line 99, in __init__
    self.model_runner: GPUModelRunnerBase = ModelRunnerClass(
  File "/usr/local/lib/python3.10/dist-packages/vllm/worker/model_runner.py", line 1008, in __init__
    self.attn_backend = get_attn_backend(
  File "/usr/local/lib/python3.10/dist-packages/vllm/attention/selector.py", line 108, in get_attn_backend
    backend = which_attn_to_use(num_heads, head_size, num_kv_heads,
  File "/usr/local/lib/python3.10/dist-packages/vllm/attention/selector.py", line 248, in which_attn_to_use
    from vllm.attention.backends.flash_attn import (  # noqa: F401
  File "/usr/local/lib/python3.10/dist-packages/vllm/attention/backends/flash_attn.py", line 607, in <module>
    @torch.library.custom_op("vllm::unified_flash_attention",
AttributeError: module 'torch.library' has no attribute 'custom_op'
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/opt/jetson-containers/jetson_containers/build.py", line 112, in <module>
[... cut ...]
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --runtime=nvidia --network=host --volume /opt/jetson-containers/packages/llm/vllm:/test --volume /opt/jetson-containers/data:/data --workdir /test llama-factory:r36.3.0-vllm /bin/bash -c 'python3 test.py' 2>&1 | tee /opt/jetson-containers/logs/20241011_165801/test/llama-factory_r36.3.0-vllm_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

johnnynunez, 2024-10-11:
> I seem to have encountered a version compatibility issue with PyTorch. The log indicates: `AttributeError: module 'torch.library' has no attribute 'custom_op'`
>
> Looking through the previous logs: [...]

you have to wait for fix: https://github.com/dusty-nv/jetson-containers/pull/670

johnnynunez, 2024-10-11:
Hello,
@dusty-nv @JIANGTUNAN I've fixed doing a PR on llama-factory repository. Now it support 0.6.3 vllm version that it is compatible with Jetson AGX Orin.

johnnynunez, 2024-10-11:
it is working:
[image]

JIANGTUNAN, 2024-10-12:
> Hello, @dusty-nv @JIANGTUNAN I've fixed doing a PR on llama-factory repository. Now it support 0.6.3 vllm version that it is compatible with Jetson AGX Orin.
> 
> #670

Cool! What an incredible process!