dusty-nv/jetson-containers #676: dustynv/ollama:r36.3.0 doesn't start the API

bogdanr, 2024-10-11:
I have an Orin Nano with Jetpack 6.1, CUDA 12.6.68

I am starting the container with this compose.yaml:
```
services:
  ollama:
    runtime: nvidia
    volumes:
      - ./ollama:/root/.ollama
      - ./data:/data
    container_name: ollama
    pull_policy: always
    tty: true
    restart: unless-stopped
    image: "dustynv/ollama:r36.3.0"
```

And it's producing this error when trying to run ollama serve:
```
2024/10/11 14:28:43 routes.go:1153: INFO server config env="map[CUDA_VISIBLE_DEVICES: GPU_DEVICE_ORDINAL: HIP_VISIBLE_DEVICES: HSA_OVERRIDE_GFX_VERSION: HTTPS_PROXY: HTTP_PROXY: NO_PROXY: OLLAMA_DEBUG:false OLLAMA_FLASH_ATTENTION:false OLLAMA_GPU_OVERHEAD:0 OLLAMA_HOST:http://0.0.0.0:11434 OLLAMA_INTEL_GPU:false OLLAMA_KEEP_ALIVE:5m0s OLLAMA_LLM_LIBRARY: OLLAMA_LOAD_TIMEOUT:5m0s OLLAMA_MAX_LOADED_MODELS:0 OLLAMA_MAX_QUEUE:512 OLLAMA_MODELS:/data/models/ollama/models OLLAMA_NOHISTORY:false OLLAMA_NOPRUNE:false OLLAMA_NUM_PARALLEL:0 OLLAMA_ORIGINS:[http://localhost https://localhost http://localhost:* https://localhost:* http://127.0.0.1 https://127.0.0.1 http://127.0.0.1:* https://127.0.0.1:* http://0.0.0.0 https://0.0.0.0 http://0.0.0.0:* https://0.0.0.0:* app://* file://* tauri://*] OLLAMA_SCHED_SPREAD:false OLLAMA_TMPDIR: ROCR_VISIBLE_DEVICES: http_proxy: https_proxy: no_proxy:]"
time=2024-10-11T14:28:43.439Z level=INFO source=images.go:753 msg="total blobs: 5"
time=2024-10-11T14:28:43.441Z level=INFO source=images.go:760 msg="total unused blobs removed: 0"
time=2024-10-11T14:28:43.441Z level=INFO source=routes.go:1200 msg="Listening on [::]:11434 (version 0.0.0)"
time=2024-10-11T14:28:43.455Z level=INFO source=common.go:135 msg="extracting embedded files" dir=/tmp/ollama2917682647/runners
time=2024-10-11T14:28:45.737Z level=INFO source=common.go:49 msg="Dynamic LLM libraries" runners="[cpu cuda_v12]"
time=2024-10-11T14:28:45.737Z level=INFO source=gpu.go:199 msg="looking for compatible GPUs"
time=2024-10-11T14:28:45.738Z level=WARN source=gpu.go:669 msg="unable to locate gpu dependency libraries"
time=2024-10-11T14:28:45.738Z level=WARN source=gpu.go:669 msg="unable to locate gpu dependency libraries"
time=2024-10-11T14:28:45.738Z level=WARN source=gpu.go:669 msg="unable to locate gpu dependency libraries"
double free or corruption (out)
SIGABRT: abort
PC=0xffffaf6ef200 m=7 sigcode=18446744073709551610
signal arrived during cgo execution

goroutine 1 gp=0x40000021c0 m=7 mp=0x4000586008 [syscall]:
runtime.cgocall(0xe4a770, 0x4000046668)
	runtime/cgocall.go:157 +0x44 fp=0x4000046630 sp=0x40000465f0 pc=0x422c34
github.com/ollama/ollama/gpu._Cfunc_nvcuda_init(0xffff50000cb0, 0x4000229480)
	_cgo_gotypes.go:545 +0x30 fp=0x4000046660 sp=0x4000046630 pc=0x7cfbd0
github.com/ollama/ollama/gpu.LoadNVCUDAMgmt.func2(0xffff50000cb0, 0x4000229480)
	github.com/ollama/ollama/gpu/gpu.go:552 +0x50 fp=0x4000046690 sp=0x4000046660 pc=0x7d4b50
github.com/ollama/ollama/gpu.LoadNVCUDAMgmt({0x40001a71e0, 0x2, 0x47a5780?})
	github.com/ollama/ollama/gpu/gpu.go:552 +0x10c fp=0x4000046850 sp=0x4000046690 pc=0x7d462c
github.com/ollama/ollama/gpu.initCudaHandles()
	github.com/ollama/ollama/gpu/gpu.go:116 +0x374 fp=0x4000046a20 sp=0x4000046850 pc=0x7d0494
github.com/ollama/ollama/gpu.GetGPUInfo()
	github.com/ollama/ollama/gpu/gpu.go:231 +0x368 fp=0x4000047b80 sp=0x4000046a20 pc=0x7d0e88
github.com/ollama/ollama/server.Serve({0x408e3d8, 0x40004ba060})
	github.com/ollama/ollama/server/routes.go:1233 +0x520 fp=0x4000047d00 sp=0x4000047b80 pc=0xe21110
github.com/ollama/ollama/cmd.RunServer(0x4000251400?, {0x4905a20?, 0x4?, 0x116f91b?})
	github.com/ollama/ollama/cmd/cmd.go:1167 +0x44 fp=0x4000047d40 sp=0x4000047d00 pc=0xe41cc4
github.com/spf13/cobra.(*Command).execute(0x40000a7808, {0x4905a20, 0x0, 0x0})
	github.com/spf13/cobra@v1.7.0/command.go:940 +0x654 fp=0x4000047e60 sp=0x4000047d40 pc=0x5d7344
github.com/spf13/cobra.(*Command).ExecuteC(0x40000a6908)
	github.com/spf13/cobra@v1.7.0/command.go:1068 +0x320 fp=0x4000047f20 sp=0x4000047e60 pc=0x5d7ab0
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.7.0/command.go:992
github.com/spf13/cobra.(*Command).ExecuteContext(...)
	github.com/spf13/cobra@v1.7.0/command.go:985
main.main()
	github.com/ollama/ollama/main.go:12 +0x54 fp=0x4000047f40 sp=0x4000047f20 pc=0xe495b4
runtime.main()
	runtime/proc.go:271 +0x28c fp=0x4000047fd0 sp=0x4000047f40 pc=0x459a9c
runtime.goexit({})
	runtime/asm_arm64.s:1222 +0x4 fp=0x4000047fd0 sp=0x4000047fd0 pc=0x48e864

goroutine 2 gp=0x4000002c40 m=nil [force gc (idle)]:
runtime.gopark(0x0?, 0x0?, 0x0?, 0x0?, 0x0?)
	runtime/proc.go:402 +0xc8 fp=0x4000060f90 sp=0x4000060f70 pc=0x459ec8
runtime.goparkunlock(...)
[... 184 lines cut ...]
r10     0xa
r11     0x0
r12     0x6974707572726f63
r13     0x2974756f28206e6f
r14     0x0
r15     0xffff500008e0
r16     0x1
r17     0xffffaf6fdbd4
r18     0x0
r19     0x1d
r20     0xffff61fbf0e0
r21     0x6
r22     0xffff61fbd460
r23     0x1
r24     0xffffaf7bf7b0
r25     0x20
r26     0x1
r27     0xffff5003a870
r28     0xffff61fbd758
r29     0xffff61fbd1f0
lr      0xffffaf6ef1ec
sp      0xffff61fbd1f0
pc      0xffffaf6ef200
fault   0x0
```

johnnynunez, 2024-10-11:
[image]

bogdanr, 2024-10-13:
dustynv/ollama:r36.4.0 work just fine!