dusty-nv/jetson-containers #681: Bitsdandbytes && openai conflicts

johnnynunez, 2024-10-13:
When I tried to build containers with bitsandbytes and triton, I get conflicts what cannot import in bistsandbytes operations

johnnynunez, 2024-10-21:
Thank you @dusty-nv now is solved!
Super

closed 2024-10-21 by johnnynunez

je1lee, 2024-10-29:
@johnnynunez

could you give me more information how it is solved??

below is my scripts while importing bitsandbytes in dustynv's base image(**dustynv/openai-triton:r36.3.0**)
```
python3

import bitsandbytes
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/bitsandbytes/__init__.py", line 15, in <module>
    from .nn import modules
  File "/usr/local/lib/python3.10/dist-packages/bitsandbytes/nn/__init__.py", line 21, in <module>
    from .triton_based_modules import (
  File "/usr/local/lib/python3.10/dist-packages/bitsandbytes/nn/triton_based_modules.py", line 7, in <module>
    from bitsandbytes.triton.int8_matmul_mixed_dequantize import (
  File "/usr/local/lib/python3.10/dist-packages/bitsandbytes/triton/int8_matmul_mixed_dequantize.py", line 12, in <module>
    from triton.ops.matmul_perf_model import early_config_prune, estimate_matmul_time
ModuleNotFoundError: No module named 'triton.ops'
```

johnnynunez, 2024-10-29:
> @johnnynunez
>
> could you give me more information how it is solved??
>
> below is my scripts while importing bitsandbytes in dustynv's base image(**dustynv/openai-triton:r36.3.0**)
>
> ```
> python3
>
> import [...]

Did you update jetson-containers? I mean, now there are more recent versions