dusty-nv/jetson-containers #687: NanoLLM - WebRTC - coredump - 'sinkpad' should not be nullptr

TangmereCottage, 2024-10-15:
@dusty-nv thanks for NanoLLM for CUDA=12.6 - works well!! 

However, when I invoke it with: 

```
sudo jetson-containers run $(autotag nano_llm) \
  python3 -m nano_llm.agents.video_query --api=mlc \
    --model Efficient-Large-Model/VILA1.5-3b \
    --max-context-len 256 \
    --max-new-tokens 32 \
    --video-input /dev/video0 \
    --video-output webrtc://@:8554/output
```
and then go to `https://192.168.X.X:8554` to view the output, `nano_llm` core dumps due to `'sinkpad' should not be nullptr`. I'm invoking this from the Orin - the same thing happens for `https://127.0.0.1:8554` etc.

This is a known issue (see related discussion in `jetson-inference` https://github.com/dusty-nv/jetson-inference/issues/1619) and so I've disabled `#enable-webrtc-hide-local-ips-with-mdns` in `chrome://flags`. (did not help). The other suggestion seems to have to do with the STUN server (whatever that is).

Anyone out there - any success with WebRTC from `jetson-containers` or `jetson-utils` or `jetson-inference`? If so, how did you get it to work?

TangmereCottage, 2024-10-15:
Ok, so according to https://patchwork.libcamera.org/patch/14939/ (from 2021) "gstreamer: Avoid call to deprecated gst_element_get_request_pad()". `gst_element_get_request_pad()` is the one that is failing. The current version of gst replaces `gst_element_get_request_pad()` with `gst_element_request_pad_simple()`. Will dig a bit more... Would be good to have the WebRTC be solid.

TangmereCottage, 2024-10-15:
Please see upstream PR. https://github.com/dusty-nv/jetson-utils/pull/219/files
When recompiling utils on CUDA=12.6 and the above change, everything seems to work

TangmereCottage, 2024-10-16:
Update - changing to `gst_element_request_pad_simple` is not the fix. 

1/ When I run `video-viewer /dev/video webrtc://@:8554/output` on the Orin, not in a docker, then everything works and I can see the video feed with Chromium at https://192.168.X.X:8554

2/ When I run`video-viewer /dev/video webrtc://@:8554/output` inside a Jetson docker, and then try to access the the video data from the host, then I get the same coredump as before.

TangmereCottage, 2024-10-22:
No progress so far - I'll test on an older 6.0 Orin to confirm if it's a 6.0->6.1 issue. @dusty-nv - how can I be most useful in debugging this, or is it something dumb I'm doing?

TangmereCottage, 2024-10-31:
Update - part of `video-viewer` works inside CUDA=12.6 containers. Working outputs are:
* display://0
* rtp://remote_ip:port

Appear to be broken:
* webrtc: `sink pad = nullptr
* rstp: `streaming stopped, reason not-linked (-1)`

So if you desperately need to get video data out at reasonable frame rates, then rtp works....

sped-evops, 2025-02-04:
Having same issue reported on dGPU CUDA 12.6 as well as jetson nano board. video-viewer crashes with sink_pad nullptr. Tried all above methods, but still in vain.

sped-evops, 2025-02-04:
@dusty-nv need to use webrtc but getting stuck here.

piter-24, 2025-02-04:
@dusty-nv I am having the same issue
**
ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Bail out! ERROR:/opt/jetson-utils/codec/gstEncoder.cpp:876:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be nullptr
Fatal Python error: Aborted
**

piter-24, 2025-02-05:
```
[network] getHostByName() unable to resolve host '0287ad3d-b146-4d71-9d3e-ded491cc8822.local'
[webrtc] couldn't resolve 0287ad3d-b146-4d71-9d3e-ded491cc8822.local from SDP candidate string

```
> For those who have similar issues: I found that installing avahi-daemon fixed the problem.

@sped-evops Thanks! Now I am running in to that error, how did you run avahi-daemon ? I am getting an error when I try to start it   

`root@ubuntu:/#     service avahi-daemon start
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon                                                                                                          [fail]
`

piter-24, 2025-02-05:
Inside the container I needed to stop avahi-daemon because is already running in host causing port conflict
the commands I ran were:

`    apt install -y gstreamer1.0-nice
    apt-get install -y avahi-utils libnss-mdns 
    service avahi-daemon stop`

closed 2025-02-21 by johnnynunez