dusty-nv/jetson-containers #688: NanoLLM(VLM) + ROS2 (Iron)

Fibo27, 2024-10-16:
@dusty-nv I started this new thread just in case it helps others too (apologies for the lengthy post).

1. dustynv/nano_llm:r36.4.0  : Thank you for that. I have tested it out.
2. I encountered the same issues as @TangmereCottage in https://github.com/dusty-nv/jetson-containers/issues/687. However changing the output to DISPLAY works. The docker therefore needs to be run ensuring that Display is allowed.

**ROS related** 

1. I am an advanced user of ROS and have a set-up that includes a remote Robot with various sensors (Camera, LiDAR, RADAR, IMU). I use an Orin-Nano running Jetpack 5 with ROS2 Iron where the base image is yours which i have modified to my requirement.
2. This Robot is remotely managed and I use VsCode with SSH tunneling and remote Devcontainer. VSCode has a ROS extension which makes building and managing a remote Devcontainer very convenient.
3. My main set-up is now on AGX Orin running Jetpack 6.1 (i have tried x86, then Orin-NX before). Given the better compute power and all the cool stuff that you have been doing, this set-up is the best for controlling the remote Robot.
4. **VLM Package**: The Robot streams camera video to the AGX Orin using its rtp address and the AGX uses the NanoLLM (VLM) package to display the outcome.
5. During the build process of '_CUDA_VERSION=12.6 jetson-containers build nano_llm:iron_', an intermediate ROS2 image was built (ROS IMAGE1). 
6. This intermediate image is loaded with advanced packages of ROS (SLAM, NAV2) as I modify your **Dockerfile.ros** in the package prior to the build.
7. I tried to build another ROS2 image using your '_dustynv/nano_llm:r36.4.0_' and my modified Dockerfile.ros. However, I encountered a number of issues especially while building the NAV2 packages. Eventually I built an image (ROS IMAGE2) without the NAV2 packages.
8. Eventually I used ROS IMAGE1 and ROS IMAGE2 to build a ROSCombo image.
9. The combo image now satisfies my requirement.

**My Query**

- I have one issue and it relates to using VSCode. Given the advantages of using the ROS extension in VSCode, i create a Devcontainer with Docker Compose and use your install.sh and build.sh in the process. While all other packages work well, the VLM based ROS2 package encounters the same error as mentioned in this thread https://forums.developer.nvidia.com/t/libnvdla-compiler-so-cannot-open-shared-object-file-no-such-file-or-directory/240750 ie '_libnvdla_compiler.so: cannot open shared object file'_. Note not using VSCode and creating another image using docker run command using the cli works fine. So i guess it must be related to how the VSCode container is created. Any help from anyone is welcome.

Fibo27, 2024-10-16:
I guess I found my answer given that this has been an issue for many others, I just had to set the runtime correctly in docker daemon.json!

Fibo27, 2024-10-16:
> @Fibo27 could you elaborate on the "The docker therefore needs to be run ensuring that Display is allowed." Currently `run.sh` sets: `-e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix -v /tmp/.docker.xauth:/tmp/.docker.xauth -e XAUTHORITY=/tmp/.docker.xauth`. Which part of that needs to be changed? Thanks!

See the file run.sh lines 228 through 238. You have most of them correct. I do have  DISPLAY=$DISPLAY though

And then you change output to video_output='display://0 in the code

TangmereCottage, 2024-10-16:
@Fibo27 Thanks, something like `--video_output display://0` gets you a local X11 window, originating from the docker container, that gives you the live image+caption. That's super useful, but i was specifically trying to get the WebRTC option to work (`--video-output webrtc://@:8554/output`) so that I can stream video+captions from inside the docker to a browser outside of the docker.

Fibo27, 2024-10-16:
@TangmereCottage the webrtc option is not working as you pointed out - i get the same errors as you. It though works fine with 36.2.0, i have tested that out as well on an Orin-Nx that is still on Jetpack 6.0 - so i guess something is broken with the upgrade to 6.1

closed 2024-10-17 by Fibo27

Shahar-Nadiv, 2026-02-21:
For your remote management setup, are you finding SSH tunneling via VSCode to be stable enough for production? We’re building  Ajime (https://ajime.io/) to provide native, secure cloud connectivity for remote fleets without the manual tunnel setup. It handles the deployment and health monitoring over the cloud so you can manage every jetson version remotely as if it were on your local desk.