dusty-nv/jetson-containers #691: Add support for building this repo on x86

gabriel-st-angel-aerospace, 2024-10-17:
We have totally dockerized our rootfs build for our NVIDIA Orins.  For example, instead of using the stock sample rootfs we are building our rootfs from source in Docker and installing nvidia-jetpack such that when we build our firmware, all dependanices are already installed.

It seems like an oversight that building this on x86 is not supported.  It seems like it'd be possible to do with QEMU and an environment variable specifying what the target is (orin, tx2, etc...)

Any plans to support this in the future?

johnnynunez, 2024-10-17:
> We have totally dockerized our rootfs build for our NVIDIA Orins. For example, instead of using the stock sample rootfs we are building our rootfs from source in Docker and installing nvidia-jetpack [...]

Do you talking about cross compilation?

dusty-nv, 2024-10-18:
A number of the ML packages actually use CUDA during the build/install process so it needs GPU enabled (and typically of the same SM...that being from the same generation of Jetson), and also there are tests between every build stage to ensure their GPU acceleration is functioning correctly.  And my understanding with cross-complication is that it often requires project-specific support, and we build a large number of libraries many of which have quirks/patches in their install or web of dependencies.  Also heard that QEMU arm64 emulation is slow to the point that AGX Orin is several times faster than a 64-core Threadripper.

All that aside though, if you have a specific application in mind you can just install the wheels this repo builds in your 'deployment container', or take our dockerfile recipes and condense them down.  Totally agree this repo can use more features on the deployment side, but am most occupied by enabling the latest releases.  In fact if you just ran this in QEMU, disabled the tests and the L4T checks, it may just install the wheels from our pip server and be quick. Would also like to support x86 for the actual containers themselves, its just a bandwidth issue like above.

closed 2024-10-23 by gabriel-st-angel-aerospace