dusty-nv/jetson-containers #695: Error building Tensorflow - `Couldn't find package`

chrmel, 2024-10-23:
Hello,

first of all, thank you for your work here. It really helps us out!

### Problem
Unfortunaltely, if I want to build the jetson container for `tensorflow` or `tensorflow2` I get an error that the package cannot be found. I am working on the current (at the time of writing 53eb0e69) jetson-containers master. I had no problems on an older version (685d9329).

### Details
Setup
```shell
git clone https://github.com/dusty-nv/jetson-containers.git
cd jetson-containers
bash install.sh
```
Build attempt (example with `tensorflow` but the same problem occurs with `tensorflow2`)
```shell
$ ./jetson-containers build tensorflow
Namespace(base='', build_args='', build_flags='', list_packages=False, logs='', multiple=False, name='', no_github_api=False, package_dirs=[''], packages=['tensorflow'], push='', show_packages=False, simulate=False, skip_errors=False, skip_packages=[''], skip_tests=[''], test_only=[''], verbose=False)
-- L4T_VERSION=35.4.1
-- JETPACK_VERSION=5.1.2
-- CUDA_VERSION=11.4
-- PYTHON_VERSION=3.8
-- LSB_RELEASE=20.04 (focal)
-- Package l4t-ml has missing dependencies, disabling...  ("couldn't find package:  tensorflow2")
-- Package l4t-tensorflow:tf1 has missing dependencies, disabling...  ("couldn't find package:  tensorflow")
-- Package l4t-tensorflow:tf2 has missing dependencies, disabling...  ("couldn't find package:  tensorflow2")
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/jetson/tmp/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/home/jetson/tmp/jetson-containers/jetson_containers/container.py", line 66, in build_container
    packages = resolve_dependencies(packages)
  File "/home/jetson/tmp/jetson-containers/jetson_containers/packages.py", line 316, in resolve_dependencies
    packages, changed = add_depends(packages)
  File "/home/jetson/tmp/jetson-containers/jetson_containers/packages.py", line 288, in add_depends
    for dependency in find_package(package).get('depends', []):
  File "/home/jetson/tmp/jetson-containers/jetson_containers/packages.py", line 174, in find_package
    raise KeyError(f"couldn't find package:  {package}")
KeyError: "couldn't find package:  tensorflow"
```

Help to build the tensorflow containers will be much appreciated. Thank you and regards.

johnnynunez, 2024-10-23:
> Hello,
> 
> first of all, thank you for your work here. It really helps us out!
> 
> ### Problem
> Unfortunaltely, if I want to build the jetson container for `tensorflow` or `tensorflow2` I get an error that the package cannot be found. I am working on the current (at the time of writing [53eb0e6](https://github.com/dusty-nv/jetson-containers/commit/53eb0e69d5a9931a21044a52299609bd16f0f6d5)) jetson-containers master. I had no problems on an older version (685d9329).
> 
> ### Details
> Setup
> 
> ```shell
> git clone https://github.com/dusty-nv/jetson-containers.git
> cd jetson-containers
> bash install.sh
> ```
> 
> Build attempt (example with `tensorflow` but the same problem occurs with `tensorflow2`)
> 
> ```shell
> $ ./jetson-containers build tensorflow
> Namespace(base='', build_args='', build_flags='', list_packages=False, logs='', multiple=False, name='', no_github_api=False, package_dirs=[''], packages=['tensorflow'], push='', show_packages=False, simulate=False, skip_errors=False, skip_packages=[''], skip_tests=[''], test_only=[''], verbose=False)
> -- L4T_VERSION=35.4.1
> -- JETPACK_VERSION=5.1.2
> -- CUDA_VERSION=11.4
> -- PYTHON_VERSION=3.8
> -- LSB_RELEASE=20.04 (focal)
> -- Package l4t-ml has missing dependencies, disabling...  ("couldn't find package:  tensorflow2")
> -- Package l4t-tensorflow:tf1 has missing dependencies, disabling...  ("couldn't find package:  tensorflow")
> -- Package l4t-tensorflow:tf2 has missing dependencies, disabling...  ("couldn't find package:  tensorflow2")
> Traceback (most recent call last):
>   File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
>     exec(code, run_globals)
>   File "/home/jetson/tmp/jetson-containers/jetson_containers/build.py", line 112, in <module>
>     build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
>   File "/home/jetson/tmp/jetson-containers/jetson_containers/container.py", line 66, in build_container
>     packages = resolve_dependencies(packages)
>   File "/home/jetson/tmp/jetson-containers/jetson_containers/packages.py", line 316, in resolve_dependencies
>     packages, changed = add_depends(packages)
>   File "/home/jetson/tmp/jetson-containers/jetson_containers/packages.py", line 288, in add_depends
>     for dependency in find_package(package).get('depends', []):
>   File "/home/jetson/tmp/jetson-containers/jetson_containers/packages.py", line 174, in find_package
>     raise KeyError(f"couldn't find package:  {package}")
> KeyError: "couldn't find package:  tensorflow"
> ```
> 
> Help to build the tensorflow containers will be much appreciated. Thank you and regards.

```bash
jetson-containers build tensorflow2
```

chrmel, 2024-10-23:
Unfortunately, as already stated in the description, using the package ‚tensorflow2‘ leads to the exact same behaviour.

chrmel, 2024-10-23:
master pulled today!

johnnynunez, 2024-10-24:
> master pulled today!

[image]

chrmel, 2024-10-25:
I have to say, this is not much of a help. The fact that it seems to work on your setup (with different CUDA and Python versions) does not help me solve my problem. I was hoping to get professional help on how to get it to build on my system, not - it works on my machine... sink or swim.

That said, I would like to try it one more time.

I try to reproduce everything you seem to be doing... still.

```shell
$ date
Fri Oct 25 05:22:35 UTC 2024
```
```shell
$ git pull origin master
```
```shell
$ bash install.sh 
+++ readlink -f install.sh
++ dirname /home/jetson/tmp/jetson-containers-dustynv/install.sh
+ ROOT=/home/jetson/tmp/jetson-containers-dustynv
+ INSTALL_PREFIX=/usr/local/bin
+ pip3 --version
pip 24.2 from /home/jetson/.local/lib/python3.8/site-packages/pip (python 3.8)
+ pip3 install -r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt
Defaulting to user installation because normal site-packages is not writeable
Collecting git+https://github.com/Granulate/DockerHub-API.git (from -r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 4))
  Cloning https://github.com/Granulate/DockerHub-API.git to /tmp/pip-req-build-df7vds8x
  Running command git clone --filter=blob:none --quiet https://github.com/Granulate/DockerHub-API.git /tmp/pip-req-build-df7vds8x
  Resolved https://github.com/Granulate/DockerHub-API.git to commit 27015dbd00c0e0550dd34287db4b70625985edc3
  Preparing metadata (setup.py) ... done
Requirement already satisfied: packaging>=20.0 in /home/jetson/.local/lib/python3.8/site-packages (from -r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 1)) (24.1)
Requirement already satisfied: pyyaml>=6 in /home/jetson/.local/lib/python3.8/site-packages (from -r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 2)) (6.0.2)
Requirement already satisfied: wget in /home/jetson/.local/lib/python3.8/site-packages (from -r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 3)) (3.2)
Requirement already satisfied: furl in /home/jetson/.local/lib/python3.8/site-packages (from DockerHub-API==0.5->-r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 4)) (2.1.3)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from DockerHub-API==0.5->-r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 4)) (2.22.0)
Requirement already satisfied: six>=1.8.0 in /usr/lib/python3/dist-packages (from furl->DockerHub-API==0.5->-r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 4)) (1.14.0)
Requirement already satisfied: orderedmultidict>=1.0.1 in /home/jetson/.local/lib/python3.8/site-packages (from furl->DockerHub-API==0.5->-r /home/jetson/tmp/jetson-containers-dustynv/requirements.txt (line 4)) (1.0.1)
WARNING: Error parsing dependencies of distro-info: Invalid version: '0.23ubuntu1'
WARNING: Error parsing dependencies of python-debian: Invalid version: '0.1.36ubuntu1'
+ sudo ln -sf /home/jetson/tmp/jetson-containers-dustynv/autotag /usr/local/bin/autotag
[sudo] password for jetson: 
+ sudo ln -sf /home/jetson/tmp/jetson-containers-dustynv/jetson-containers /usr/local/bin/jetson-containers
```
I tried building it with your version setup (these are not the version on my jetson - see original post)
```shell
CUDA_VERSION=12.6 PYTHON_VERSION=3.10 ./jetson-containers build tensorflow2
Namespace(base='', build_args='', build_flags='', list_packages=False, logs='', multiple=False, name='', no_github_api=False, package_dirs=[''], packages=['tensorflow2'], push='', show_packages=False, simulate=False, skip_errors=False, skip_packages=[''], skip_tests=[''], test_only=[''], verbose=False)
-- L4T_VERSION=35.4.1
-- JETPACK_VERSION=5.1.2
-- CUDA_VERSION=12.6
-- PYTHON_VERSION=3.10
-- LSB_RELEASE=20.04 (focal)
-- Package raft has missing dependencies, disabling...  ("couldn't find package:  cuda-python")
-- Package cuda-python:11.4 has missing dependencies, disabling...  ("couldn't find package:  cuda:11.4")
-- Package local_llm has missing dependencies, disabling...  ("couldn't find package:  cuda-python")
-- Package nano_llm:main has missing dependencies, disabling...  ("couldn't find package:  cuda-python")
-- Package nano_llm:main-foxy has missing dependencies, disabling...  ("couldn't find package:  nano_llm:main")
-- Package nano_llm:main-galactic has missing dependencies, disabling...  ("couldn't find package:  nano_llm:main")
-- Package nano_llm:main-humble has missing dependencies, disabling...  ("couldn't find package:  nano_llm:main")
-- Package nano_llm:main-iron has missing dependencies, disabling...  ("couldn't find package:  nano_llm:main")
-- Package nano_llm:24.4 has missing dependencies, disabling...  ("couldn't find package:  cuda-python")
-- Package nano_llm:24.4-foxy has missing dependencies, disabling...  ("couldn't find package:  nano_llm:24.4")
-- Package nano_llm:24.4-galactic has missing dependencies, disabling...  ("couldn't find package:  nano_llm:24.4")
-- Package nano_llm:24.4-humble has missing dependencies, disabling...  ("couldn't find package:  nano_llm:24.4")
-- Package nano_llm:24.4-iron has missing dependencies, disabling...  ("couldn't find package:  nano_llm:24.4")
-- Package nano_llm:24.4.1 has missing dependencies, disabling...  ("couldn't find package:  cuda-python")
-- Package nano_llm:24.4.1-foxy has missing dependencies, disabling...  ("couldn't find package:  nano_llm:24.4.1")
[... cut ...]
  File "/home/jetson/tmp/jetson-containers-dustynv/jetson_containers/packages.py", line 290, in add_depends
    for dependency in find_package(package).get('depends', []):
  File "/home/jetson/tmp/jetson-containers-dustynv/jetson_containers/packages.py", line 174, in find_package
    raise KeyError(f"couldn't find package:  {package}")
KeyError: "couldn't find package:  tensorflow"
```
May it be that my jetson is running 20.04, you seem to be running 22.04? I am genuinly looking for a solution.

Thank you!

johnnynunez, 2024-10-25:
Aaah tensorflow2 right now it is only for jetpack 6

chrmel, 2024-10-28:
Ok, thanks.  
How is it then that I was able to build tensorflow for JP5.1 with a [previous commit](https://github.com/dusty-nv/jetson-containers/commit/5dece56e74b6cc14ced9557cc103affcdde728c1) and you are able to [provide](https://github.com/dusty-nv/jetson-containers/tree/master/packages/ml/tensorflow) [dustynv/tensorflow:r35.2.1](https://hub.docker.com/r/dustynv/tensorflow/tags) and [dustynv/tensorflow2:r35.2.1](https://hub.docker.com/r/dustynv/tensorflow2/tags) when [JP5.1 has CUDA 11.4.19](https://developer.nvidia.com/embedded/jetpack-sdk-51)?

johnnynunez, 2024-10-28:
> Ok, thanks. How is it then that I was able to build tensorflow for JP5.1 with a [previous commit](https://github.com/dusty-nv/jetson-containers/commit/5dece56e74b6cc14ced9557cc103affcdde728c1) and [...]

because they are old versions of tensorflow:
https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html

chrmel, 2024-10-29:
> because they are old versions of tensorflow: https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html
> 
> Now tensorflow and JAX use XLA backend by default. https://github.com/openxla/xla XLA has Hermetic CUDA, so only from +2.18.0 support 11.8 cuda version and above

Ok, thank you! That clarifies things.

closed 2024-10-29 by chrmel