dusty-nv/jetson-containers #702: ROS2 Jazzy fails to build because of OpenCV

0Unkn0wn, 2024-10-28:
I tried building an image with ROS2 Jazzy and Jax and it seems that OpenCV failed to install which stops me from creating the image. 
I have also tried Humble, and it also fails on the same step if the Python version is set to 3.11. Any idea what happened?

**Command**:
`PYTHON_VERSION=3.11 jetson-containers build --name=ros2jazzy_jax_base ros:jazzy-desktop jax:0.4.32`

<details>
<summary>Build log</summary>

```
jetson@ubuntu ~/r/jetson-containers (master)> PYTHON_VERSION=3.11 jetson-containers build --name=ros2jazzy_jax_base ros:jazzy-desktop jax:0.4.32
Namespace(packages=['ros:jazzy-desktop', 'jax:0.4.32'], name='ros2jazzy_jax_base', base='', multiple=False, build_flags='', build_args='', package_dirs=[''], list_packages=False, show_packages=False, skip_packages=[''], skip_errors=False, skip_tests=[''], test_only=[''], simulate=False, push='', logs='', verbose=False, no_github_api=False)
-- L4T_VERSION=36.4.0
-- JETPACK_VERSION=6.1
-- CUDA_VERSION=12.6
-- PYTHON_VERSION=3.11
-- LSB_RELEASE=22.04 (jammy)
-- Building containers  ['build-essential', 'pip_cache:cu126', 'cuda:12.6', 'cudnn:9.4', 'python', 'tensorrt', 'numpy', 'opencv', 'cmake', 'ros:jazzy-desktop', 'jax:0.4.32']
-- Building container ros2jazzy_jax_base:r36.4.0-cp311-build-essential

DOCKER_BUILDKIT=0 docker build --network=host --tag ros2jazzy_jax_base:r36.4.0-cp311-build-essential \
--file /home/jetson/repos/jetson-containers/packages/build/build-essential/Dockerfile \
--build-arg BASE_IMAGE=ubuntu:22.04 \
/home/jetson/repos/jetson-containers/packages/build/build-essential \
2>&1 | tee /home/jetson/repos/jetson-containers/logs/20241028_012330/build/ros2jazzy_jax_base_r36.4.0-cp311-build-essential.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  19.97kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> 981912c48e9a
Step 3/5 : ENV DEBIAN_FRONTEND=noninteractive     LANGUAGE=en_US:en     LANG=en_US.UTF-8     LC_ALL=en_US.UTF-8
 ---> Using cache
 ---> 978f6db217f5
Step 4/5 : RUN set -ex     && apt-get update     && apt-get install -y --no-install-recommends         locales         locales-all         tzdata     && locale-gen en_US $LANG     && update-locale LC_ALL=$LC_ALL LANG=$LANG     && locale         && apt-get install -y --no-install-recommends         build-essential         software-properties-common         apt-transport-https         ca-certificates         lsb-release         pkg-config         gnupg         git         gdb         wget         curl         nano         zip         unzip         time         sshpass         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*         && gcc --version     && g++ --version
 ---> Using cache
 ---> bdeb5fdb0222
Step 5/5 : COPY tarpack /usr/local/bin/
 ---> Using cache
 ---> d87978b51f05
Successfully built d87978b51f05
Successfully tagged ros2jazzy_jax_base:r36.4.0-cp311-build-essential
-- Building container ros2jazzy_jax_base:r36.4.0-cp311-pip_cache_cu126

DOCKER_BUILDKIT=0 docker build --network=host --tag ros2jazzy_jax_base:r36.4.0-cp311-pip_cache_cu126 \
--file /home/jetson/repos/jetson-containers/packages/cuda/cuda/Dockerfile.pip \
--build-arg BASE_IMAGE=ros2jazzy_jax_base:r36.4.0-cp311-build-essential \
--build-arg TAR_INDEX_URL="http://jetson.webredirect.org:8000/jp6/cu126" \
--build-arg PIP_INDEX_REPO="http://jetson.webredirect.org/jp6/cu126" \
--build-arg PIP_TRUSTED_HOSTS="jetson.webredirect.org" \
--build-arg PIP_UPLOAD_REPO="http://localhost/jp6/cu126" \
--build-arg PIP_UPLOAD_USER="jp6" \
--build-arg PIP_UPLOAD_PASS="none" \
--build-arg SCP_UPLOAD_URL="localhost:/dist/jp6/cu126" \
--build-arg SCP_UPLOAD_USER="None" \
--build-arg SCP_UPLOAD_PASS="None" \
/home/jetson/repos/jetson-containers/packages/cuda/cuda \
2>&1 | tee /home/jetson/repos/jetson-containers/logs/20241028_012330/build/ros2jazzy_jax_base_r36.4.0-cp311-pip_cache_cu126.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  49.66kB
Step 1/4 : ARG BASE_IMAGE
Step 2/4 : FROM ${BASE_IMAGE}
 ---> d87978b51f05
[... 822 lines cut ...]
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|  No running processes found                                                           |
+---------------------------------------------------------------------------------------+
```

</details>

<details>
<summary>Storage info</summary>

```
jetson@ubuntu ~/r/jetson-containers (master)> df -h
Filesystem       Size  Used Avail Use% Mounted on
/dev/nvme0n1p1   915G   35G  834G   5% /
tmpfs            3.8G  136K  3.8G   1% /dev/shm
tmpfs            1.5G   35M  1.5G   3% /run
tmpfs            5.0M  4.0K  5.0M   1% /run/lock
/dev/nvme0n1p10   63M  118K   63M   1% /boot/efi
tmpfs            762M   92K  762M   1% /run/user/128
tmpfs            762M   76K  762M   1% /run/user/1000
```

</details>

0Unkn0wn, 2024-10-28:
Also tried building opencv:4.8.1 to see if it would build and it failed as well.

**Command**:
`PYTHON_VERSION=3.11 jetson-containers build --name=opencv-p311 opencv:4.8.1`

<details>
<summary>Build log</summary>

```
...

  -- General configuration for OpenCV 4.8.1 =====================================
  --   Version control:               4.8.1-dirty
  --
  --   Extra modules:
  --     Location (extra):            /opt/opencv-python/opencv_contrib/modules
  --     Version control (extra):     4.8.1
  --
  --   Platform:
  --     Timestamp:                   2024-10-28T12:21:43Z
  --     Host:                        Linux 5.15.148-tegra aarch64
  --     CMake:                       3.30.5
  --     CMake generator:             Unix Makefiles
  --     CMake build tool:            /usr/bin/gmake
  --     Configuration:               RELEASE
  --
  --   CPU/HW features:
  --     Baseline:                    NEON FP16
  --       required:                  NEON
  --
  --   C/C++:
  --     Built as dynamic libs?:      NO
  --     C++ standard:                11
  --     C++ Compiler:                /usr/bin/c++  (ver 11.4.0)
  --     C++ flags (Release):         -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
  --     C++ flags (Debug):           -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
  --     C Compiler:                  /usr/bin/cc
  --     C flags (Release):           -fsigned-char -W -Wall -Wreturn-type -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
  --     C flags (Debug):             -fsigned-char -W -Wall -Wreturn-type -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
  --     Linker flags (Release):      -Wl,--gc-sections -Wl,--as-needed -Wl,--no-undefined
  --     Linker flags (Debug):        -Wl,--gc-sections -Wl,--as-needed -Wl,--no-undefined
  --     ccache:                      NO
  --     Precompiled headers:         NO
  --     Extra dependencies:          /usr/lib/aarch64-linux-gnu/liblapack.so /usr/lib/aarch64-linux-gnu/libcblas.so /usr/lib/aarch64-linux-gnu/libatlas.so /usr/lib/aarch64-linux-gnu/libjpeg.so /usr/lib/aarch64-linux-gnu/libpng.so /usr/lib/aarch64-linux-gnu/libz.so Iconv::Iconv m pthread cudart_static dl rt nppc nppial nppicc nppidei nppif nppig nppim nppist nppisu nppitc npps cublas cudnn cufft -L/usr/local/cuda/lib64 -L/usr/lib/aarch64-linux-gnu
  --     3rdparty dependencies:       libprotobuf ade ittnotify libwebp libtiff libopenjp2 IlmImf quirc tegra_hal
  --
  --   OpenCV modules:
  --     To be built:                 alphamat aruco bgsegm bioinspired calib3d ccalib core cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev datasets dnn dnn_objdetect dnn_superres dpm face features2d flann fuzzy gapi hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot python3 quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking video videoio videostab wechat_qrcode xfeatures2d ximgproc xobjdetect xphoto
  --     Disabled:                    freetype world
  --     Disabled by dependency:      -
  --     Unavailable:                 cvv hdf java julia matlab ovis python2 sfm ts viz
  --     Applications:                -
  --     Documentation:               NO
  --     Non-free algorithms:         YES
  --
  --   GUI:                           GTK3
  --     GTK+:                        YES (ver 3.24.33)
  --       GThread :                  YES (ver 2.72.4)
  --       GtkGlExt:                  NO
  --     VTK support:                 NO
  --
  --   Media I/O:
  --     ZLib:                        /usr/lib/aarch64-linux-gnu/libz.so (ver 1.2.11)
  --     JPEG:                        /usr/lib/aarch64-linux-gnu/libjpeg.so (ver 80)
  --     WEBP:                        build (ver encoder: 0x020f)
[... cut ...]
subprocess.CalledProcessError: Command 'docker run -t --rm --runtime=nvidia --network=host --volume /home/jetson/repos/jetson-containers/packages/opencv:/test --volume /home/jetson/repos/jetson-containers/data:/data --workdir /test opencv-p311:r36.4.0-cp311-opencv_4.8.1 /bin/bash -c 'python3 test.py' 2>&1 | tee /home/jetson/repos/jetson-containers/logs/20241028_130807/test/opencv-p311_r36.4.0-cp311-opencv_4.8.1_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

</details>

dusty-nv, 2024-10-31:
Hi @0Unkn0wn, thanks for the logs, I just tried `jetson-containers build ros:jazzy-desktop` on JetPack 6.1 and hit some opencv errors in imageproc package (different than yours I think?)

Also going back and trying it with OpenCV 4.8.1 now.  However prior to that OpenCV may encounter errors with the newer versions of CUDA, that had needed fixed in OpenCV.  So there could encounter a situation where it's not possible to build both ROS + OpenCV/CUDA together.  So either ROS would need patched for newer OpenCV, or use the vanilla OpenCV built without CUDA.

0Unkn0wn, 2024-10-31:
Hi @dusty-nv, thank you for the updates. Just wondering, if would it be possible to skip OpenCV for my build with ROS2 Jazzy and Jax? I’m not planning to use it right away, so if there’s a way to leave it out or any workaround you’d suggest, that’d be great. If it works with OpenCV 4.8.1, that would work for me as well. Thanks again!

dusty-nv, 2024-10-31:
Hi @0Unkn0wn, you can try removing opencv from the depends list here:  https://github.com/dusty-nv/jetson-containers/blob/20e7292f0d9745f7badfe638afb68b57a7222bb3/packages/ros/config.py#L10

And then remove opencv from these skip_keys:  https://github.com/dusty-nv/jetson-containers/blob/20e7292f0d9745f7badfe638afb68b57a7222bb3/packages/ros/ros2_build.sh#L109

0Unkn0wn, 2024-11-03:
Hi @dusty-nv I managed to build the container with Jazzy and Jax by removing OpenCV as you have shown me so thank you very much!

For future reference, this is the build that worked: `CUDA_VERSION=12.6 jetson-containers build --skip-tests=all --name=ros_jax ros:jazzy-ros-base jax:0.4.32`

Just one last question, any idea why it only works when I don’t set the Python version to 3.11 or 3.12?

<details>
<summary>Error log</summary>

```
#All required rosdeps installed successfully
Traceback (most recent call last):
  File "/usr/bin/colcon", line 33, in <module>
    sys.exit(load_entry_point('colcon-core==0.18.1', 'console_scripts', 'colcon')())
  File "/usr/lib/python3/dist-packages/colcon_core/command.py", line 128, in main
    return _main(
  File "/usr/lib/python3/dist-packages/colcon_core/command.py", line 186, in _main
    args = parser.parse_args(args=argv)
  File "/usr/lib/python3/dist-packages/colcon_defaults/argument_parser/defaults.py", line 166, in parse_args
    return self._parser.parse_args(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_argcomplete/argument_parser/argcomplete/__init__.py", line 85, in parse_args
    from argcomplete import autocomplete
ModuleNotFoundError: No module named 'argcomplete'
The command '/bin/bash -c /tmp/ros2_build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/jetson/repos/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/home/jetson/repos/jetson-containers/jetson_containers/container.py", line 147, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag ros_jax:r36.4.0-cu126-cp311-ros_jazzy-ros-base --file /home/jetson/repos/jetson-containers/packages/ros/Dockerfile.ros2 --build-arg BASE_IMAGE=ros_jax:r36.4.0-cu126-cp311-cmake --build-arg ROS_VERSION="jazzy" --build-arg ROS_PACKAGE="ros_base" /home/jetson/repos/jetson-containers/packages/ros 2>&1 | tee /home/jetson/repos/jetson-containers/logs/20241103_203530/build/ros_jax_r36.4.0-cu126-cp311-ros_jazzy-ros-base.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```
</details>

martincerven, 2024-11-21:
There is some [API breaking] https://github.com/ros-perception/image_pipeline/issues/1030#issuecomment-2373146016) from ROS's image_proc with higher OpenCV versions. (4.6.0 is default OpenCV shipped with 24.04 )

Also for the 12.6 CUDA support in OpenCV you need to use [latest branch](https://github.com/opencv/opencv/issues/26392), but that breaks some packages in ROS such as image_proc

But for those of us who just want to run Jazzy with GPU support and don't care about CUDA or OpenCV versions this is still amazing! Thanks @dusty-nv @0Unkn0wn

martincerven, 2024-11-21:
I think so. I built both base and desktop. (although I just skipped CUDA OpenCV and used the one that comes with Ubuntu).

closed 2025-02-21 by johnnynunez