dusty-nv/jetson-containers #704: using the vllm container error

huangeng, 2024-10-30:
# device information
```top 4.2.12(c)2024，Raffaello Bonghi [raffaello@rnext.itWebsite: https://rnext,it/jetson stats
Platform
Serial Number: [s|XX CLICK TO READ XXX]
Machine: aarch64
Hardware
System: LinuxModel: NVIDIA Jetson Orin NX Engineering Reference Developer KitDistribution: Ubuntu 22.04 Jammy Jellyfish699-level Part Number: 699-13767-0000-300 P.1
Release: 5.15.148-tegra
P-Number: p3767-0000
Module: NVIDIA Jetson Orin NX(16GB ram)
Python: 3.10.12
Soc: tegra234
Libraries
CUDA ArCh BIN: 8.7
CUDA: 12.6.68
L4T: 36.4.0
CUDNN: 9.3.0.75
Jetpack: 6.1
TensorRT: 10.3.0.30
VPI: 3.2.4
Hostname: jetson-desktop
Vulkan: 1.3.204
Interfaces
opency: 4.8.0 with CuDA:
wIP1p1s0: 192.168.1.64
dockere: 172.17.0.1 
```

# steps
```

1. docker run --runtime nvidia     -p 8000:8000     --ipc=host -it  dustynv/vllm:0.6.3-r36.4.0
2. vllm serve Qwen/Qwen2.5-0.5B-Instruct

```
# error logs

```
root@jetson-desktop:/# vllm serve Qwen/Qwen2.5-0.5B-Instruct
/usr/local/lib/python3.10/dist-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 10-30 17:15:10 api_server.py:528] vLLM API server version 0.6.3
INFO 10-30 17:15:10 api_server.py:529] args: Namespace(subparser='serve', model_tag='Qwen/Qwen2.5-0.5B-Instruct', config='', host=None, port=8000, uvicorn_log_level='info', allow_credentials=False, allowed_origins=['*'], allowed_methods=['*'], allowed_headers=['*'], api_key=None, lora_modules=None, prompt_adapters=None, chat_template=None, response_role='assistant', ssl_keyfile=None, ssl_certfile=None, ssl_ca_certs=None, ssl_cert_reqs=0, root_path=None, middleware=[], return_tokens_as_token_ids=False, disable_frontend_multiprocessing=False, enable_auto_tool_choice=False, tool_call_parser=None, tool_parser_plugin='', model='Qwen/Qwen2.5-0.5B-Instruct', tokenizer=None, skip_tokenizer_init=False, revision=None, code_revision=None, tokenizer_revision=None, tokenizer_mode='auto', trust_remote_code=False, download_dir=None, load_format='auto', config_format='auto', dtype='auto', kv_cache_dtype='auto', quantization_param_path=None, max_model_len=None, guided_decoding_backend='outlines', distributed_executor_backend=None, worker_use_ray=False, pipeline_parallel_size=1, tensor_parallel_size=1, max_parallel_loading_workers=None, ray_workers_use_nsight=False, block_size=16, enable_prefix_caching=False, disable_sliding_window=False, use_v2_block_manager=True, num_lookahead_slots=0, seed=0, swap_space=4, cpu_offload_gb=0, gpu_memory_utilization=0.9, num_gpu_blocks_override=None, max_num_batched_tokens=None, max_num_seqs=256, max_logprobs=20, disable_log_stats=False, quantization=None, rope_scaling=None, rope_theta=None, enforce_eager=False, max_context_len_to_capture=None, max_seq_len_to_capture=8192, disable_custom_all_reduce=False, tokenizer_pool_size=0, tokenizer_pool_type='ray', tokenizer_pool_extra_config=None, limit_mm_per_prompt=None, mm_processor_kwargs=None, enable_lora=False, max_loras=1, max_lora_rank=16, lora_extra_vocab_size=256, lora_dtype='auto', long_lora_scaling_factors=None, max_cpu_loras=None, fully_sharded_loras=False, enable_prompt_adapter=False, max_prompt_adapters=1, max_prompt_adapter_token=0, device='auto', num_scheduler_steps=1, multi_step_stream_outputs=True, scheduler_delay_factor=0.0, enable_chunked_prefill=None, speculative_model=None, speculative_model_quantization=None, num_speculative_tokens=None, speculative_disable_mqa_scorer=False, speculative_draft_tensor_parallel_size=None, speculative_max_model_len=None, speculative_disable_by_batch_size=None, ngram_prompt_lookup_max=None, ngram_prompt_lookup_min=None, spec_decoding_acceptance_method='rejection_sampler', typical_acceptance_sampler_posterior_threshold=None, typical_acceptance_sampler_posterior_alpha=None, disable_logprobs_during_spec_decoding=None, model_loader_extra_config=None, ignore_patterns=[], preemption_mode=None, served_model_name=None, qlora_adapter_name_or_path=None, otlp_traces_endpoint=None, collect_detailed_traces=None, disable_async_output_proc=False, override_neuron_config=None, scheduling_policy='fcfs', disable_log_requests=False, max_log_len=None, disable_fastapi_docs=False, dispatch_function=<function serve at 0xffff37191ab0>)
INFO 10-30 17:15:10 api_server.py:166] Multiprocessing frontend to use ipc:///tmp/7872893e-c189-48f7-a7dc-b43b2a2be836 for IPC Path.
INFO 10-30 17:15:10 api_server.py:179] Started engine process with PID 41
/usr/local/lib/python3.10/dist-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 10-30 17:15:30 llm_engine.py:237] Initializing an LLM engine (v0.6.3) with config: model='Qwen/Qwen2.5-0.5B-Instruct', speculative_config=None, tokenizer='Qwen/Qwen2.5-0.5B-Instruct', skip_tokenizer_init=False, tokenizer_mode=auto, revision=None, override_neuron_config=None, rope_scaling=None, rope_theta=None, tokenizer_revision=None, trust_remote_code=False, dtype=torch.bfloat16, max_seq_len=32768, download_dir=None, load_format=LoadFormat.AUTO, tensor_parallel_size=1, pipeline_parallel_size=1, disable_custom_all_reduce=False, quantization=None, enforce_eager=False, kv_cache_dtype=auto, quantization_param_path=None, device_config=cuda, decoding_config=DecodingConfig(guided_decoding_backend='outlines'), observability_config=ObservabilityConfig(otlp_traces_endpoint=None, collect_model_forward_time=False, collect_model_execute_time=False), seed=0, served_model_name=Qwen/Qwen2.5-0.5B-Instruct, use_v2_block_manager=True, num_scheduler_steps=1, chunked_prefill_enabled=False multi_step_stream_outputs=True, enable_prefix_caching=False, use_async_output_proc=True, use_cached_outputs=True, mm_processor_kwargs=None)
INFO 10-30 17:15:33 model_runner.py:1060] Starting to load model Qwen/Qwen2.5-0.5B-Instruct...
INFO 10-30 17:15:34 weight_utils.py:243] Using model weights format ['*.safetensors']
INFO 10-30 17:15:34 weight_utils.py:288] No model.safetensors.index.json found in remote.
Loading safetensors checkpoint shards:   0% Completed | 0/1 [00:00<?, ?it/s]
Loading safetensors checkpoint shards: 100% Completed | 1/1 [00:00<00:00,  1.27it/s]
Loading safetensors checkpoint shards: 100% Completed | 1/1 [00:00<00:00,  1.27it/s]

INFO 10-30 17:15:35 model_runner.py:1071] Loading model weights took 0.9276 GB
INFO 10-30 17:15:45 gpu_executor.py:122] # GPU blocks: 62004, # CPU blocks: 21845
INFO 10-30 17:15:45 gpu_executor.py:126] Maximum concurrency for 32768 tokens per request: 30.28x
Traceback (most recent call last):
  File "/usr/local/bin/vllm", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/vllm/scripts.py", line 195, in main
    args.dispatch_function(args)
  File "/usr/local/lib/python3.10/dist-packages/vllm/scripts.py", line 41, in serve
    uvloop.run(run_server(args))
  File "/usr/local/lib/python3.10/dist-packages/uvloop/__init__.py", line 82, in run
    return loop.run_until_complete(wrapper())
  File "uvloop/loop.pyx", line 1518, in uvloop.loop.Loop.run_until_complete
  File "/usr/local/lib/python3.10/dist-packages/uvloop/__init__.py", line 61, in wrapper
    return await main
  File "/usr/local/lib/python3.10/dist-packages/vllm/entrypoints/openai/api_server.py", line 552, in run_server
    async with build_async_engine_client(args) as engine_client:
  File "/usr/lib/python3.10/contextlib.py", line 199, in __aenter__
    return await anext(self.gen)
  File "/usr/local/lib/python3.10/dist-packages/vllm/entrypoints/openai/api_server.py", line 107, in build_async_engine_client
    async with build_async_engine_client_from_engine_args(
  File "/usr/lib/python3.10/contextlib.py", line 199, in __aenter__
    return await anext(self.gen)
  File "/usr/local/lib/python3.10/dist-packages/vllm/entrypoints/openai/api_server.py", line 194, in build_async_engine_client_from_engine_args
    raise RuntimeError(
RuntimeError: Engine process failed to start
root@jetson-desktop:/# /usr/lib/python3.10/multiprocessing/resource_tracker.py:224: UserWarning: resource_tracker: There appear to be 1 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '

```

johnnynunez, 2024-10-30:
are you using the latest one?
http://jetson.webredirect.org/jp6/cu126/vllm/0.6.4.dev0+gfd47e57f.d20241015.cu126 ?

huangeng, 2024-10-30:
> http://jetson.webredirect.org/jp6/cu126/vllm/0.6.4.dev0+gfd47e57f.d20241015.cu126？​

pip list | grep vllm
vllm                              0.6.3+cu126

How can I use the latest one? Uninstall vllm in the container and then pip install vllm?

huangeng, 2024-10-30:
> are you using the latest one? http://jetson.webredirect.org/jp6/cu126/vllm/0.6.4.dev0+gfd47e57f.d20241015.cu126 ?

pip list | grep vllm
vllm 0.6.3+cu126

How can I use the latest one? Uninstall vllm in the container and then pip install vllm?

huangeng, 2024-10-30:
pip list | grep vllm
vllm 0.6.4.dev0+gfd47e57f.d20241015.cu126
I installed the latest vllm (pip install vllm-0.6.4.dev0+gfd47e57f.d20241015.cu126) and it seems that other errors occurred.
```
root@jetson-desktop:/# vllm serve Qwen/Qwen2.5-0.5B-Instruct
/usr/local/lib/python3.10/dist-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 10-30 17:39:48 api_server.py:528] vLLM API server version 0.6.4.dev0+gfd47e57f.d20241015
INFO 10-30 17:39:48 api_server.py:529] args: Namespace(subparser='serve', model_tag='Qwen/Qwen2.5-0.5B-Instruct', config='', host=None, port=8000, uvicorn_log_level='info', allow_credentials=False, allowed_origins=['*'], allowed_methods=['*'], allowed_headers=['*'], api_key=None, lora_modules=None, prompt_adapters=None, chat_template=None, response_role='assistant', ssl_keyfile=None, ssl_certfile=None, ssl_ca_certs=None, ssl_cert_reqs=0, root_path=None, middleware=[], return_tokens_as_token_ids=False, disable_frontend_multiprocessing=False, enable_auto_tool_choice=False, tool_call_parser=None, tool_parser_plugin='', model='Qwen/Qwen2.5-0.5B-Instruct', tokenizer=None, skip_tokenizer_init=False, revision=None, code_revision=None, tokenizer_revision=None, tokenizer_mode='auto', trust_remote_code=False, download_dir=None, load_format='auto', config_format='auto', dtype='auto', kv_cache_dtype='auto', quantization_param_path=None, max_model_len=None, guided_decoding_backend='outlines', distributed_executor_backend=None, worker_use_ray=False, pipeline_parallel_size=1, tensor_parallel_size=1, max_parallel_loading_workers=None, ray_workers_use_nsight=False, block_size=16, enable_prefix_caching=False, disable_sliding_window=False, use_v2_block_manager=True, num_lookahead_slots=0, seed=0, swap_space=4, cpu_offload_gb=0, gpu_memory_utilization=0.9, num_gpu_blocks_override=None, max_num_batched_tokens=None, max_num_seqs=256, max_logprobs=20, disable_log_stats=False, quantization=None, rope_scaling=None, rope_theta=None, enforce_eager=False, max_context_len_to_capture=None, max_seq_len_to_capture=8192, disable_custom_all_reduce=False, tokenizer_pool_size=0, tokenizer_pool_type='ray', tokenizer_pool_extra_config=None, limit_mm_per_prompt=None, mm_processor_kwargs=None, enable_lora=False, max_loras=1, max_lora_rank=16, lora_extra_vocab_size=256, lora_dtype='auto', long_lora_scaling_factors=None, max_cpu_loras=None, fully_sharded_loras=False, enable_prompt_adapter=False, max_prompt_adapters=1, max_prompt_adapter_token=0, device='auto', num_scheduler_steps=1, multi_step_stream_outputs=True, scheduler_delay_factor=0.0, enable_chunked_prefill=None, speculative_model=None, speculative_model_quantization=None, num_speculative_tokens=None, speculative_disable_mqa_scorer=False, speculative_draft_tensor_parallel_size=None, speculative_max_model_len=None, speculative_disable_by_batch_size=None, ngram_prompt_lookup_max=None, ngram_prompt_lookup_min=None, spec_decoding_acceptance_method='rejection_sampler', typical_acceptance_sampler_posterior_threshold=None, typical_acceptance_sampler_posterior_alpha=None, disable_logprobs_during_spec_decoding=None, model_loader_extra_config=None, ignore_patterns=[], preemption_mode=None, served_model_name=None, qlora_adapter_name_or_path=None, otlp_traces_endpoint=None, collect_detailed_traces=None, disable_async_output_proc=False, override_neuron_config=None, scheduling_policy='fcfs', disable_log_requests=False, max_log_len=None, disable_fastapi_docs=False, dispatch_function=<function serve at 0xffff25f61b40>)
INFO 10-30 17:39:48 api_server.py:166] Multiprocessing frontend to use ipc:///tmp/4fc64ff5-3fe9-4b00-b9fd-63258342c276 for IPC Path.
INFO 10-30 17:39:48 api_server.py:179] Started engine process with PID 132
/usr/local/lib/python3.10/dist-packages/transformers/utils/hub.py:128: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
INFO 10-30 17:40:09 llm_engine.py:237] Initializing an LLM engine (v0.6.4.dev0+gfd47e57f.d20241015) with config: model='Qwen/Qwen2.5-0.5B-Instruct', speculative_config=None, tokenizer='Qwen/Qwen2.5-0.5B-Instruct', skip_tokenizer_init=False, tokenizer_mode=auto, revision=None, override_neuron_config=None, rope_scaling=None, rope_theta=None, tokenizer_revision=None, trust_remote_code=False, dtype=torch.bfloat16, max_seq_len=32768, download_dir=None, load_format=LoadFormat.AUTO, tensor_parallel_size=1, pipeline_parallel_size=1, disable_custom_all_reduce=False, quantization=None, enforce_eager=False, kv_cache_dtype=auto, quantization_param_path=None, device_config=cuda, decoding_config=DecodingConfig(guided_decoding_backend='outlines'), observability_config=ObservabilityConfig(otlp_traces_endpoint=None, collect_model_forward_time=False, collect_model_execute_time=False), seed=0, served_model_name=Qwen/Qwen2.5-0.5B-Instruct, use_v2_block_manager=True, num_scheduler_steps=1, chunked_prefill_enabled=False multi_step_stream_outputs=True, enable_prefix_caching=False, use_async_output_proc=True, use_cached_outputs=True, mm_processor_kwargs=None)
INFO 10-30 17:40:11 model_runner.py:1060] Starting to load model Qwen/Qwen2.5-0.5B-Instruct...
INFO 10-30 17:40:12 weight_utils.py:243] Using model weights format ['*.safetensors']
INFO 10-30 17:40:13 weight_utils.py:288] No model.safetensors.index.json found in remote.
[... cut ...]
  File "/usr/local/lib/python3.10/dist-packages/vllm/entrypoints/openai/api_server.py", line 107, in build_async_engine_client
    async with build_async_engine_client_from_engine_args(
  File "/usr/lib/python3.10/contextlib.py", line 199, in __aenter__
    return await anext(self.gen)
  File "/usr/local/lib/python3.10/dist-packages/vllm/entrypoints/openai/api_server.py", line 194, in build_async_engine_client_from_engine_args
    raise RuntimeError(
RuntimeError: Engine process failed to start

```

johnnynunez, 2024-10-30:
https://github.com/dusty-nv/jetson-containers/pull/705#issue-2623535276

johnnynunez, 2024-10-30:
@dusty-nv i'm checked with the PR and not errors. With new version,  only that @huangeng you have to limit memory use

johnnynunez, 2024-10-30:
@dusty-nv I can share you the whl

johnnynunez, 2024-10-30:
@huangeng try this one: https://drive.google.com/file/d/1_C2e0UeIBOoCkK12qCjWya2DL2n5lf1G/view?usp=sharing

dusty-nv, 2024-10-30:
OK, thanks @johnnynunez, I merged https://github.com/dusty-nv/jetson-containers/pull/705#issue-2623535276 and trying to rebuild it now 👍 

Also there is an open PR on vLLM adding Jetson support, hopefully in the future our paths here will merge and not require patches 🙏 

Also @huangeng did this issue just occur with Qwen?   We had been testing vLLM with Llama models but will expand it to Qwen.

dusty-nv, 2024-10-30:
OK this built, and the wheel is up on the pip server here:  http://jetson.webredirect.org/jp6/cu126/vllm/0.6.3.post1+cu126

johnnynunez, 2024-10-30:
> OK, thanks @johnnynunez, I merged [#705 (comment)](https://github.com/dusty-nv/jetson-containers/pull/705#issue-2623535276) and trying to rebuild it now 👍
> 
> Also there is an open PR on vLLM adding Jetson support, hopefully in the future our paths here will merge and not require patches 🙏
> 
> Also @huangeng did this issue just occur with Qwen? We had been testing vLLM with Llama models but will expand it to Qwen.

I used qwen and works for me with latest version

huangeng, 2024-10-31:
> I used qwen and works for me with latest version

I haven't tried llama model.
I tried 0.6.3. post1+cu126 and it didn't work for me. Can you send me your execution steps?

johnnynunez, 2024-10-31:
jetson-containers run -it dustynv/vllm:0.6.3-r36.4.0
vllm serve Qwen/Qwen2.5-0.5B-Instruct —gpu-memory-utilization=0.3

huangeng, 2024-10-31:
> jetson-containers run -it dustynv/vllm:0.6.3-r36.4.0 vllm serve Qwen/Qwen2.5-0.5B-Instruct —gpu-memory-utilization=0.3

Are you not using the latest 0.6.3.post1+cu126?

huangeng, 2024-10-31:
> jetson-containers run -it dustynv/vllm:0.6.3-r36.4.0 vllm serve Qwen/Qwen2.5-0.5B-Instruct —gpu-memory-utilization=0.3

I limited the memory to 0.5 and it worked, why does it fail to start when I give more memory?

huangeng, 2024-10-31:
> jetson-containers run -it dustynv/vllm:0.6.3-r36.4.0 vllm serve Qwen/Qwen2.5-0.5B-Instruct —gpu-memory-utilization=0.3

Limiting memory to 0.5 worked for me👍

huangeng, 2024-10-31:
> > jetson-containers run -it dustynv/vllm:0.6.3-r36.4.0 vllm serve Qwen/Qwen2.5-0.5B-Instruct —gpu-memory-utilization=0.3
> 
> I limited the memory to 0.5 and it worked, why does it fail to start when I give more memory?

Could this be a machine configuration issue where the machine is killed when the memory is full?

huangeng, 2024-10-31:
> jetson-containers run -it dustynv/vllm:0.6.3-r36.4.0 vllm serve Qwen/Qwen2.5-0.5B-Instruct —gpu-memory-utilization=0.3

My qwen2.5 0.5B model can run successfully with a memory limit of 0.5, but the 7B model will fail.

johnnynunez, 2024-10-31:
> My qwen2.5 0.5B model can run successfully with a memory limit of 0.5, but the 7B model will fail.

AH you have 16gb right?
So, you have to increase swap memory or not use cuda graph.
https://docs.vllm.ai/en/stable/models/engine_args.html

huangeng, 2024-11-01:
> > My qwen2.5 0.5B model can run successfully with a memory limit of 0.5, but the 7B model will fail.
> 
> AH! Do you have 16gb, right? So, you have to increase swap memory or not use cuda graph. https://docs.vllm.ai/en/stable/models/engine_args.html
> 
> the problem of vllm is that the control of use memory is from user, so if you don't limit memory, you will have segmentation fault or crash directly.

I have started it successfully, but an error occurred during the request.
```
OSError: ctypes.util.find_library() did not manage to locate a library called 'sndfile'

```
Need to install libsndfile1 package

apt-get install libsndfile1

johnnynunez, 2024-11-01:
We can add to container

closed 2024-11-19 by huangeng