dusty-nv/jetson-containers #706: Unit docker.service not found

ccl-private, 2024-10-31:
rt

ToeiRei, 2024-11-02:
Please give more details. I mean, docker.service points to docker not being installed - but that's just a guess.

dusty-nv, 2024-11-04:
@ccl-private normally jetpack comes with nvidia-container-runtime and docker preinstalled - but you can try `apt-get install docker.io nvidia-container-toolkit`

ccl-private, 2024-11-07:
I only flashed the Ubuntu system. 
And I modified Nvidia apt source manually.
wget https://repo.download.nvidia.com/jetson/jetson-ota-public.asc -o /etc/apt/trusted.gpg.d/jetson-ota-public.asc
chmod 644 /etc/apt/trusted.gpg.d/jetson-ota-public.asc \
&& apt-get update && apt-get install -y --no-install-recommends \
    ca-certificates \
&& echo "deb https://repo.download.nvidia.com/jetson/common r36.2 main" > /etc/apt/sources.list.d/nvidia-l4t-apt-source.list \
&& echo "deb https://repo.download.nvidia.com/jetson/${SOC} r36.2 main" >> /etc/apt/sources.list.d/nvidia-l4t-apt-source.list \
&& apt-get update \

ccl-private, 2024-11-07:
I have given up on manually modifying the apt source.
I'm trying to re-flash it with Cuda-runtime.

closed 2025-02-21 by johnnynunez