dusty-nv/jetson-containers #710: Error build any container dev branch

johnnynunez, 2024-11-01:
```bash
Successfully tagged comfyui:r36.4.0-cu126-cp310-cuda_12.6
-- Testing container comfyui:r36.4.0-cu126-cp310-cuda_12.6 (cuda:12.6/test.sh)

sudo docker run -t --rm --runtime=nvidia --network=host \
--volume /home/johnny/Projects/jetson-containers/packages/cuda/cuda:/test \
--volume /home/johnny/Projects/jetson-containers/data:/data \
--workdir /test \
comfyui:r36.4.0-cu126-cp310-cuda_12.6 \
/bin/bash -c '/bin/bash test.sh' \
2>&1 | tee /home/johnny/Projects/jetson-containers/logs/20241101_113724/test/comfyui_r36.4.0-cu126-cp310-cuda_12.6_test.sh.txt; exit ${PIPESTATUS[0]}

docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #1: error running hook: exit status 1, stdout: , stderr: time="2024-11-01T11:45:44+01:00" level=info msg="Symlinking /var/lib/docker/overlay2/dcd26e757ebfe4ad2594fafbd35068b9fec840b219a362faba166eda10551b04/merged/usr/lib/aarch64-linux-gnu/nvidia/libgstnvcustomhelper.so to libgstnvcustomhelper.so.1.0.0"
time="2024-11-01T11:45:44+01:00" level=info msg="Symlinking /var/lib/docker/overlay2/dcd26e757ebfe4ad2594fafbd35068b9fec840b219a362faba166eda10551b04/merged/usr/lib/aarch64-linux-gnu/nvidia/libgstnvdsseimeta.so to libgstnvdsseimeta.so.1.0.0"
time="2024-11-01T11:45:44+01:00" level=error msg="failed to create link [/usr/lib/aarch64-linux-gnu/nvidia/nvidia_icd.json /etc/vulkan/icd.d/nvidia_icd.json]: failed to check if link exists: unexpected link target: /var/lib/docker/overlay2/dcd26e757ebfe4ad2594fafbd35068b9fec840b219a362faba166eda10551b04/merged/etc/vulkan/icd.d/nvidia_icd.json": unknown.
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/johnny/Projects/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/home/johnny/Projects/jetson-containers/jetson_containers/container.py", line 154, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/johnny/Projects/jetson-containers/jetson_containers/container.py", line 327, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo docker run -t --rm --runtime=nvidia --network=host --volume /home/johnny/Projects/jetson-containers/packages/cuda/cuda:/test --volume /home/johnny/Projects/jetson-containers/data:/data --workdir /test comfyui:r36.4.0-cu126-cp310-cuda_12.6 /bin/bash -c '/bin/bash test.sh' 2>&1 | tee /home/johnny/Projects/jetson-containers/logs/20241101_113724/test/comfyui_r36.4.0-cu126-cp310-cuda_12.6_test.sh.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 125.

```

dusty-nv, 2024-11-01:
> @dusty-nv<https://github.com/dusty-nv> master also doesn't work... seems that nvidia changed docker image l4t?

Had you apt-upgraded recently?  That OCI error can happen in the past if the NV container runtime has a version conflict...

johnnynunez, 2024-11-01:
> Had you apt-upgraded recently? That OCI error can happen in the past if the NV container runtime has a version conflict...
> […](#)

Yes,
I had an upgrade today from apt, tensorrt and I don't remember if libcontainer too.
```bash
johnny@johnny-jetson:~/Projects/Asier$ apt list --installed | grep nvidia

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libnvidia-container-tools/unknown,now 1.17.0-1 arm64 [installed]
libnvidia-container1/unknown,now 1.17.0-1 arm64 [installed]
nvidia-container-toolkit-base/unknown,now 1.17.0-1 arm64 [installed]
nvidia-container-toolkit/unknown,now 1.17.0-1 arm64 [installed]
nvidia-l4t-3d-core/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-apt-source/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-bootloader/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-camera/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-configs/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-core/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-cuda-utils/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-cuda/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-cudadebuggingsupport/stable,now 12.6-34622040.0 arm64 [installed]
nvidia-l4t-display-kernel/stable,now 5.15.148-tegra-36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-dla-compiler/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-firmware/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-gbm/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-graphics-demos/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-gstreamer/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-init/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-initrd/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-jetson-io/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-jetson-multimedia-api/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-jetsonpower-gui-tools/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-kernel-dtbs/stable,now 5.15.148-tegra-36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-kernel-headers/stable,now 5.15.148-tegra-36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-kernel-oot-headers/stable,now 5.15.148-tegra-36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-kernel-oot-modules/stable,now 5.15.148-tegra-36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-kernel/stable,now 5.15.148-tegra-36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-libwayland-client0/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-libwayland-cursor0/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-libwayland-egl1/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-libwayland-server0/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-multimedia-utils/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-multimedia/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-nvfancontrol/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-nvml/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-nvpmodel-gui-tools/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-nvpmodel/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-nvsci/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-oem-config/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-openwfd/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-optee/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-pva/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-tools/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-vulkan-sc-dev/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-vulkan-sc-samples/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-vulkan-sc-sdk/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-vulkan-sc/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-wayland/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-weston/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-x11/stable,now 36.4.0-20240912212859 arm64 [installed]
nvidia-l4t-xusb-firmware/stable,now 36.4.0-20240912212859 arm64 [installed]

```

dusty-nv, 2024-11-01:
Ok, typically one doesn't not upgrade the nv container runtime because those kind of OCI errors can happen

johnnynunez, 2024-11-01:
interesting:

closed 2024-11-01 by johnnynunez