dusty-nv/jetson-containers #711: Building Ollama Container for Jetson Xavier-NX failed for a reference to llm/generate

apulache, 2024-11-03:
I run jetson-container build ollama and got an error:

Step 8/17 : ENV LD_LIBRARY_PATH=/usr/local/nvidia/lib:/usr/local/nvidia/lib64:/usr/local/nvidia/compat:${LD_LIBRARY_PATH}     CMAKE_CUDA_ARCHITECTURES=${CMAKE_CUDA_ARCHITECTURES}
 ---> Running in adf11ae4f612
Removing intermediate container adf11ae4f612
 ---> 8d0320fcebd5
Step 9/17 : RUN cd /opt/ollama &&     export VERSION="0.0.0" &&     export GOFLAGS="'-ldflags=-w -s \"-X=github.com/ollama/ollama/version.Version=$VERSION\" \"-X=github.com/ollama/ollama/server.mode=release\"'" &&     export GOARCH=arm64 &&     echo "OLLAMA_VERSION=${VERSION}  GOFLAGS=${GOFLAGS}" &&     cd llm/generate &&     sed 's|-j8|-j$(nproc)|' -i gen_common.sh &&     bash gen_linux.sh &&     cd ../../ &&     go build -trimpath .
 ---> Running in b57e5ae15583
**/bin/sh: 1: cd: OLLAMA_VERSION=0.0.0  GOFLAGS='-ldflags=-w -s "-X=github.com/ollama/ollama/version.Version=0.0.0" "-X=github.com/ollama/ollama/server.mode=release"'
can't cd to llm/generate**
The command '/bin/sh -c cd /opt/ollama &&     export VERSION="0.0.0" &&     export GOFLAGS="'-ldflags=-w -s \"-X=github.com/ollama/ollama/version.Version=$VERSION\" \"-X=github.com/ollama/ollama/server.mode=release\"'" &&     export GOARCH=arm64 &&     echo "OLLAMA_VERSION=${VERSION}  GOFLAGS=${GOFLAGS}" &&     cd llm/generate &&     sed 's|-j8|-j$(nproc)|' -i gen_common.sh &&     bash gen_linux.sh &&     cd ../../ &&     go build -trimpath .' returned a non-zero code: 2
Traceback (most recent call last):
  File "/home/apulache/miniforge3/envs/jetson_containers/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/apulache/miniforge3/envs/jetson_containers/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/apulache/Workspace/jetson-containers/jetson_containers/build.py", line 112, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.build_args, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api, args.skip_packages)
  File "/home/apulache/Workspace/jetson-containers/jetson_containers/container.py", line 147, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/home/apulache/miniforge3/envs/jetson_containers/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag ollama:r35.5.0-ollama --file /home/apulache/Workspace/jetson-containers/packages/llm/ollama/Dockerfile --build-arg BASE_IMAGE=ollama:r35.5.0-cuda --build-arg OLLAMA_REPO="ollama/ollama" --build-arg OLLAMA_BRANCH="main" --build-arg GOLANG_VERSION="1.22.1" --build-arg CMAKE_VERSION="3.22.1" --build-arg JETPACK_VERSION="5.1" --build-arg CMAKE_CUDA_ARCHITECTURES="72;87" /home/apulache/Workspace/jetson-containers/packages/llm/ollama 2>&1 | tee /home/apulache/Workspace/jetson-containers/logs/20241103_011216/build/ollama_r35.5.0-ollama.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 2.

I'm using a Jetson Xavier-NX with Jetpack 5.1.3 and CUDA 11.4.315 on Ubuntu 20.04
My python environment was created with Mamba and has Python 3.10

Here the full log:

jetson-containers run $(autotag ollama)
Namespace(packages=['ollama'], prefer=['local', 'registry', 'build'], disable=[''], user='dustynv', output='/tmp/autotag', quiet=False, verbose=False)
-- L4T_VERSION=35.5.0  JETPACK_VERSION=5.1  CUDA_VERSION=11.4
-- Finding compatible container image for ['ollama']

Found compatible container dustynv/ollama:r35.4.1 (2024-06-25, 5.4GB) - would you like to pull it? [Y/n] n

Couldn't find a compatible container for ollama, would you like to build it? [y/N] y
-- Package l4t-ml has missing dependencies, disabling...  ("couldn't find package:  tensorflow2")
-- Package l4t-tensorflow:tf1 has missing dependencies, disabling...  ("couldn't find package:  tensorflow")
-- Package l4t-tensorflow:tf2 has missing dependencies, disabling...  ("couldn't find package:  tensorflow2")
-- Package comfyui has missing dependencies, disabling...  ("couldn't find package:  torchao")
-- Building containers  ['build-essential', 'pip_cache:cu114', 'cuda', 'ollama']
-- Building container ollama:r35.5.0-build-essential

DOCKER_BUILDKIT=0 docker build --network=host --tag ollama:r35.5.0-build-essential \
--file /home/apulache/Workspace/jetson-containers/packages/build/build-essential/Dockerfile \
--build-arg BASE_IMAGE=nvcr.io/nvidia/l4t-jetpack:r35.4.1 \
/home/apulache/Workspace/jetson-containers/packages/build/build-essential \
2>&1 | tee /home/apulache/Workspace/jetson-containers/logs/20241103_005258/build/ollama_r35.5.0-build-essential.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  19.97kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> 5c923ac521a3
Step 3/5 : ENV DEBIAN_FRONTEND=noninteractive     LANGUAGE=en_US:en     LANG=en_US.UTF-8     LC_ALL=en_US.UTF-8
 ---> Running in 85833e70c483
Removing intermediate container 85833e70c483
 ---> 7bab7e573b8a
Step 4/5 : RUN set -ex     && apt-get update     && apt-get install -y --no-install-recommends         locales         locales-all         tzdata     && locale-gen en_US $LANG     && update-locale LC_ALL=$LC_ALL LANG=$LANG     && locale         && apt-get install -y --no-install-recommends         build-essential         software-properties-common         apt-transport-https         ca-certificates         lsb-release         pkg-config         gnupg         git         gdb         wget         curl         nano         zip         unzip         time 	   sshpass 	   ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*         && gcc --version     && g++ --version
 ---> Running in b9b5d38123e8
+ apt-get update
Get:1 https://repo.download.nvidia.com/jetson/common r35.4 InRelease [2,555 B]
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Get:3 https://repo.download.nvidia.com/jetson/common r35.4/main arm64 Packages [19.6 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [128 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [128 kB]
[... 798 lines cut ...]
/bin/sh: 1: cd: can't cd to llm/generate
The command '/bin/sh -c cd /opt/ollama &&     export VERSION="0.0.0" &&     export GOFLAGS="'-ldflags=-w -s \"-X=github.com/ollama/ollama/version.Version=$VERSION\" \"-X=github.com/ollama/ollama/server.mode=release\"'" &&     export GOARCH=arm64 &&     echo "OLLAMA_VERSION=${VERSION}  GOFLAGS=${GOFLAGS}" &&     cd llm/generate &&     sed 's|-j8|-j$(nproc)|' -i gen_common.sh &&     bash gen_linux.sh &&     cd ../../ &&     go build -trimpath .' returned a non-zero code: 2
Traceback (most recent call last):
  File "/home/apulache/miniforge3/envs/jetson_containers/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/apulache/miniforge3/envs/jetson_containers/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/apulache/Workspace/jetson-containers/jetson_containers/tag.py", line 58, in <module>
    image = find_container(args.packages[0], prefer_sources=args.prefer, disable_sources=args.disable, user=args.user, quiet=args.quiet)
  File "/home/apulache/Workspace/jetson-containers/jetson_containers/container.py", line 537, in find_container
    return build_container('', package) #, simulate=True)
  File "/home/apulache/Workspace/jetson-containers/jetson_containers/container.py", line 147, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/home/apulache/miniforge3/envs/jetson_containers/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag ollama:r35.5.0-ollama --file /home/apulache/Workspace/jetson-containers/packages/llm/ollama/Dockerfile --build-arg BASE_IMAGE=ollama:r35.5.0-cuda --build-arg OLLAMA_REPO="ollama/ollama" --build-arg OLLAMA_BRANCH="main" --build-arg GOLANG_VERSION="1.22.1" --build-arg CMAKE_VERSI [... cut ...] user/1000/pulse:/run/user/1000/pulse --device /dev/bus/usb --device /dev/i2c-0 --device /dev/i2c-1 --device /dev/i2c-2 --device /dev/i2c-3 --device /dev/i2c-4 --device /dev/i2c-5 --device /dev/i2c-6 --device /dev/i2c-7 --device /dev/i2c-8 --device /dev/i2c-9 -v /run/jtop.sock:/run/jtop.sock --name jetson_container_20241103_005656
"docker run" requires at least 1 argument.
See 'docker run --help'.

Usage:  docker run [OPTIONS] IMAGE [COMMAND] [ARG...]

Create and run a new container from an image

dusty-nv, 2024-11-05:
Hi @apulache, ollama updated their build process in https://github.com/ollama/ollama/commit/b754f5a6a36d6f3068e938af525d8b056c7f9bce#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52 , and I just updated jetson-containers for it with https://github.com/dusty-nv/jetson-containers/commit/43406820000146896b72ec946d93ac0f60343ce0 - let me know if you encounter additional errors after pulling the latest :+1:

LukeGotWood, 2024-11-07:
I did a little bit more digging, I rebuilt the container to confirm it wasn't just a transient issue but it still persists for me.

I did notice during the build that libggml_cuda_v12.so is generated and zipped up so hopefully its just an easy fix?
I don't know where to look for it in the build process but currently but I can have a dig tomorrow.
```
cp -af /opt/ollama/llama/build/linux-arm64/runners/cuda_v12/libggml_cuda_v12.so /opt/ollama/dist/linux-arm64/lib/ollama/libggml_cuda_v12.so

gzip --best -c /opt/ollama/llama/build/linux-arm64/runners/cuda_v12/ollama_llama_server > /opt/ollama/build/linux/arm64/cuda_v12/ollama_llama_server.gz
cp -af /opt/ollama/llama/build/linux-arm64/runners/cuda_v12/ollama_llama_server /opt/ollama/dist/linux-arm64/lib/ollama/runners/cuda_v12/ollama_llama_server
```

LukeGotWood, 2024-11-07:
Your fix worked great, thank you 👍

And agree with the extra testing. The SmolLM2 model you have set as default is the smallest I could find at a glance too so seems like a really good suggestion that shouldn't slow down the build process to much ether!

hillct, 2024-11-08:
I see the same issue when building on an AGX Orin 32GB.
Pulled updates, got commit https://github.com/dusty-nv/jetson-containers/commit/f38a23025619afd95ae79ca7c3e1c609964c3117
ran `jetson-containers run --name ollama $(autotag ollama)`
There were no built-time errors, however the tests failed primarily with the same missing CUDA_12 library:

```Successfully built 24205df91fa7
Successfully tagged ollama:r36.3.0-ollama
-- Testing container ollama:r36.3.0-ollama (ollama/test.sh)

docker run -t --rm --runtime=nvidia --network=host \
--volume /home/hillct/jetson-containers/packages/llm/ollama:/test \
--volume /home/hillct/jetson-containers/data:/data \
--workdir /test \
ollama:r36.3.0-ollama \
/bin/bash -c '/bin/bash test.sh' \
2>&1 | tee /home/hillct/jetson-containers/logs/20241108_151255/test/ollama_r36.3.0-ollama_test.sh.txt; exit ${PIPESTATUS[0]}

testing ollama
Warning: could not connect to a running Ollama instance
Warning: client version is 0.0.0
Large language model runner

Usage:
  ollama [flags]
  ollama [command]

Available Commands:
  serve       Start ollama
  create      Create a model from a Modelfile
  show        Show information for a model
  run         Run a model
  stop        Stop a running model
  pull        Pull a model from a registry
  push        Push a model to a registry
  list        List models
  ps          List running models
  cp          Copy a model
  rm          Remove a model
  help        Help about any command

Flags:
  -h, --help      help for ollama
  -v, --version   Show version information

Use "ollama [command] --help" for more information about a command.
-- Tagging container ollama:r36.3.0-ollama -> ollama:r36.3.0
docker tag ollama:r36.3.0-ollama ollama:r36.3.0

-- Testing container ollama:r36.3.0 (cuda:12.2/test.sh)

docker run -t --rm --runtime=nvidia --network=host \
--volume /home/hillct/jetson-containers/packages/cuda/cuda:/test \
--volume /home/hillct/jetson-containers/data:/data \
--workdir /test \
ollama:r36.3.0 \
/bin/bash -c '/bin/bash test.sh' \
2>&1 | tee /home/hillct/jetson-containers/logs/20241108_151255/test/ollama_r36.3.0_test.sh.txt; exit ${PIPESTATUS[0]}

{
   "cuda" : {
      "name" : "CUDA SDK",
      "version" : "12.2.12"
   },
   "cuda_cccl" : {
      "name" : "CUDA C++ Core Compute Libraries",
      "version" : "12.2.140"
   },
   "cuda_compat" : {
      "name" : "CUDA Specific Libraries",
      "version" : "12.2.34086590"
[... 198 lines cut ...]
time=2024-11-08T15:29:51.454-05:00 level=INFO source=images.go:762 msg="total unused blobs removed: 0"
time=2024-11-08T15:29:51.454-05:00 level=INFO source=routes.go:1240 msg="Listening on [::]:11434 (version 0.0.0)"
time=2024-11-08T15:29:51.455-05:00 level=INFO source=common.go:135 msg="extracting embedded files" dir=/tmp/ollama881695612/runners
time=2024-11-08T15:29:51.555-05:00 level=INFO source=common.go:49 msg="Dynamic LLM libraries" runners="[cuda_v12 cpu]"
time=2024-11-08T15:29:51.555-05:00 level=INFO source=gpu.go:221 msg="looking for compatible GPUs"
time=2024-11-08T15:29:51.555-05:00 level=WARN source=gpu.go:732 msg="unable to locate gpu dependency libraries"
time=2024-11-08T15:29:51.556-05:00 level=WARN source=gpu.go:732 msg="unable to locate gpu dependency libraries"
time=2024-11-08T15:29:51.556-05:00 level=WARN source=gpu.go:732 msg="unable to locate gpu dependency libraries"
time=2024-11-08T15:29:51.662-05:00 level=INFO source=types.go:123 msg="inference compute" id=GPU-128dc705-e97d-5be0-a41d-6e8b4aec6c7e library=cuda variant=jetpack6 compute=8.7 driver=12.2 name=Orin total="30.0 GiB" available="27.6 GiB"
[GIN] 2024/11/08 - 15:30:03 | 200 |        85.6µs |       127.0.0.1 | HEAD     "/"
[GIN] 2024/11/08 - 15:30:03 | 200 |    2.018337ms |       127.0.0.1 | GET      "/api/tags"
[GIN] 2024/11/08 - 15:30:25 | 200 |      52.384µs |       127.0.0.1 | HEAD     "/"
[GIN] 2024/11/08 - 15:30:25 | 200 |   13.130759ms |       127.0.0.1 | POST     "/api/show"
time=2024-11-08T15:30:25.597-05:00 level=INFO source=sched.go:714 msg="new model will fit in available VRAM in single GPU, loading" model=/data/models/ollama/models/blobs/sha256-22a849aafe3ded20e9b6551b02684d8fa911537c35895dd2a1bf9eb70da8f69e gpu=GPU-128dc705-e97d-5be0-a41d-6e8b4aec6c7e parallel=4 available=29649371136 required="14.8 GiB"
time=2024-11-08T15:30:25.714-05:00 level=INFO source=server.go:105 msg="system memory" total="30.0 GiB" free="27.7 GiB" free_swap="15.0 GiB"
time=2024-11-08T15:30:25.715-05:00 level=INFO source=memory.go:343 msg="offload to cuda" layers.requested=-1 layers.model=57 layers.offload=57 layers.split="" memory.available="[27.6 GiB]" memory.gpu_overhead="0 B" memory.required.full="14.8 GiB" memory.required.partial="14.8 GiB" memory.required.kv="1.8 GiB" memory.required.allocations="[14.8 GiB]" memory.weights.total="13.2 GiB" memory.weights.repeating="13.0 GiB" memory.weights.nonrepeating="157.5 MiB" memory.graph.full="832.0 MiB" memory.graph.partial="860.3 MiB"
time=2024-11-08T15:30:25.717-05:00 level=INFO source=server.go:383 msg="starting llama server" cmd="/tmp/ollama881695612/runners/cuda_v12/ollama_llama_server --model /data/models/ollama/models/blobs/sha256-22a849aafe3ded20e9b6551b02684d8fa911537c35895dd2a1bf9eb70da8f69e --ctx-size 8192 --batch-size 512 --n-gpu-layers 57 --threads 12 --parallel 4 --port 38685"
time=2024-11-08T15:30:25.717-05:00 level=INFO source=sched.go:449 msg="loaded runners" count=1
[... cut ...]
time=2024-11-08T15:30:25.718-05:00 level=INFO source=server.go:596 msg="waiting for server to become available" status="llm server error"
**/tmp/ollama881695612/runners/cuda_v12/ollama_llama_server: error while loading shared libraries: libggml_cuda_v12.so: cannot open shared object file: No such file or directory**
time=2024-11-08T15:30:25.969-05:00 level=ERROR source=sched.go:455 msg="error loading llama server" error="llama runner process has terminated: exit status 127"
[GIN] 2024/11/08 - 15:30:25 | 500 |  636.163544ms |       127.0.0.1 | POST     "/api/generate"
root@orindev:/# exit
```

dusty-nv, 2024-11-08:
Thanks @jk-vtp-one and everyone - will cherry-pick your fixes and tests for these 👍 

And you are right @jk-vtp-one about pinning ollama to specific versions like I do other packages that frequently break.  I will also add a `main` config that will always pull the latest for those who want to build it.

dusty-nv, 2024-11-09:
@jk-vtp-one thanks, it is working for me with your tests on JetPack 6.1 / ollama v0.4.0 - https://github.com/dusty-nv/jetson-containers/commit/e18584dfa62d57a031193b4206734dc7014bc502 (this is in jetson-containers dev branch)

davidADSP, 2024-11-10:
This works great for running llama 3.2 11b through ollama on 64gb Orin Dev kit, thanks! However the ollama llama 3.2 90b with q4-k-m still fails for me - can anyone confirm if this should work as well?

Spencer90, 2024-11-23:
> @jk-vtp-one thanks, it is working for me with your tests on JetPack 6.1 / ollama v0.4.0 - [e18584d](https://github.com/dusty-nv/jetson-containers/commit/e18584dfa62d57a031193b4206734dc7014bc502)
> 
> (this is in jetson-containers dev branch and the image is on DockerHub at `dustynv/ollama:0.4.0-r36.4.0`)

[FYI] Hi @dusty-nv, running the most recent `dustynv/ollama:0.4.0-r36.4.0` results in the following error. 
/bin/sh: 1: /start_ollama: Permission denied

hillct, 2024-11-23:
The file not found error relates to the logs directory not having been
created inside the data directory. In my case I was mapping it to a volume
so had to create the directory prior to starting the container. This could
be easily solved by adding a mkdir -p to the start ohana script.

closed 2025-02-21 by johnnynunez