dusty-nv/jetson-containers #712: KeyError: 'alias' in `packages/ml/tensorflow/config.py`

hugomiralles-manta, 2024-11-04:
**Description**:
When attempting to build a container using the `jetson-containers` repository, I encountered a `KeyError` in the `config.py` script located in `packages/ml/tensorflow`. This error appears because the `pkg` dictionary does not contain an `'alias'` key when the code tries to extend it.

**Error Log**:
```
Traceback (most recent call last):
  File "/home/manta/jetson-containers/jetson_containers/packages.py", line 147, in scan_packages
    packages = config_package(package)  # returns a list (including subpackages)
  ...
  File "/home/manta/jetson-containers/packages/ml/tensorflow/config.py", line 99, in <module>
    *tensorflow(
     ^^^^^^^^^^^
  File "/home/manta/jetson-containers/packages/ml/tensorflow/config.py", line 92, in tensorflow
    pkg['alias'].extend(['tensorflow','tensorflow2'])
KeyError: 'alias'
```

**Steps to Reproduce**:
1. Clone the `jetson-containers` repository.
2. Run the `jetson-containers build pytorch` command.
3. Observe the `KeyError` during the build process.

**Environment**:
- L4T Version: 32.7.5
- JetPack Version: 5.1
- Python Version: 3.12
- Host System: Jetson TX2
- OS: Ubuntu 18.04 (Bionic)

johnnynunez, 2024-11-04:
> **Description**: When attempting to build a container using the `jetson-containers` repository, I encountered a `KeyError` in the `config.py` script located in `packages/ml/tensorflow`. This error [...]

tensorflow is for jetpack 6.1

dusty-nv, 2024-11-05:
@hugomiralles-manta just fixed this error on JetPack 4 with https://github.com/dusty-nv/jetson-containers/commit/f38a23025619afd95ae79ca7c3e1c609964c3117 - if you have further issues after pulling the latest, please let me know.

@johnnynunez the containers config scripts still need to not produce errors regardless of jetpack version, because those always get run during every `jetson-containers build` when it enumerates the available package configurations.  For example, here pytorch was trying to be built, not TF.

hugomiralles-manta, 2024-11-06:
Thanks a lot for your quick answer !

I am encountering a new issue when attempting to build a container for PyTorch with Python 3.8 using the `jetson-containers` repository on my Jetson TX2 device. The build process fails during the `install_python.sh` step due to an error in finding a suitable pip distribution.

**Environment**:
- **Device**: Jetson TX2
- **L4T Version**: 32.7.5
- **JetPack Version**: 5.1
- **Python Version**: 3.8
- **CUDA Version**: 10.2
- **OS**: Ubuntu 18.04 (Bionic)

**Steps to Reproduce**:
- Ran the command: `PYTHON_VERSION=3.8 jetson-containers build pytorch`.
- Observed the following error during the build step involving `install_python.sh`:

**Error Log**:
```
+ curl -sS https://bootstrap.pypa.io/get-pip.py
+ python3.8
Looking in indexes: http://jetson.webredirect.org/jp5/cu102
ERROR: Could not find a version that satisfies the requirement pip (from versions: none)
ERROR: No matching distribution found for pip
+ curl -sS https://bootstrap.pypa.io/pip/3.6/get-pip.py
+ python3.6
Looking in indexes: http://jetson.webredirect.org/jp5/cu102
ERROR: Could not find a version that satisfies the requirement pip<22.0 (from versions: none)
ERROR: No matching distribution found for pip<22.0
The command '/bin/sh -c /tmp/install_python.sh' returned a non-zero code: 1
```

hugomiralles-manta, 2025-01-24:
@dusty-nv any idea on how to solve this issue on jetson-tx-2 ?
Thanks for your work

closed 2025-02-21 by johnnynunez