dusty-nv/jetson-containers #714: Undefined symbol error when trying to import tensorrt in dustynv/l4t-pytorch:r36.4.0 container

J4Q8, 2024-11-05:
Hi! 

I have recently started working in a docker container on Jetson AGX Orin 64GB. I started with TensorRT 8.x but then I decided to upgrade to 10.x to hopefully take an advantage of newer optimizations. I tried to follow the steps [here](https://docs.nvidia.com/deeplearning/tensorrt/install-guide/index.html#installing) and using one of your newer containers: [dustynv/l4t-pytorch:r36.4.0](https://hub.docker.com/layers/dustynv/l4t-pytorch/r36.4.0/images/sha256-a05c85def9139c21014546451d3baab44052d7cabe854d937f163390bfd5201b?context=explore) but both resulted in the following error:

`root@tree:/# python3 -c "import tensorrt"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/tensorrt/__init__.py", line 75, in <module>
    from .tensorrt import *
ImportError: /lib/aarch64-linux-gnu/libnvinfer.so.10: undefined symbol: _ZN5nvdla19IConcatenationLayer7setAxisEi`

I tried to resolve it myself for quite some time, however, I was unsuccessful. These are the things I have tried/notices:

1. This happens only for [dustynv/l4t-pytorch:r36.4.0](https://hub.docker.com/layers/dustynv/l4t-pytorch/r36.4.0/images/sha256-a05c85def9139c21014546451d3baab44052d7cabe854d937f163390bfd5201b?context=explore), if I use earlier version [dustynv/l4t-pytorch:r36.2.0](https://hub.docker.com/layers/dustynv/l4t-pytorch/r36.2.0/images/sha256-916e0301423b809edd089f7008692bdfe4567e3a21563dea8779b1edc3b2d697?context=explore) I can easily import tensorrt.
2. When following the official [steps](https://docs.nvidia.com/deeplearning/tensorrt/install-guide/index.html#installing) or installing official jetpack [nvcr.io/nvidia/l4t-jetpack:r36.4.0](https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-jetpack/tags) I first get a different error:
`root@tree:/# python3 -c "import tensorrt"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.10/dist-packages/tensorrt/__init__.py", line 76, in <module>
    from .tensorrt import *
ImportError: libnvcudla.so: cannot open shared object file: No such file or directory`
However, this can be resolved by modifying LD_LIBRARY_PATH such that it `includes/usr/local/cuda/compat`. Once I do it I get the same error as above.
3. I have found this [thread ](https://forums.developer.nvidia.com/t/jetpack-6-0-container-importerror-libnvcudla-so-cannot-open-shared-object-file-no-such-file-or-directory/280296/4) with a similar error and followed the steps suggested by @dusty-nv, i.e. checking the drivers with `root@artifact:/# ls -ll /usr/lib/aarch64-linux-gnu/tegra/libnvdla*` which resulted in:
-rw-r--r-- 1 root root 7487576 Aug 28 09:03 /usr/lib/aarch64-linux-gnu/tegra/libnvdla_compiler.so
-rw-r--r-- 1 root root 6480496 Aug 28 09:03 /usr/lib/aarch64-linux-gnu/tegra/libnvdla_runtime.so`
and checked the csv file in `/etc/nvidia-container-runtime/host-files-for-container.d/`, which did not contain `/usr/lib/aarch64-linux-gnu/tegra/libnvcudla.so`.
4. Given the previous two points I thought that there is something wrong with DLA installation so I tried to reinstall CUDA using steps listed [here ](https://forums.developer.nvidia.com/t/is-it-possible-to-already-use-tensorrt-10-on-jestson-agx-orin/295744/10?u=uzytkownik1363), but for cuda-12.6. Interestingly I could install cuda-toolkit this way but cuda-compat threw the following error:
`root@artifact:/# apt-get -y install cuda-toolkit-12-6 cuda-compat-12-6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
cuda-toolkit-12-6 is already the newest version (12.6.1-1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 cuda-compat-12-6 : PreDepends: nvidia-l4t-core but it is not installable
E: Unable to correct problems, you have held broken packages.`
5. One more suspicious thing I noticed is that after starting the [dustynv/l4t-pytorch:r36.4.0](https://hub.docker.com/layers/dustynv/l4t-pytorch/r36.4.0/images/sha256-a05c85def9139c21014546451d3baab44052d7cabe854d937f163390bfd5201b?context=explore) docker (I ran all dockers using `docker run -it --rm --net=host --runtime nvidia dustynv/l4t-pytorch:r36.4.0`) I cannot find tensorrt installation by using `dpkg -l | grep nvinfer` or `dpkg-query -W tensorrt`, the only way I can find it is through `pip3 show tensorrt`. If this is expected behaviour could you elaborate?

Orin runs on, Ubuntu 20.04 and has CUDA 11.4.

I would be very grateful for help with this issue!

Best,
Jakub

dusty-nv, 2024-11-05:
> Orin runs on, Ubuntu 20.04 and has CUDA 11.4.

Hi @J4Q8, I think your issue is related to trying to run a JetPack 6 container on JetPack 5, and would recommend to reflash the system with JetPack 6.1 / L4T r36.4 if possible.  Actually I am surprised that the r36.2 image worked for you, under some circumstances things like that may work and others not.

J4Q8, 2024-11-05:
I will try that, thanks for quick response!

J4Q8, 2024-11-06:
I reflashed Orin with JetPack 6.1 and the error is no more. Thanks!

closed 2024-11-06 by J4Q8

EricYang3721, 2025-07-15:
Hi @dusty-nv Is that possible to run JetPack 6 container on JetPack 5 jetson board? Thank you! I have some code depend on packages in JetPack 6.

johnnynunez, 2025-07-15:
> Hi [@dusty-nv](https://github.com/dusty-nv) Is that possible to run JetPack 6 container on JetPack 5 jetson board? Thank you! I have some code depend on packages in JetPack 6.

dusty is OOO.... i think that it can't due that is taking some internal files... and also due nvidia drivers