dusty-nv/jetson-containers #716: Reducing downloads / increasing build time

ToeiRei, 2024-11-07:
Hi guys,

I am building containers myself using `jetson-containers build` and it's "a tad slow" - so I decided to look at the build process; My 'benchmark image' is ollama as it builds relatively fast.

My first hurdle is the huge apt repo pull which I solved by having a repo with a local apt-cacher. Speeds things a bit with the builds.

Looking at the Dockerfile and the dependency files, there are - in the case of ollama 
- https://github.com/Kitware/CMake/releases/download/v${CMAKE_VERSION}/cmake-${CMAKE_VERSION}-linux-${INSTALL_ARCH}.tar.gz
- https://dl.google.com/go/go${GOLANG_VERSION}.linux-${GO_ARCH}.tar.gz
among dependencies like cuda.

Is there a 'sane' way to cache or proxy those files?

As to my knowledge, squid won't help me there unless I intercept the HTTPS traffic (i.e. SSL bump) and thus get access to the decrypted content.

dusty-nv, 2024-11-08:
@ToeiRei feel your pain, ollama isn't a great benchmark indicative of most packages though, because most are using pip and those get cached on my pip server at `http://jetson.webredirect.org`    then you see that design pattern in the dockerfiles with `install.sh` trying to install these wheels, then falling back to `build.sh` if needed (wherein it uploads the wheels at the end)

This way has proven to scale for us across many packages, but some like ollama have different install processes that we try to remain compatible with as to avoid breaking changes.  Ollama I typically have to update on a weekly or bi-weekly basis to keep it going...

ToeiRei, 2024-11-08:
Yeah, it's a pain as I want to get a lot stuff built locally.
Pip is the next thing I am able to cache here as well.

Staged builds really help if you're able to keep the images and all the layers. Sadly my ssd isn't too big so I'm forced to purge.
Also spotted some stuff coming in from nvidia which I'd love to cache as that stuff really slows down things...

closed 2025-02-21 by johnnynunez