dusty-nv/jetson-containers #722: failed to build ctranslate2 on Jetson Orin AGX

KlausReinhard, 2024-11-13:
Hei dusty-nv,

I'm currently trying to compile ctranslate2 (jetson-containers build ctranslate2) with various CUDA versions but fail. Im running jp5 with Ubunut 20.04 on the Orin.

CUDA_VERSION=11.4: Will find required/linked packages under http://jetson.webredirect.org/jp5, but only for jp5/cu114, but will fail later due to old CUDA version.

CUDA_VERSION=12.0: Obviously, can't find required/linked packages under http://jetson.webredirect.org/jp5 as stated above due to missing link to jp5/cu120 and fails already there.

Any suggestions how to successfully compile ctranslate2 I would need for faster-whisper. Many thanks in advance for any support.

cheers, klaus

dusty-nv, 2024-11-13:
Hi @KlausReinhard, do you remember the error you were getting later?   I recall ctranslate being one of the trickier ones to build.  I mostly build for JetPack 6.1 now, so if there is nothing else keeping you on JetPack 5, you might want to reflash your Orin going forward if you continue to have issues.  Also you may be aware there is also [whisper_trt](https://github.com/dusty-nv/jetson-containers/tree/master/packages/speech/whisper_trt) which is faster than faster_whisper, but IIRC that only builds on JP6 also.

KlausReinhard, 2024-11-13:
Hi @dusty-nv , for the CUDA=11.4 ist starts building ctranslate 4.2.0...

+ wget --quiet --show-progress --progress=bar:force:noscroll http://jetson.webredirect.org:8000/jp5/cu114/ctranslate2-4.2.0.tar.gz
Building CTranslate2 4.2.0
+ echo 'Building CTranslate2 4.2.0'
+ git clone --branch=v4.2.0 --recursive https://github.com/OpenNMT/CTranslate2.git /opt/CTranslate2
Cloning into '/opt/CTranslate2'...
Note: switching to 'e491a5146c88e61a91da132043aced8e606a0859'.

with C++/C compiler 9.4 (from Ubuntu 20.04) 
 [I guess I read somewhere that ctranslate2 needs to  compiled with C++ 17 ??]
it starts "make -j8" and runs quickly into the following error:

/opt/CTranslate2/include/ctranslate2/ops/flash-attention/flash_fwd_launch_template.h(15): error: incomplete type is not allowed
/opt/CTranslate2/include/ctranslate2/ops/flash-attention/flash_fwd_launch_template.h(15): error: identifier "__grid_constant__" is undefined
/opt/CTranslate2/include/ctranslate2/ops/flash-attention/flash_fwd_launch_template.h(15): error: expected a ")"
/opt/CTranslate2/include/ctranslate2/ops/flash-attention/flash_fwd_launch_template.h(15): error: expected a ";"

But thanks for the advice, will probably ramp-up to JetPack 6.1 and have a look at the whisper_trt.

cheers, klaus

KlausReinhard, 2024-11-14:
I guess issue is closed, thx again.

closed 2024-11-14 by KlausReinhard