dusty-nv/jetson-containers #725: Can run the previous versions images on a Jetpack host?

Ailiee, 2024-11-18:
I tested that the NVIDIA Jetson AGX Orin Developer Kit jetpack 6.4 [L4T 36.4.0] host does not support the torch in the dustynv/l4t-ml: r35.4.1 image to use cuda, but supports the torch in the dustynv/l4t-ml: r36.2.0 image to use cuda.

Some projects are based on the dustynv/l4t-ml: r35.4.1 mirror configuration environment, but it does not work in the jetpack 6.4 [L4T36.4.0] host.

 Does a jetpack version upgrade mean that the environment needs to be reconfigured according to the corresponding version of the base image? Is there a simple way for older versions of Jetpack to call images from previous versions so that the torch in them can use cuda?

dusty-nv, 2024-11-18:
Hi @Ailiee, it depends, but the images are typical not compatible with each other, except for some minor changes in JetPack versions sometimes.  r35.4 -> r36 was a big jump from JetPack5/Ubuntu 20.04 to JetPack6/Ubuntu 22.02, so those definitely needed rebuilt and upgraded.   r36.3 -> r36.4 (JetPack 6.0->6.1) also needed a rebuild to go from CUDA 12.2 to 12.6

Ailiee, 2024-11-22:
Ok, thanks for your open-source project and detailed answer. @dusty-nv

closed 2024-11-22 by Ailiee