dusty-nv/jetson-containers #726: Some python packages installed only for current user cause --user docker run errors

ms1design, 2024-11-18:
Hello @dusty-nv 👋 

I noticed that in some of your `nano...` and `clip_trt` packages, the python modules are built and installed using the `--user` flag:

https://github.com/dusty-nv/jetson-containers/blob/f38a23025619afd95ae79ca7c3e1c609964c3117/packages/vit/nanosam/Dockerfile#L19
https://github.com/dusty-nv/jetson-containers/blob/f38a23025619afd95ae79ca7c3e1c609964c3117/packages/vit/nanosam/Dockerfile#L24
https://github.com/dusty-nv/jetson-containers/blob/f38a23025619afd95ae79ca7c3e1c609964c3117/packages/vectordb/nanodb/Dockerfile#L17
https://github.com/dusty-nv/jetson-containers/blob/f38a23025619afd95ae79ca7c3e1c609964c3117/packages/vit/clip_trt/Dockerfile#L16
https://github.com/dusty-nv/jetson-containers/blob/f38a23025619afd95ae79ca7c3e1c609964c3117/packages/vit/nanoowl/Dockerfile#L25

Is there a specific reason for choosing `--user` instead of a global installation?

The issue arises when running Docker containers using `docker run` with the `--user` flag to specify a custom user and group ID. Since `clip_trt` (or other similar modules) is installed only for the original user, it becomes unavailable when a different user context is used inside the container. This causes errors like `ModuleNotFoundError: No module named 'clip_trt'`.

Would it be possible to modify the installation process to support global installations, or provide guidance on how to make the packages accessible in such scenarios? I can issue a PR for that.

Thank you for your work, and I look forward to your insights! 🙌

dusty-nv, 2024-11-18:
Oh interesting, I was unaware of that limitation.  Yes I use `pip install -e` option on many of the packages that we develop or edit the source for.  Because it makes it easy to mount in a local copy of the repo from the host and just edit it interactively.  

And actually I don't explicitly remember adding the `--user`, just the `-e` flag.   So maybe the `--user` is just some remnant I was carrying on.  Do you think we can still use `-e` and drop `--user`, to keep the mountability but resolve the multiuser issue you are encountering?

ms1design, 2024-11-18:
Hey @dusty-nv, thanks for reply. Correct, dropping just the `--user` flag should resolve my issues. Doing that this way you can still build in editable mode (`-e`) but the module will be installed globally and should be available for all users in  container.

ms1design, 2024-11-20:
resolved in #727

PR #727 (2024-11-19): removed --user flag from pip install commands
fix #726
closed 2024-11-20 by ms1design