dusty-nv/jetson-containers #745: /start_ollama:Permission denied

shouji000, 2024-12-12:
In my jetson agx orin(64G), jetpack 6.1, I have met the follwing the question when i run "jetson-containers run --name ollama $(autotag ollama)", it pulled "dustynv/ollama:0.4.0-r36.4.0"
[image]
But, i run " jetson-containers run --name ollama dustynv/ollama:r36.4.0" , this works well

intelemodelseth, 2024-12-12:
I've previously opened [issue 742](https://github.com/dusty-nv/jetson-containers/issues/742) on this same issue. Please join me there.

shouji000, 2024-12-13:
> I've previously opened [issue 742](https://github.com/dusty-nv/jetson-containers/issues/742) on this same issue. Please join me there.

ok

dusty-nv, 2024-12-13:
https://github.com/dusty-nv/jetson-containers/issues/742#issuecomment-2541040046