dusty-nv/jetson-containers #748: I am unable to install any dependencies, including pytorch, from the jetson-webredirect.org source

cthulhu-tww, 2024-12-17:
RT
[image]
[image]

junzquan, 2024-12-17:
jp6/cu122 has the same issue.

m-gangloff, 2024-12-17:
Same issue on my side for jp6/cu122 since 16.12.2024
[image]

It also affects the mirrored upstream Pypi server, i.e. packages that aren't in the custom pip server are not automatically pulled from PyPi. Another thing I noticed is that https://pypi.ngc.nvidia.com  redirects to https://developer.download.nvidia.com/compute/redist/ which is a blank page.

johnnynunez, 2024-12-18:
@dusty-nv He is moving to new page with certificates

Auc7us, 2025-02-17:
is  https://pypi.jetson-ai-lab.dev down? I keep running into the same error in the last couple of days

> 
> 1.089 Requirement already satisfied: numpy==1.26.4 in /usr/local/lib/python3.10/dist-packages (1.26.4)
> 1.487 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7400>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> 2.078 WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e76d0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> 3.204 WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7880>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> 5.244 WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7a30>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> 9.287 WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7be0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> 9.385 ERROR: Could not find a version that satisfies the requirement pyopengl (from versions: none)
> 9.386 ERROR: No matching distribution found for pyopengl

johnnynunez, 2025-02-17:
> is https://pypi.jetson-ai-lab.dev down? I keep running into the same error for the last couple of days
> 
> > 1.089 Requirement already satisfied: numpy==1.26.4 in /usr/local/lib/python3.10/dist-packages (1.26.4)
> > 1.487 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7400>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> > 2.078 WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e76d0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> > 3.204 WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7880>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> > 5.244 WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7a30>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> > 9.287 WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff8c9e7be0>: Failed to establish a new connection: [Errno 111] Connection refused')': /jp6/cu126/pyopengl/
> > 9.385 ERROR: Could not find a version that satisfies the requirement pyopengl (from versions: none)
> > 9.386 ERROR: No matching distribution found for pyopengl

Yes, it’s down. Dustin is on vacation

closed 2025-02-21 by johnnynunez