dusty-nv/jetson-containers #751: Errors building with L4T-Pytorch and ROS Humble for the Jetpack 6.1

Georgini82, 2024-12-18:
Hello.
I'm using a Jetson Orin 64GB with the JetsonPack 6.1 (36.4.0) installed.
I'm trying to build a container with L4T-pytorch and ros humble, but I'm having some issues. 
This is what have done so far. 

When running the command:

`CUDA_VERSION=12.6 ./jetson-containers build --base=dustynv/ros:humble-desktop-l4t-r36.4.0 --verbose --name vision_base l4t-pytorch`

It start to build the following containers:

`Building containers  ['build-essential', 'pip_cache:cu126', 'cuda:12.6', 'cudnn:9.4', 'python', 'numpy', 'cmake', 'onnx', 'pytorch:2.5', 'torchvision', 'torchaudio', 'tensorrt', 'torch2trt', 'opencv', 'pycuda', 'l4t-pytorch']`

I got an error when building the 'python' image on the /tmp/install_python.sh script:

`+ ln -s /usr/bin/python3.10 /usr/local/bin/python3`
`ln: failed to create symbolic link '/usr/local/bin/python3': File exists`

I force it by adding  **-f**. But after that, when I got to the pytorch image I get more errors. It seems torch==2.5 tries to install sympy==1.13.1 and it can't uninstall sympy==1.9 because it was previous installed by the distutils

`× Cannot uninstall sympy 1.9`
`It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.`

Thanks for the help.

KimDoYoung1997, 2024-12-19:
same with me

mzahana, 2025-01-26:
I am getting the same error. Any solution ?

mzahana, 2025-03-17:
@johnnynunez I see that you closed the issue, but I can not find a solution. Where you able to solve this issue?

Thanks.

Georgini82, 2025-03-18:
Hello all, sorry for the late reply.
I manage to make it work doing the following:

- Edit the packages/build/python/install.sh → add the -f to force the symbolic link creation

[image]

- Edit the packages/pytorch/install.sh → add apt remove python3-sympy -y to avoid compatibility issues

[image]

 
- Build it
 
`CUDA_VERSION=12.6 ./jetson-containers build --base=dustynv/ros:humble-desktop-l4t-r36.4.0 --verbose --name vision_base_jetpack_6.1 l4t-pytorch`

- After the build, install libopencv-dev in the container

`apt install libopencv-dev`

shihangw, 2025-03-19:
This is issue still happening to me

mfruhner, 2025-03-19:
I am seeing the same when running:
`CUDA_VERSION=12.8 jetson-containers build --base=dustynv/pytorch:2.6-r36.4.0-cu128 --name=ros_yolo_torchbase ros:humble-desktop realsense zed`

johnnynunez, 2025-03-19:
we will fix it after GTC, sorry guys, but there are some live demos here @dusty-nv