dusty-nv/jetson-containers #762: Failt buil jetson-inference r36.4.2 (Nano 8gb)

michelc33, 2024-12-31:
Hi,

After a clean Jetson nano 8gb install from SD image and SSD addition 

Trying to run jetson-inference 
```
docker/run.sh 
```
and that is a FAIL

from the terminal ouput
```
Unable to find image 'dustynv/jetson-inference:r36.4.2' locally
docker: Error response from daemon: manifest for dustynv/jetson-inference:r36.4.2 not found: manifest unknown: manifest unknown.
See 'docker run --help'.
```

so I try to build using jetson-containers
```
jetson-containers build jetson-inference
```
and it fail 

From the terminal outpout
```
apt-get clean
+ curl -sS https://bootstrap.pypa.io/get-pip.py
+ python3.10
Looking in indexes: http://jetson.webredirect.org/jp5/cu126
ERROR: Could not find a version that satisfies the requirement pip (from versions: none)
ERROR: No matching distribution found for pip
+ curl -sS https://bootstrap.pypa.io/pip/3.6/get-pip.py
+ python3.6
/tmp/install_python.sh: line 27: python3.6: command not found
curl: (23) Failure writing output to destination
The command '/bin/sh -c /tmp/install_python.sh' returned a non-zero code: 127
```
Thanks in advance !

dusty-nv, 2024-12-31:
Sorry, there is an issue with TRT10 compatibility for >=r36.4 in this repo - even after you compile it, most of the models from this project are no longer supported (this project was made when caffe models were most popular...)

I have been updating the Jetson AI Lab and genAI containers, and hope to circle back to jetson-inference, but it is also all CNNs.  If you are just looking for realtime object detection , I would go with Ultralytics YOLOv11 - they already have good TRT support in their repo.

jetson-utils still builds/runs fine on r36.4 btw

walidnegm, 2025-01-13:
Same issue: 

walidnegm@walid-orin-nano:~/Desktop/HelloAI/jetson-inference$ docker/run.sh
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R36.4.2
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/jetson-inference:r36.4.2
DATA_VOLUME:      --volume /home/walidnegm/Desktop/HelloAI/jetson-inference/data:/jetson-inference/data --volume /home/walidnegm/Desktop/HelloAI/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/walidnegm/Desktop/HelloAI/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/walidnegm/Desktop/HelloAI/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/walidnegm/Desktop/HelloAI/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models --volume /home/walidnegm/Desktop/HelloAI/jetson-inference/python/www/recognizer/data:/jetson-inference/python/www/recognizer/data 
DISPLAY_DEVICE:   -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix 
**Unable to find image 'dustynv/jetson-inference:r36.4.2' locally
docker: Error response from daemon: manifest for dustynv/jetson-inference:r36.4.2 not found: manifest unknown: manifest unknown.
See 'docker run --help'.**

QUESTION: Can I use an existing tag ;ole 36.3.0? 

 sudo docker pull dustynv/jetson-inference:r36.3.0

kyflores, 2025-01-14:
> Hi,
>
> After a clean Jetson nano 8gb install from SD image and SSD addition
>
> Trying to run jetson-inference
>
> ```
> docker/run.sh
> ```
>
> and that is a FAIL
>
> from the terminal ouput
>
> ```
> Unable to find [...]

Hi, I had a similar issue where the final failed command was something about python3.6. The problem was that it was incorrectly detecting jetpack 5 because I had a version of the jetson-containers repo prior it being updated for 36.4.2.

You can see that in this line of your output, it should be jp6:
`Looking in indexes: http://jetson.webredirect.org/jp5/cu126`

If you pull the latest version of jetson-containers and try again the container build should succeed at least.