dusty-nv/jetson-containers #763: GPU Access Issue with NVIDIA JetPack 6.1 Docker Image

kabirkedia, 2025-01-02:
Hello dusty,  
Happy New Year! I’m reaching out for help with an issue we’ve encountered while using the official NVIDIA JetPack 6.1 image. Here's the image we’re using: [[NVIDIA JetPack 6.1 - PyTorch](https://hub.docker.com/layers/dustynv/l4t-pytorch/r36.4.0/images/sha256-a05c85def9139c21014546451d3baab44052d7cabe854d937f163390bfd5201b)](https://hub.docker.com/layers/dustynv/l4t-pytorch/r36.4.0/images/sha256-a05c85def9139c21014546451d3baab44052d7cabe854d937f163390bfd5201b).  

When we run a container based on this image, everything works fine as long as it’s connected to a display and a user is logged in. However, when accessing the container via SSH, GPU access is problematic. Specifically:  

- A non-root user with the same permissions as the user outside the container cannot access the GPU.  
- GPU access works when logged in as root.  

We also tested the same setup with JetPack 5.1, and this issue does not occur in that version.  

If anyone has insights or suggestions on resolving this, please let me know!  

Thanks,  
Kabir Kedia

dusty-nv, 2025-01-10:
Thanks @kabirkedia , admittedly have not seen this ... is the non-root user part of `docker` group ?   Does `--privileged` change the behavior?   I'm not sure it is the containers themselves rather than a change in the docker daemon?

closed 2025-02-21 by johnnynunez

tonynajjar, 2025-02-24:
I have the exact same setup and issue. Works with root and not with my non-root user (that is part of the `docker` group). It used to work with Jetpack 5. I'll try the mentioned workarounds but it would be nice if that could be solved upstream somehow

tonynajjar, 2025-02-24:
I think I found the issue, the non-root user in the container must be part of the `render` group, which from my experience unfortunately does not even exist and needs to be created. See [here](https://developer.ridgerun.com/wiki/index.php/JetPack_6_Migration_and_Developer_Guide/JetPack_6_Key_Features/New_Linux_Distributions#:~:text=%23%20Add%20user%20to%20group%20render%20allows%20running%20GPU%20related%20commands%20as%20non%20root). @kabirkedia maybe you can give it a shot?

Maybe @dusty-nv you have a clearer explanation as why this is necessary?

tonynajjar, 2025-02-24:
https://github.com/dusty-nv/jetson-containers/blob/9889aedb049622f8bbaab3d24ddb6da922ea07af/jetson_containers/l4t_version.py#L231

@dusty-nv why is it that for Jetpack 6 and above we use `ubuntu:22.04` as the base image and not `nvcr.io/nvidia/l4t-jetpack:r36.4.0`? I believe this is where the difference is when it comes to the existence of the `render` group

flxinxout, 2025-06-07:
Hello. I experience exactly the same issue. I have an image based on pytorch:2.1.0 r36. If I change my user to a new user, I cant access the GPU inside my docker container. My host is a jetson orin nx 16BG.

Here is my dockerfile (the base image is based on the pytorch:2.1.0 r36). I run 2 tests:

1. with the line USER $USERNAME => no access GPU
2. without this line => access to GPU

```

FROM ghcr.io/epflxplore/docker_commons:humble-jetson-nx-hd

ENV DEBIAN_FRONTEND=noninteractive

# RUN apt-get update && apt-get install -y --no-install-recommends \
#     git \
#     cmake \
#     build-essential \
#     libssl-dev \
#     libusb-1.0-0-dev \
#     libusb-1.0-0 \
#     libudev-dev \
#     pkg-config \
#     libgtk-3-dev \
#     python3 \
#     python3-setuptools \
#     python3-dev \
#     python3-pip \
#     python3-pybind11 \
#     v4l-utils \
#     libxinerama-dev \
#     libxcursor-dev \
#     libcanberra-gtk-module \
#     libcanberra-gtk3-module \
#     && apt-get clean && rm -rf /var/lib/apt/lists/*

# # Install ROS 2 HD packages
# RUN apt-get update && apt-get upgrade -y
# RUN apt-get install -y --no-install-recommends \
#     ros-${ROS_DISTRO}-moveit \
#     ros-${ROS_DISTRO}-depthai-ros \
#     ros-${ROS_DISTRO}-hardware-interface \
#     ros-${ROS_DISTRO}-control-toolbox \
#     ros-${ROS_DISTRO}-moveit-servo \
#     ros-${ROS_DISTRO}-controller-manager \
#     ros-${ROS_DISTRO}-librealsense2*\
#     ros-${ROS_DISTRO}-realsense2-*\
#     ros-${ROS_DISTRO}-cv-bridge \
#     python3-opencv \
#     libopencv-dev \
#     python3-zmq \
#     libyaml-cpp-dev \
#     lcov

# # Add USB rules
# RUN echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
# RUN /etc/init.d/udev restart 

# RUN pip uninstall torch -y
USER $USERNAME

# # sudo is required for this package (documentation)
# RUN sudo pip install -U jetson-stats

# RUN pip install --no-cache-dir -v evdev depthai pyrealsense2 ultralytics opencv-python opencv-contrib-python==4.7.0.72 numpy==1.24.0 /opt/torch-2.1.0-cp310-cp310-linux_aarch64.whl 

# Set a diretory to store the project
WORKDIR /home/$USERNAME/dev_ws/src
COPY . .

# Set a directory to build the project
WORKDIR /home/$USERNAME/dev_ws

# Clean up
RUN sudo rm -rf /var/lib/apt/lists/*

# Remove all the confidential Xplore source code from the image
RUN sudo rm -rf /home/$USERNAME/dev_ws/src/*
```