dusty-nv/jetson-containers #764: TypeError: unsupported operand type(s) for |: 'type' and 'NoneType' when running jetson-containers command

romris, 2025-01-02:
### Summary
Encountered `TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'` when running the `jetson-containers run $(autotag l4t-pytorch)` command. Same error occurs when running autotag.

### System Information
- **Device**: Jetson AGX Orin 64GB Dev Kit
- **JetPack Version**: 5.1.2
- **Python Version**: 3.8.10
- **Docker Version**: 24.0.7

romris, 2025-01-03:
@mzahana 
Thanks, this works for me! However, will upgrade to the latest JetPack to avoid such issues.

closed 2025-01-03 by romris

Galaxy-Ding, 2025-01-23:
> [@mzahana](https://github.com/mzahana) Thanks, this works for me! However, will upgrade to the latest JetPack to avoid such issues.

I would like to ask what version this upgrades to that solves the problem, currently I have the following version:

Package: nvidia-jetpack
Version: 5.1.2-b104
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-jetpack-runtime (= 5.1.2-b104), nvidia-jetpack-dev (= 5.1.2-b104)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_5.1.2-b104_arm64.deb

Galaxy-Ding, 2025-01-23:
> [@romris](https://github.com/romris) It seems that this error happens because of Python compatibility issue. The current `jetson-containers` assumes Python 3.10. So, on older JetPacks that use Python 3.8 it will throw this error.
> 
> A possible fix is to modify the `handle_text_request` function as follows.
> 
> [jetson-containers/jetson_containers/utils.py](https://github.com/dusty-nv/jetson-containers/blob/63debc7211304b698c1ae00167808328c6290063/jetson_containers/utils.py#L135)
> 
> Line 135 in [63debc7](/dusty-nv/jetson-containers/commit/63debc7211304b698c1ae00167808328c6290063)
> 
>  def handle_text_request(url) -> str | None: 
> Replace with:
> 
> from typing import Optional
> 
> def handle_text_request(url) -> Optional[str]:
>     """
>     Handles a request to fetch text data from the given URL.
> 
>     Args:
>         url (str): The URL from which to fetch text data.
> 
>     Returns:
>         Optional[str]: The fetched text data, stripped of leading and trailing whitespace, 
>                      or None if an error occurs.
>     """
>     try:
>         response = requests.get(url)
>         if response.status_code == 200:
>             return response.text.strip()
>         else:
>             print("Failed to fetch version information. Status code:", response.status_code)
>             return None
>     except Exception as e:
>         print("An error occurred:", e)
>         return None

thanks! Following your changes, I managed to solve the problem