dusty-nv/jetson-containers #765: The latest version of transformers drops support for Python 3.8

Jadeiin, 2025-01-03:
Since [4.47.0](https://github.com/huggingface/transformers/commit/f0e640adfa3cedea53912b95e3093f05cc2b66b5) transformers no longer supports for Python 3.8. And the default python version in Jetpack 5 is 3.8. Should the build version of transformers in config be fixed to 4.46.3?

johnnynunez, 2025-01-03:
> Since [4.47.0](https://github.com/huggingface/transformers/commit/f0e640adfa3cedea53912b95e3093f05cc2b66b5) transformers no longer supports for Python 3.8. And the default python version in Jetpack 5 is 3.8. Should the build version of transformers in config be fixed to 4.46.3?

omg we will need jetpack 7 based on ubuntu 24.04 hahaha
well, you can build containers with "PYTHON_VERSION=3.11 jetcon-containers build pytorch"

Jadeiin, 2025-01-03:
> > Since [4.47.0](https://github.com/huggingface/transformers/commit/f0e640adfa3cedea53912b95e3093f05cc2b66b5) transformers no longer supports for Python 3.8. And the default python version in Jetpack 5 is 3.8. Should the build version of transformers in config be fixed to 4.46.3?
> 
> omg we will need jetpack 7 based on ubuntu 24.04 hahaha well, you can build containers with
> 
> ```shell
> PYTHON_VERSION=3.11 jetson-containers build transformers
> ```

here's a quick fix:
```
package = [
    transformers_pypi('latest', default=(L4T_VERSION.major >= 36), requires='>=36'),            # will always resolve to the latest pypi version
    transformers_pypi('4.46.3', default=(L4T_VERSION.major < 36), requires='<36'),              # 4.46.3 is the last version that supports Python 3.8
    transformers_git('latest', codename='git'),                                                 # will always resolve to the latest git version from huggingface/transformers
    transformers_git('latest', codename='nvgpt', repo='ertkonuk/transformers'),     # will always resolve to the latest git version from ertkonuk/transformers
]

```

dusty-nv, 2025-01-10:
Thanks @Jadeiin !  I just merged this into `dev` branch with commit https://github.com/dusty-nv/jetson-containers/commit/28e011a82994d2aab2344801bc6df57e2c654db4 :+1:

closed 2025-02-21 by johnnynunez