dusty-nv/jetson-containers #768: wyoming-piper build failed

pipacsba, 2025-01-05:
Hello Dusty,

I am struggling with the build process for wyoming-piper.
Any help would be appretiated.

Best,
pipacsba

Build command:
`jetson-containers/jetson-containers run $(autotag wyoming-piper)`

Last lines of the build process:
_[ 94%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/opt/onnxruntime/onnxruntime/contrib_ops/cuda/bert/flash_attention/flash_fwd_hdim192_fp16_sm80.cu.o
Killed
gmake[2]: *** [CMakeFiles/onnxruntime_providers_cuda.dir/build.make:6203: CMakeFiles/onnxruntime_providers_cuda.dir/opt/onnxruntime/onnxruntime/contrib_ops/cuda/bert/flash_attention/flash_fwd_hdim128_bf16_sm80.cu.o] Error 137
gmake[2]: *** Waiting for unfinished jobs...._

Module: NVIDIA Jetson Orin Nano (8GB ram)
SoC: tegra234
L4T: 36.4.0
Jetpack: 6.1

LukeGotWood, 2025-01-07:
I am having trouble building onnxruntime also.
I was hitting the same error you were and was able to get past this error by increasing my swap memory to 16GB using the mounting swap guide [here]( https://www.jetson-ai-lab.com/tips_ram-optimization.html)

I'm now stuck with tests failing which I am assuming is to do with this error:

```+ sshpass -p None scp -o 'StrictHostKeyChecking no' /tmp/tarpack/uploads/onnxruntime-gpu-1.20.0.tar.gz None@localhost:/dist/jp6/cu126/
Permission denied, please try again.
+ echo 'failed to upload tarball'```

pipacsba, 2025-01-11:
Increasing the swap memory to 16GB indeed changed the behavior, but now the Jetson reboots without any failure indication.

```
[ 94%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/opt/onnxruntime/onnxruntime/contrib_ops/cuda/bert/flash_attention/flash_fwd_hdim224_fp16_sm80.cu.o
Connection to orin.lan closed by remote host.

```

IAmIlliest, 2025-08-27:
I'm having a similar reboot issue with no indication as to why when building onnxruntime with CUDA support, on a Jetson Orin Nano Super. The build is happening inside a docker container with all the build tools installed and based off Nvidia's L4T JetPack container (R36.4.4)

These are some relevant flags related to the Orin's cuda arch and recommendations from the onnx docs to speed up build time, They are arg's for onnx's build.sh (wrapper script over CMake):
--cmake_extra_defines \
	     'CMAKE_LIBRARY_ARCHITECTURE=aarch64-linux-gnu' \
            # 'CMAKE_CUDA_ARCHITECTURES=87' \
	    # 'CMAKE_CUDA_ARCHITECTURES=native' \
	    # 'CMAKE_CUDA_ARCHITECTURES=87-real;87-virtual' \
            'onnxruntime_BUILD_UNIT_TESTS=OFF' \
            'onnxruntime_USE_FLASH_ATTENTION=OFF' \
            'onnxruntime_USE_MEMORY_EFFICIENT_ATTENTION=OFF' \
--skip_tests \

I have tried building the specific 87 cuda arch for the Orin and it always fails at the stage below. building for 'native' rather than for any of the 87/87-real/87-virutal specifically, will work. I've tried building with Flash Attention and Memory Efficent Attention set to on, but it fails just the same.

I piped the build output to a log file, with the flags for the build configured as above and these are the last few lines (below). The failure appears to be related to flash attention, which is something that is specifically recommended to be disabled to speed up the build in the onnx docs, but fails whether i enable it or not.
```
#16 3796.7 [ 95%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/000_build/git_clones/onnxruntime/onnxruntime/contrib_ops/cuda/bert/flash_attention/flash_fwd_split_hdim96_fp16_sm80.cu.o
#16 3797.4 [ 95%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/000_build/git_clones/onnxruntime/onnxruntime/contrib_ops/cuda/bert/gemma_rotary_emb_impl.cu.o
#16 3800.9 [ 95%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/000_build/git_clones/onnxruntime/onnxruntime/contrib_ops/cuda/bert/group_query_attention_impl.cu.o
#16 3835.1 [ 95%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/000_build/git_clones/onnxruntime/onnxruntime/contrib_ops/cuda/bert/lean_attention/lean_fwd_hdim128_fp16.cu.o
#16 3855.7 [ 96%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/000_build/git_clones/onnxruntime/onnxruntime/contrib_ops/cuda/bert/lean_attention/lean_fwd_hdim64_fp16.cu.o
#16 3906.8 [ 96%] Building CUDA object CMakeFiles/onnxruntime_providers_cuda.dir/000_build/git_clones/onnxruntime/onnxruntime/contrib_ops/cuda/bert/longformer_attention_impl.cu.o
```
I will try mounting some extra swap and see if I have any luck.

closed 2025-09-28 by pipacsba