dusty-nv/jetson-containers #773: Add support for MoonDream2 on nano_llm.chat

shubhamgupto, 2025-01-10:
Hello

Im just getting started with nano llm and I found https://huggingface.co/vikhyatk/moondream2 perfect for my application. I tried using the nano llm interface and currently the `mlc` api says moondream is currently not supported. 

What would be the best way to add support to this interface? 

Thanks

dusty-nv, 2025-01-10:
@shubhamgupto it does not appear Moondream is in MLC, and I have been moving the VLMs to OpenAI endpoint servers instead of NanoLLM so it is easier to keep updating moving forward.   MLC, TRT-LLM, vLLM, llama.cpp and others have various VLM and support for images in OpenAI chat.completion protocol, however you have to cross-reference what is available.   

Otherwise I would recommend following the Moondream repo or what it's model card advises from deployment, and dig in from there.   All of the VLMs tend to be different unfortunately.   My previous approach was to "deconstruct" them back into LLM and VIT, which typically they have common bases (like Llama or Qwen), so the quantization through MLC/ect would work again.   However in practice this heavy-handed approach became too much to support for all the future VLMs coming out.

shubhamgupto, 2025-01-10:
Makes sense, redeploying VLMs from ground up sounds like a lot of work indeed, I will refer to the authors original deployment code then. Thanks

closed 2025-01-10 by shubhamgupto