dusty-nv/jetson-containers #795: Why does vllm consume so much memory?

HouLingLXH, 2025-01-23:
Jetson Orin NX 16G
jetpack 6.2
docker： [dustynv/vllm:0.6.3-r36.4.0](https://hub.docker.com/r/dustynv/vllm/tags)

Consume more than 16G of memory and terminate:
INFO 01-23 07:32:01 model_runner.py:1060] Starting to load model ./Qwen2-7B-Instruct.Q4_K_M.gguf...
/usr/local/lib/python3.10/dist-packages/torch/nested/__init__.py:226: UserWarning: The PyTorch API of nested tensors is in prototype stage and will change in the near future. (Triggered internally at /opt/pytorch/aten/src/ATen/NestedTensorImpl.cpp:178.)
  return _nested.nested_tensor(
INFO 01-23 07:32:42 model_runner.py:1071] Loading model weights took 4.3752 GB
INFO 01-23 07:34:26 gpu_executor.py:122] # GPU blocks: 11836, # CPU blocks: 4681
INFO 01-23 07:34:26 gpu_executor.py:126] Maximum concurrency for 200 tokens per request: 946.88x
Killed

[image]

By ollama , I can run qwen2.5-7b on the same device.

johnnynunez, 2025-01-23:
Because each framework has differente behavior of how to process the model.
Execution Requirements: The vLLM framework uses additional memory to handle token generation, input/output management, and other tasks. Also, with vllm is all up to you, memory size, input tokens length etc

Also, you are using an old version of vllm. @dusty-nv This docker also must be update because with 0.6.3.post1 we have natively vllm on jetson and it has optimizations for unified memory

HouLingLXH, 2025-01-23:
Is 16G enough to run Qwen2-7B-Instruct.Q4_K_M.gguf in Jetson Orin NX?

dusty-nv, 2025-01-23:
I will be rebuilding vLLM, yes.  And often in "server mode", the default setting may be to pre-allocate more memory than necessary.

leon-seidel, 2025-01-28:
I would recommend to add `--swap_space 0` in the command, otherwise vLLM will allocate another 4 GB of unified memory. Adding `--enforce-eager` helps with some models, too,

closed 2025-04-24 by johnnynunez